MarlinFirmware/Marlin #26064: [BUG] Start print and Configuration icons are missing on status screen

woj-o, 2023-07-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Downloaded today 08.07.2023 - 2.1x bugfix branch.
After compiling and reseting EEPROM printer boots up, heating of nozzle and bed works correctly, but there is not icon to enter menu nor icon to start print from SD card.

Same branch downloaded 12.05.2023 compiled with same configuration files works correctly.

Most probably connected with bugs reported already for UI_480_320.cpp for MKS43 screen.

My printer is using MKS35 screen on Robin nano 1.2 board.
But it's same resolution so same file is used.

Screen of status scree after booting and reseting EEPROM:
[image]

### Bug Timeline

New bug - noticed on 08.07.2023 

### Expected behavior

Both icons for menu are visible and working

### Actual behavior

Buttons for menu and "New print" are not visible

### Steps to Reproduce

1. Download bugfix branch
2. Compile for Robin Nano 1.2 with MKS35 screen
3. Upload, start printer
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/11991342/Configuration.zip)

### Version of Marlin Firmware

Merlin 2.1.x bugfix branch from 08.07.2023

### Printer model

Two Trees Sapphire Pro (SP3) 

### Electronics

MKS Robin Nano v 1.2 

### Add-ons

Replaced driver on Extruder to 2209

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-07-12:
Since a PR to fix this exists, closing this issue

closed 2023-07-12 by ellensp

woj-o, 2023-07-12:
All right - looks good!.
Thanks you for update :).

Wojciech