MarlinFirmware/Marlin #26066: [BUG] VSCode An error ocurred setting up the container

liam-clink, 2023-07-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When trying to open the project in a devcontainer in vscode, both using the most recent release commit and the bugfix-2.1.x branch, on linux, there is an error opening the container. When I open the logfile, it says "No dev container config found. No repository connection string found." I haven't modified any files in the cloned repository.

### Bug Timeline

First time setting it up, so don't know

### Expected behavior

I expected the project to open in a devcontainer

### Actual behavior

Instead it can't open

### Steps to Reproduce

Open project in VSCode, let it install wanted extensions, install docker as instructed, enable virtualization for docker, and then attempt to open the project in a DevContainer

### Version of Marlin Firmware

2.1.2.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

EvilGremlin, 2023-07-09:
afaik, all this new devcontainer stuff is not supported by platformio
so don't use devcontainer (remove plugin)

liam-clink, 2023-07-09:
Oh, I thought platformio depended on docker. So I can just use platformio and I'll be able to build just fine? Perhaps it would be good to add info about this to the installation guide

liam-clink, 2023-07-09:
The reason I say this is that current documentation says to use docker

EvilGremlin, 2023-07-09:
No it doesn't https://marlinfw.org/docs/basics/install_platformio_vscode.html
Though we probably should add note to discard installing recommended plugins.

thisiskeithb, 2023-07-09:
> Though we probably should add note to discard install of recommended plugins.

Better yet, update [`extensions.json`](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.vscode/extensions.json) to exclude Dev Container and Docker.

wilsnat, 2023-07-12:
I think it does still recommend Docker. [When you click on the Install button on the top bar](https://marlinfw.org/docs/basics/install.html):
[image]
"2.1.x | 8-bit AVR, 32-bit ARM | Install Marlin with PlatformIO in a DevContainer"

EvilGremlin, 2023-07-12:
This is an option, not recommendation.

classicrocker883, 2023-07-19:
it's possible to use docker to develop. but there are several steps in order to do so (outside of VS code). but the easiest way to do something like that is Codespaces.

closed 2023-09-28 by github-actions[bot]

thinkyhead, 2023-10-06:
> the easiest way to do something like that is Codespaces

I've lately become kind of fond of these Jupyter thingies at https://colab.research.google.com/ . It would be cool to see one of those set up to build Marlin with some options. Maybe that would be a good option for folks who don't feel like installing anything on their computer, but also don't need something like https://marlin.crc.id.au .

quotschmacher, 2023-10-24:
@EvilGremlin of course dev Containers work with Platformio and the latest Version just needs some fixes in the Container Definition file. I have it running in a devcontainer.