MarlinFirmware/Marlin #26070: [FR] FIX Wifi ESP32U PIN

gianguardo, 2023-07-11:
### Is your feature request related to a problem? Please describe.

Hello, an update has been uploaded regarding the pins for the octopus max ez, to facilitate pairing with the wi-fi if it is possible to change them, I am also attaching the documentation of the biqu.
I would also like to point out that when I activate the #define SERIAL_PORT_3 7 I get errors compiling, how can I fix it?
Thank you
[image]

### Are you looking for hardware support?

Octopus Max Ez 

### Describe the feature you want

  /**
   *                 --------
   *                       GND  |  9   8 | 3.3V
   *          (ESP-CS) PG1  | 10   7 | PB15 (ESP-MOSI) -> PC12
   *                         3.3V | 11   6 | PB14 (ESP-MISO) -> PC11
   *        (ESP-IO0) PG0  | 12   5 | PB13 (ESP-CLK)    -> PC10
   *        (ESP-IO4) PF15 | 13   4 | --
   *                             -- | 14   3 | 3.3V (ESP-EN)
   *          (ESP-RX) PE7  | 15   2 | --
   *          (ESP-TX) PE8  | 16   1 | PB2 (ESP-RST)
   *                 --------
   *                   WIFI
   */

gianguardo, 2023-07-11:
I also tried to enable it but the compiler gives these errors:
serial.h
'MYSERIAL2' was not declared in this scope; did you mean 'MYSERIAL1'?
template argument 2 is invalid
request for member 'write' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
hal.h
#error "SERIAL_PORT_2 must be from 1 to 6, or -1 for Native USB."

My configuration
#define SERIAL_PORT_2 7
#define WIFISUPPORT
How can I fix it? Thank you

EvilGremlin, 2023-07-11:
WIFISUPPORT is not applicable (pay attention to notes) 
but yep UARTs should be expanded to 9

gianguardo, 2023-07-11:
hi, i followed this setup on octopus max ez pin

//
// WIFI
//
#if ENABLED(WIFISUPPORT)
  /**
   *                 --------
   *           GND  |  9   8 | 3.3V
   *  (ESP-CS) PG1  | 10   7 | PB15 (ESP-MOSI)
   *           3.3V | 11   6 | PB14 (ESP-MISO)
   * (ESP-IO0) PG0  | 12   5 | PB13 (ESP-CLK)
   * (ESP-IO4) PF15 | 13   4 | --
   *             -- | 14   3 | 3.3V (ESP-EN)
   *  (ESP-RX) PE7  | 15   2 | --
   *  (ESP-TX) PE8  | 16   1 | PB2 (ESP-RST)
   *                 --------
   *                   WIFI
   */
  #define ESP_WIFI_MODULE_COM                  7  // Must also set either SERIAL_PORT or SERIAL_PORT_2 to this
  #define ESP_WIFI_MODULE_BAUDRATE      BAUDRATE  // Must use same BAUDRATE as SERIAL_PORT & SERIAL_PORT_2
  #define ESP_WIFI_MODULE_RESET_PIN         PB2
  #define ESP_WIFI_MODULE_GPIO0_PIN         PG0
  #define ESP_WIFI_MODULE_GPIO4_PIN         PF15
#endif

EvilGremlin, 2023-07-11:
then disable SERIAL_PORT_2

gianguardo, 2023-07-11:
Ok now compile, mount it and try. Thank you

thisiskeithb, 2023-07-11:
Here are some minimal changes to get the ESP working on the Octopus Max EZ. @ellensp had me test this out a little while ago, but no PR was put in:

<details><summary>expand me:</summary>
<p>

```diff
diff --git a/Marlin/Configuration.h b/Marlin/Configuration.h
index c59bbba..d49261e 100644
--- a/Marlin/Configuration.h
+++ b/Marlin/Configuration.h
@@ -68,7 +68,7 @@
 
 // Choose the name from boards.h that matches your setup
 #ifndef MOTHERBOARD
-  #define MOTHERBOARD BOARD_RAMPS_14_EFB
+  #define MOTHERBOARD BOARD_BTT_OCTOPUS_MAX_EZ_V1_0
 #endif
 
 /**
@@ -79,7 +79,7 @@
  *
  * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
  */
-#define SERIAL_PORT 0
+#define SERIAL_PORT -1
 
 /**
  * Serial Port Baud Rate
@@ -92,7 +92,7 @@
  *
  * :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000]
  */
-#define BAUDRATE 250000
+#define BAUDRATE 115200
 
 //#define BAUD_RATE_GCODE     // Enable G-code M575 to set the baud rate
 
@@ -101,7 +101,7 @@
  * Currently Ethernet (-2) is only supported on Teensy 4.1 boards.
  * :[-2, -1, 0, 1, 2, 3, 4, 5, 6, 7]
  */
-//#define SERIAL_PORT_2 -1
+#define SERIAL_PORT_2 7
 //#define BAUDRATE_2 250000   // :[2400, 9600, 19200, 38400, 57600, 115200, 250000, 500000, 1000000] Enable to override BAUDRATE
 
 /**
diff --git a/Marlin/src/HAL/STM32/HAL.h b/Marlin/src/HAL/STM32/HAL.h
index 013390f..e81c13d 100644
--- a/Marlin/src/HAL/STM32/HAL.h
+++ b/Marlin/src/HAL/STM32/HAL.h
@@ -75,7 +75,7 @@
   #elif SERIAL_PORT_2 == -1
     #define MYSERIAL2 MSerialUSB
   #else
-    #error "SERIAL_PORT_2 must be from 1 to 6, or -1 for Native USB."
+    //#error "SERIAL_PORT_2 must be from 1 to 6, or -1 for Native USB."
   #endif
 #endif
 
diff --git a/Marlin/src/HAL/STM32/MarlinSerial.cpp b/Marlin/src/HAL/STM32/MarlinSerial.cpp
index 37a8f40..ee2ebe5 100644
--- a/Marlin/src/HAL/STM32/MarlinSerial.cpp
+++ b/Marlin/src/HAL/STM32/MarlinSerial.cpp
@@ -43,6 +43,11 @@
   MSerialT MSerial ## ser_num (true, USART ## ser_num, &_rx_complete_irq_ ## ser_num); \
   void _rx_complete_irq_ ## ser_num (serial_t * obj) { MSerial ## ser_num ._rx_complete_irq(obj); }
 
+#define DECLARE_SERIAL_PORT_UART(ser_num) \
+  void _rx_complete_irq_ ## ser_num (serial_t * obj); \
+  MSerialT MSerial ## ser_num (true, UART ## ser_num, &_rx_complete_irq_ ## ser_num); \
+  void _rx_complete_irq_ ## ser_num (serial_t * obj) { MSerial ## ser_num ._rx_complete_irq(obj); }
+
 #if USING_HW_SERIAL1
   DECLARE_SERIAL_PORT(1)
 #endif
@@ -62,7 +67,8 @@
   DECLARE_SERIAL_PORT(6)
 #endif
 #if USING_HW_SERIAL7
-  DECLARE_SERIAL_PORT(7)
+  //DECLARE_SERIAL_PORT(7)
+  DECLARE_SERIAL_PORT_UART(7)
 #endif
 #if USING_HW_SERIAL8
   DECLARE_SERIAL_PORT(8)
```

</p>
</details>

gianguardo, 2023-07-11:
OK thank you

thisiskeithb, 2023-07-11:
Remember to disable `#define WIFISUPPORT` in your config since the WiFi module runs separate firmware & acts like a serial host.

gianguardo, 2023-07-11:
Following the steps I get these errors:
'MYSERIAL2' was not declared in this scope; did you mean 'MYSERIAL1'?
template argument 2 is invalid
request for member 'write' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}
request for member 'print' in 'multiSerial', which is of non-class type 'SerialOutputT' {aka 'int'}

I'm writing you the configuration to be sure I haven't made a mistake in modifying
Configuration.h

#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BTT_OCTOPUS_MAX_EZ_V1_0
#endif
#define SERIAL_PORT -1
#define BAUDRATE 115200
#define SERIAL_PORT_2 7

STM32/HAL.h
//#error "SERIAL_PORT_2 must be from 1 to 6, or -1 for Native USB."

MarlinSerial.cpp

#define DECLARE_SERIAL_PORT(ser_num) \
  void _rx_complete_irq_ ## ser_num (serial_t * obj); \
  MSerialT MSerial ## ser_num (true, USART ## ser_num, &_rx_complete_irq_ ## ser_num); \
  void _rx_complete_irq_ ## ser_num (serial_t * obj) { MSerial ## ser_num ._rx_complete_irq(obj); }

#define DECLARE_SERIAL_PORT_UART(ser_num) \
  void _rx_complete_irq_ ## ser_num (serial_t * obj); \
  MSerialT MSerial ## ser_num (true, UART ## ser_num, &_rx_complete_irq_ ## ser_num); \
  void _rx_complete_irq_ ## ser_num (serial_t * obj) { MSerial ## ser_num ._rx_complete_irq(obj); }

#if USING_HW_SERIAL7
  //DECLARE_SERIAL_PORT(7)
  DECLARE_SERIAL_PORT_UART(7)
#endif

gianguardo, 2023-07-11:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/12016800/Configuration.zip)
[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/12016801/Configuration_adv.zip)

thisiskeithb, 2023-07-11:
The first error is:

```prolog
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:52,
                 from Marlin/src/HAL/STM32/MarlinSerial.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2052:6: error: #error "TEMP_SENSOR_0 MAX thermocouple requires TEMP_0_CS_PIN."
 2052 |     #error "TEMP_SENSOR_0 MAX thermocouple requires TEMP_0_CS_PIN."
      |      ^~~~~
```

Did you modify any other files, like `pins_BTT_OCTOPUS_MAX_EZ.h`?

As for the patch not working, it *was* working, but now I'm also seeing the same compile errors.

ellensp, 2023-07-11:
disable #define TMC_USE_SW_SPI   its using hardware SPI and is the same SPI port your using for sdcard and your MAX's

ellensp, 2023-07-12:
This Compiles. I have no way to test it

Updated Config files

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/12022348/Configuration.files.zip)

Changes to STM32/HAL.h

```DIFF
diff --git a/Marlin/src/HAL/STM32/HAL.h b/Marlin/src/HAL/STM32/HAL.h
index 013390fcaf..37a718b08d 100644
--- a/Marlin/src/HAL/STM32/HAL.h
+++ b/Marlin/src/HAL/STM32/HAL.h
@@ -57,61 +57,61 @@
 #define _MSERIAL(X) MSerial##X
 #define MSERIAL(X) _MSERIAL(X)
 
-#if WITHIN(SERIAL_PORT, 1, 6)
+#if WITHIN(SERIAL_PORT, 1, 9)
   #define MYSERIAL1 MSERIAL(SERIAL_PORT)
 #elif !defined(USBCON)
-  #error "SERIAL_PORT must be from 1 to 6."
+  #error "SERIAL_PORT must be from 1 to 9."
 #elif SERIAL_PORT == -1
   #define MYSERIAL1 MSerialUSB
 #else
-  #error "SERIAL_PORT must be from 1 to 6, or -1 for Native USB."
+  #error "SERIAL_PORT must be from 1 to 9, or -1 for Native USB."
 #endif
 
 #ifdef SERIAL_PORT_2
-  #if WITHIN(SERIAL_PORT_2, 1, 6)
+  #if WITHIN(SERIAL_PORT_2, 1, 9)
     #define MYSERIAL2 MSERIAL(SERIAL_PORT_2)
   #elif !defined(USBCON)
-    #error "SERIAL_PORT_2 must be from 1 to 6."
+    #error "SERIAL_PORT_2 must be from 1 to 9."
   #elif SERIAL_PORT_2 == -1
     #define MYSERIAL2 MSerialUSB
   #else
-    #error "SERIAL_PORT_2 must be from 1 to 6, or -1 for Native USB."
+    #error "SERIAL_PORT_2 must be from 1 to 9, or -1 for Native USB."
   #endif
 #endif
 
 #ifdef SERIAL_PORT_3
-  #if WITHIN(SERIAL_PORT_3, 1, 6)
+  #if WITHIN(SERIAL_PORT_3, 1, 9)
     #define MYSERIAL3 MSERIAL(SERIAL_PORT_3)
   #elif !defined(USBCON)
-    #error "SERIAL_PORT_3 must be from 1 to 6."
+    #error "SERIAL_PORT_3 must be from 1 to 9."
   #elif SERIAL_PORT_3 == -1
     #define MYSERIAL3 MSerialUSB
   #else
-    #error "SERIAL_PORT_3 must be from 1 to 6, or -1 for Native USB."
+    #error "SERIAL_PORT_3 must be from 1 to 9, or -1 for Native USB."
   #endif
 #endif
 
 #ifdef MMU2_SERIAL_PORT
-  #if WITHIN(MMU2_SERIAL_PORT, 1, 6)
+  #if WITHIN(MMU2_SERIAL_PORT, 1, 9)
     #define MMU2_SERIAL MSERIAL(MMU2_SERIAL_PORT)
   #elif !defined(USBCON)
-    #error "MMU2_SERIAL_PORT must be from 1 to 6."
+    #error "MMU2_SERIAL_PORT must be from 1 to 9."
   #elif MMU2_SERIAL_PORT == -1
     #define MMU2_SERIAL MSerialUSB
   #else
-    #error "MMU2_SERIAL_PORT must be from 1 to 6, or -1 for Native USB."
+    #error "MMU2_SERIAL_PORT must be from 1 to 9, or -1 for Native USB."
   #endif
 #endif
 
 #ifdef LCD_SERIAL_PORT
-  #if WITHIN(LCD_SERIAL_PORT, 1, 6)
+  #if WITHIN(LCD_SERIAL_PORT, 1, 9)
     #define LCD_SERIAL MSERIAL(LCD_SERIAL_PORT)
   #elif !defined(USBCON)
-    #error "LCD_SERIAL_PORT must be from 1 to 6."
+    #error "LCD_SERIAL_PORT must be from 1 to 9."
   #elif LCD_SERIAL_PORT == -1
     #define LCD_SERIAL MSerialUSB
   #else
-    #error "LCD_SERIAL_PORT must be from 1 to 6, or -1 for Native USB."
+    #error "LCD_SERIAL_PORT must be from 1 to 9, or -1 for Native USB."
   #endif
   #if HAS_DGUS_LCD
     #define LCD_SERIAL_TX_BUFFER_FREE() LCD_SERIAL.availableForWrite()
```

Changes to MarlinSerial.cpp
```DIFF
diff --git a/Marlin/src/HAL/STM32/MarlinSerial.cpp b/Marlin/src/HAL/STM32/MarlinSerial.cpp
index 37a8f40fd0..797954d559 100644
--- a/Marlin/src/HAL/STM32/MarlinSerial.cpp
+++ b/Marlin/src/HAL/STM32/MarlinSerial.cpp
@@ -37,6 +37,9 @@
 #ifndef USART5
   #define USART5 UART5
 #endif
+#ifndef USART7
+  #define USART7 UART7
+#endif
 
 #define DECLARE_SERIAL_PORT(ser_num) \
   void _rx_complete_irq_ ## ser_num (serial_t * obj); \

```

Changes to pins_BTT_OCTOPUS_MAX_EZ.h

```DIFF
diff --git a/Marlin/src/pins/stm32h7/pins_BTT_OCTOPUS_MAX_EZ.h b/Marlin/src/pins/stm32h7/pins_BTT_OCTOPUS_MAX_EZ.h
index bd230da6fc..374975b44e 100644
--- a/Marlin/src/pins/stm32h7/pins_BTT_OCTOPUS_MAX_EZ.h
+++ b/Marlin/src/pins/stm32h7/pins_BTT_OCTOPUS_MAX_EZ.h
@@ -183,6 +183,17 @@
   #define E0_CS_PIN                         PG10
 #endif
 
+#if TEMP_SENSOR_0 == -5
+#define TEMP_0_CS_PIN PG9
+#define TEMP_0_SCK_PIN PE12
+#define TEMP_0_MISO_PIN PE13
+#define TEMP_0_MOSI_PIN PE14
+//#define SOFTWARE_SPI
+//#define FORCE_SOFT_SPI
+#else
+#define TEMP_0_PIN PB0
+#endif
+
 #define E1_STEP_PIN                         PG15  // MOTOR 6
 #define E1_DIR_PIN                          PB3
 #define E1_ENABLE_PIN                       PD5
@@ -374,8 +385,8 @@
 #if SD_CONNECTION_IS(ONBOARD)
   #ifndef SD_DETECT_STATE
     #define SD_DETECT_STATE HIGH
-  #elif SD_DETECT_STATE == LOW
-    #error "BOARD_BTT_OCTOPUS_MAX_EZ onboard SD requires SD_DETECT_STATE set to HIGH."
+  //#elif SD_DETECT_STATE == LOW
+  //  #error "BOARD_BTT_OCTOPUS_MAX_EZ onboard SD requires SD_DETECT_STATE set to HIGH."
   #endif
   #define SDSS                              PB12
   #define SD_SS_PIN                         SDSS
@@ -464,4 +475,4 @@
   #define ESP_WIFI_MODULE_RESET_PIN         PB2
   #define ESP_WIFI_MODULE_GPIO0_PIN         PG0
   #define ESP_WIFI_MODULE_GPIO4_PIN         PF15
-#endif
+#endif
\ No newline at end of file
```

ellensp, 2023-07-12:
I have added a PR to allow uarts 1-9

ellensp, 2023-07-12:
Since a PR has been created closing this FR

closed 2023-07-12 by ellensp