MarlinFirmware/Marlin #26076: [BUG] Input Shaping Zeta not saved

ThomasToka, 2023-07-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Sending

`stepper.set_shaping_damping_ratio(X_AXIS, x_zeta);`

Is not saved automatically.
I have to 

`settings.save();`

Sending

`stepper.set_shaping_frequency(X_AXIS, x_frequency);`

Is saved directly and survives a reboot.

### Bug Timeline

Don't know.

### Expected behavior

Zeta also saved to eeprom.

### Actual behavior

Zeta not saved to eeprom

### Steps to Reproduce

Send zeta in script or with M593 X D0.15.
Reboot printer and check zeta with M593.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

E3S1PRO

### Electronics

Srock

### Add-ons

Non3

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Configs not needed. Should be the same on every printer with IS enabled.

tombrazier, 2023-07-14:
Neither function should cause settings to be saved to EEPROM. I cannot see how `stepper.set_shaping_frequency(X_AXIS, x_frequency);` would survive a reboot.

ThomasToka, 2023-07-14:
@tombrazier i just verified it again and i think i am wrong and you are right. both are not saved. i had this false bug laying around for weeks and what i think is that i had a settings.save(); after all subfunctions in my settings save case. i then sometime removed this, but had already a settings.save(); in the zeta case function.

sorry for the wind.

naturally i close this.

user error.

closed 2023-07-14 by ThomasToka