MarlinFirmware/Marlin #26089: [BUG] Extruder stops moving at the same line of G-Code

enesdogan00, 2023-07-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have an Ender3V2 and I installed a BMG extruder that required me to increase the E-steps from 93 to 415. After that it worked find for about a week but now extrusion stops at the same line of G-code of my print. And the extruder doesn't work until rebooting the printer or sending M18 and M17. Other steppers works just fine. I have checked my cables and they seem to be fine, when I keep the wires at the same position and reboot the printer extruder works normally. I have checked the stepper temperature and it didn't feel too warm. I tried printing without filament and extruder still stops. I have tried printing with octoprint, pronterface and SD print but didn't make any difference.

### Bug Timeline

01/07/2023

### Expected behavior

I expect my printer to keep the extruder moving.

### Actual behavior

Extrusion stops at the same line of print.

### Steps to Reproduce

1. Get the marlin firmware and compile it with the given commit number and config files.
2. Upload the firmware to an Ender3V2 with a Creality 4.2.2 board.
3. Print the given gcode.

### Version of Marlin Firmware

244de2458ac6fd48f3031d0179ca1bcad6808583

### Printer model

Ender3V2

### Electronics

Creality 4.2.2

### Add-ons

BMG extruder

### Bed Leveling

No Bed Leveling

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/12078916/Marlin.zip)
On this gcode the extrusion stops at around line 1000.
[catt.gcode.zip](https://github.com/MarlinFirmware/Marlin/files/12078930/catt.gcode.zip)

IlyaFarm, 2023-07-23:
Maybe your driver is overheating?
Cause with/without filament, moves are continue, extruder spinning, driver send impulses of current and heating up...
But am I not sure.

closed 2023-10-02 by github-actions[bot]

thinkyhead, 2023-10-07:
Check to see if your extruder motor has powered off. If that is the case you will be able to turn the extruder by hand.