MarlinFirmware/Marlin #26092: [BUG] Edit Screen value drift for negative decimals

lukasradek, 2023-07-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When opening an edit screen from menu item with negative decimal value, the value increases by one at the last decimal place.

I can be nicely seen when Editing Mesh in menu.
Before opening.
[image]
Open EDIT_ITEM
[image]
No change, just confirm.
[image]

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Simulator Windows

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2023-07-19:
I think you should really add some more details. like the `Configuration.h` files. what screen are you using?

thinkyhead, 2023-10-06:
Float imprecision is a bit of a headache. So we have methods to round float values for display, which are supposed to round towards zero. If these methods are forgotten in the display implementation then a value like -0.060000001 will get improperly "rounded" to 0.007.

closed 2024-01-19 by github-actions[bot]