MarlinFirmware/Marlin #26093: SKR mini e3 v3 compiling issue

VicenteAdauy, 2023-07-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When I try to compile the latest version of marlin the following issue appears:

In file included from Marlin/src/HAL/STM32/../../inc/../HAL/../HAL/STM32/../shared/Marduino.h:89,
                 from Marlin/src/HAL/STM32/../../inc/../HAL/../HAL/STM32/HAL.h:28,
                 from Marlin/src/HAL/STM32/../../inc/../HAL/HAL.h:30,
                 from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:33,
                 from Marlin/src/HAL/STM32/eeprom_if_iic.cpp:32:
Marlin/src/HAL/STM32/../../inc/../core/mstring.h: In member function 'MString<SIZE, SAFE>& MString<SIZE, SAFE>::set()':
Marlin/src/HAL/STM32/../../inc/../HAL/../HAL/STM32/../shared/progmem.h:42:72: error: expected primary-expression before ')' token
   42 | #define F(str) (reinterpret_cast<const __FlashStringHelper *>(PSTR(str)))
      |                                                                        ^
Marlin/src/HAL/STM32/../../inc/../core/mstring.h:114:93: note: in expansion of macro 'F'
  114 |   MString& set()                          { str[0] = '\0';                            debug(F("clear"));   return *this; }
      |                                                                                             ^
Marlin/src/HAL/STM32/../../inc/../core/mstring.h: In member function 'MString<SIZE, SAFE>& MString<SIZE, SAFE>::set(char*)':
Marlin/src/HAL/STM32/../../inc/../HAL/../HAL/STM32/../shared/progmem.h:42:72: error: expected primary-expression before ')' token
   42 | #define F(str) (reinterpret_cast<const __FlashStringHelper *>(PSTR(str)))
      |                                                                        ^
Marlin/src/HAL/STM32/../../inc/../core/mstring.h:115:93: note: in expansion of macro 'F'
  115 |   MString& set(char *s)                   { strncpy(str, s, SIZE);                    debug(F("string"));  return *this; }
      |                                                                                             ^
Marlin/src/HAL/STM32/../../inc/../core/mstring.h: In member function 'MString<SIZE, SAFE>& MString<SIZE, SAFE>::set_P(const char*)':
Marlin/src/HAL/STM32/../../inc/../HAL/../HAL/STM32/../shared/progmem.h:42:72: error: expected primary-expression before ')' token
   42 | #define F(str) (reinterpret_cast<const __FlashStringHelper *>(PSTR(str)))
      |                                                                        ^
Marlin/src/HAL/STM32/../../inc/../core/mstring.h:117:93: note: in expansion of macro 'F'
  117 |   MString& set_P(PGM_P const s)           { strncpy_P(str, s, SIZE);                  debug(F("pstring")); return *this; }
      |                                                                                             ^
Marlin/src/HAL/STM32/../../inc/../core/mstring.h: In member function 'MString<SIZE, SAFE>& MString<SIZE, SAFE>::set(const bool&)':
Marlin/src/HAL/STM32/../../inc/../HAL/../HAL/STM32/../shared/progmem.h:42:72: error: expected primary-expression before ')' token
   42 | #define F(str) (reinterpret_cast<const __FlashStringHelper *>(PSTR(str)))
      |                                                                        ^
Marlin/src/HAL/STM32/../../inc/../core/mstring.h:119:60: note: in expansion of macro 'F'
  119 |   MString& set(const bool &b)             { return set(b ? F("true") : F("false")); }
      |                                                            ^
Marlin/src/HAL/STM32/../../inc/../HAL/../HAL/STM32/../shared/progmem.h:42:72: error: expected primary-expression before ')' token
   42 | #define F(str) (reinterpret_cast<const __FlashStringHelper *>(PSTR(str)))
      |                                                                        ^
Marlin/src/HAL/STM32/../../inc/../core/mstring.h:119:72: note: in expansion of macro 'F'
  119 |   MString& set(const bool &b)             { return set(b ? F("true") : F("false")); }
      |                                                                        ^
compilation terminated due to -fmax-errors=5.
*** [.pio/build/STM32G0B1RE_btt/src/src/HAL/STM32/eeprom_if_iic.cpp.o] Error 1

I have checked multiple times and I don't see any evident issue in the config files. Any help would be greatly appreciated.

### Bug Timeline

It works fine with standard config files, but doesn't work with my edit.

### Expected behavior

For the compiling to work and give me a .bin file

### Actual behavior

An error occurs, without enough explanation on what to do to solve it.

### Steps to Reproduce

1.[use latest marlin buggfix]
2.[modify config files according to my printer]
3.[hit compile]

### Version of Marlin Firmware

2.1.2.1

### Printer model

Creality CR-10 S5

### Electronics

btt skr mini e3 v3 board

### Add-ons

biqu h2v2 extruder, keenovo style bed heater.

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config files.zip](https://github.com/MarlinFirmware/Marlin/files/12099852/config.files.zip)

thisiskeithb, 2023-07-19:
> Version of Marlin Firmware
> 2.1.2.1

Your configs are for a newer version:
> `#define CONFIGURATION_H_VERSION 02010300`

You'll need to downgrade your config or upgrade your copy of Marlin.

---

Compiling with the latest `bugfix-2.1.x`, I get the following error:

```prolog
error: static assertion failed: TRAMMING_POINT_XY point 0 is not reachable with the default NOZZLE_TO_PROBE offset and PROBING_MARGIN.
```

You need to adjust your probing margin or tramming points under the `ASSISTED_TRAMMING` menu.

closed 2023-07-19 by thisiskeithb

VicenteAdauy, 2023-07-19:
Thank you for the quick response. I have fixed the tramming point issue and it doesn't change the outcome. As for the config files version, I verified it, on the standard files, the version is 02010300; if I change the version to 02010201 (2.1.2.1 is the newest release as far as I know) then the version incompatibility occurs. Even with those changes, the outcome is the same: same error message.
[Configuration_adv.h.zip](https://github.com/MarlinFirmware/Marlin/files/12100309/Configuration_adv.h.zip)
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/12100310/Configuration.h.zip)
[image]

thisiskeithb, 2023-07-19:
>  As for the config files version, I verified it, on the standard files, the version is 02010300; if I change the version to 02010201 (2.1.2.1 is the newest release as far as I know) then the version incompatibility occurs.

You'll need to downgrade your config or upgrade your copy of Marlin. You can't just change that version number and move configs between versions.

Using your new configs & reverting the version change (since these are 2.1.3 configs) & temporarily disabling `SHOW_CUSTOM_BOOTSCREEN` & `CUSTOM_STATUS_SCREEN_IMAGE` (since I don't have those), I'm now getting these errors:

```prolog
error: 'PREHEAT_3_LABEL' was not declared in this scope; did you mean 'PREHEAT_5_LABEL'?

error: 'PREHEAT_4_LABEL' was not declared in this scope; did you mean 'PREHEAT_5_LABEL'?
```

Your config is missing both `PREHEAT_3_*` & `PREHEAT_4_*` options, but has `#define PREHEAT_5_*` options, so I renamed those to `PREHEAT_3_*`.

Your config now compiles correctly:

```prolog
Indexing .pio/build/STM32G0B1RE_btt/libFrameworkArduino.a
Linking .pio/build/STM32G0B1RE_btt/firmware.elf
Checking size .pio/build/STM32G0B1RE_btt/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.0% (used 24996 bytes from 147456 bytes)
Flash: [=====     ]  49.4% (used 258780 bytes from 524288 bytes)
Building .pio/build/STM32G0B1RE_btt/firmware.bin
========================================= [SUCCESS] Took 55.16 seconds =========================================

Environment      Status    Duration
---------------  --------  ------------
STM32G0B1RE_btt  SUCCESS   00:00:55.164
========================================== 1 succeeded in 00:00:55.164 ==========================================
 *  Terminal will be reused by tasks, press any key to close it.
 *
```
---

Delete the hidden `.pio` folder at the root of your Marlin project and try compiling again.

If that doesn't work:

1. Delete the hidden `.platformio` folder from your User directory:
 
    - Windows: `%USERPROFILE%\.platformio\`
    - macOS, Unix, Unix-like: `~/.platformio/`

2. Re-install PlatformIO.

--- 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

VicenteAdauy, 2023-07-21:
Thank you so much for the help, I followed your steps and I finally could compile the firmware!