MarlinFirmware/Marlin #26100: Unable to ABL (G29) when FT Motion is enabled.

ahmetcemturan, 2023-07-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Auto Bed Leveling fails when FT Motion is enabled (M493 S1).  It works fine when I send M493 S0 before G29.
Also when I try to send e.g. M493 S1 A25 B25, I get: Wrong mode for [A] frequency. Wrong mode for [B] frequency.
I dont get that essage when I set S to S11, 12 etc..

### Bug Timeline

This is the bugfix of a few days ago.

### Expected behavior

I expect it to be able to ABl without disabling FT motion and I want to be able to change the FT Motion settings..

### Actual behavior

Doesnt work

### Version of Marlin Firmware

2.1.2 bugfix as of 17th July

### Printer model

Prusa Clone

### Electronics

SKR3 EZ

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/12134078/Marlin.zip)

narno2202, 2023-07-21:
@ahmetcemturan , i look at your config files , the config seems ok. My probe is a BLTouch and my motherboard a MKS Monster8 V1 with same stepper drivers. I test with the last Marlin bugfix-2.1.x (my actual firmware is from my FT_MOTION develop branch)

1.  With or without `FT_MOTION` , `G29` works fine (S0, S1 ...) for Bilinear bed levelling
2. With M493 S1, you can't set frequencies as only `FT_MOTION` is enabled without shaping. You need a `ft_motion_mode > 1`
3.  In your config file, `FT_MOTION` linear advance is disabled, so no K factor is applied as standard lin_advance is bypassed with `ft_motion_mode >0`

I have a question. When i submit the #26083 PR, screenshots are from Marlin Simulator : same emulated LCD as your REPRAP_DISCOUNT_SMART_CONTROLLER. When i test the menu, the simulator crashes if i change Motion Mode before sending any serial command (i.e. works if i home then change FT Motion Mode). Do you notice the same behavior with your printer?

ahmetcemturan, 2023-07-22:
@narno2202 ,  
to 1:  echo:  M913 T0 E50
echo:; Driver stepping mode:
echo:  M569 S1 X Y Z
echo:  M569 S1 T0 E
echo:; Fixed-Time Motion:
echo:  M493 S1 A45.00 B25.50 D0 F0.00 H0.00 P1 K0.00
echo:; Hotend Idle Timeout:
  M86 B0 E0 S900 T180
echo:; Linear Advance:
echo:  M900 K0.06
echo:; Filament load/unload:
echo:  M603 L0.00 U100.00 ; (mm)
//  HOMING HERE
>>> G29
SENDING:G29
//action:notification Probing Point 1/25
//action:notification Probing Failed
Error:Probing Failed
[ERROR] Error:Probing Failed

//action:notification Hornet3D SKR3 Ready.
>>> M493 S0
SENDING:M493 S0
Fixed-Time Motion disabled.
Linear Advance enabled. Gain: 0.00100
>>> G29
SENDING:G29
//action:notification Probing Point 1/25
//action:notification Probing Failed
Error:Probing Failed   //BECAUSE I DIDNT HOME (AGAIN)
[ERROR] Error:Probing Failed

//action:notification Hornet3D SKR3 Ready.
echo:busy: processing
echo:busy: processing
echo:busy: processing
>>> G29
SENDING:G29
//action:notification Probing Point 1/25
//action:notification Probing Point 2/25
.
.
.
.
.
.
echo:busy: processing
//action:notification Probing Point 23/25
//action:notification Probing Point 24/25
echo:busy: processing
//action:notification Probing Point 25/25
//action:notification Hornet3D SKR3 Ready.
Bilinear Leveling Grid:
      0      1      2      3      4
 0 -0.705 -0.813 -0.858 -0.815 -0.698
 1 -0.294 -0.313 -0.286 -0.226 -0.289
 2 -0.128 -0.046 +0.042 +0.122 +0.014
 3 -0.030 +0.019 +0.161 +0.297 +0.064
 4 +0.041 +0.100 +0.109 +0.245 +0.071
Subdivided with CATMULL ROM Leveling Grid:

to 2:   don't understand: So what frequency does it correct when I don't enter any?  (there would be the FW settings though)
Can You explain a little bit more?  

to 3: Yes in FW it was like that but I enabled it via M493 later and had the problem of overcompensation by about x10

ahmetcemturan, 2023-07-22:
To the question:
The printer doesnt "crash" in the meaning that it needs a reset but makes a rather unpleasant sound (as if hit an object) and shifts about 10mm and continues.
Pausing, changing Motion mode, Homing X and Y works fine.

ahmetcemturan, 2023-07-22:
Note: On my CoreXY with Robin Nano S 1.3,  G29 works fine with FT motion enabled

ahmetcemturan, 2023-07-22:
Done calibrating and printing a Benchy:
Settings:  M493 S14 A36.50 B40.00 D0 F0.00 H0.00 P1 K0.00
Print via USB - Pronterface
Observation:  Extreme slowdown on tiny features (Screenshot attached, GCODE attached)
I will Retry printing from SD card,
[3DBenchy_0.25mm_PLA_MK3S_25m.zip](https://github.com/MarlinFirmware/Marlin/files/12135964/3DBenchy_0.25mm_PLA_MK3S_25m.zip)

narno2202, 2023-07-22:
@ahmetcemturan

1.  M569 is useless as stealthchop is enabled in your `Configuration_ADV` file, try disable `HYBRID_TRESHOLD` in your  `Configuration_ADV` file as some people have problem with this setting (not enabled in my config)
2. To set a X and/or Y frequency with `M493` in` FT_MOTION` you need a valid shaping mode : S10 and above. On the printer start, default values are loaded if there is no user defined frequency stored.
3.  In your Gcode, `FT_MOTION` linear advance K factor is set to 0. M900 is only for lin-advance in Marlin's standard motion mode. You have to calibrate your FT_MOTION K factor (algorithms are not the same) before using it.
4. Thank's for your answer, i will try to fix the menu
5. For a simple explanation on object rotation with a corexy : [https://marlinfw.org/docs/gcode/M593.html](url)

I will have a look at your new post

narno2202, 2023-07-22:
@ahmetcemturan , you choose the more time consuming calculation algorithm. As you see my start gcode is very simple
```
;TYPE:Custom
G28 ; home all axes
M190 S80 ; wait for bed temp
M109 S225 ; wait for nozzle temp
G29 ; Bed Levelling
G1 X0 Y20 Z0.2 F3000 ; get ready to prime
G92 E0 ; reset extrusion distance
G1 X100 E20 F600 ; prime nozzle
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0
```
No more idea to solve your G29 failure as i use the same leveling process without any failure, only the board differs.

ahmetcemturan, 2023-07-22:
Update: Printing from SD is working as expected.  Same GCODE, no slowing down on small details..

ahmetcemturan, 2023-07-22:
[image]
S14 seems to work best for me and the SKR3 is apowerful board (I hope)

jamespearson04, 2023-08-19:
I'm experiencing the same levelling issue with UBL G29. Probe retracts but Z movement continues until emergency stop triggers. 
M493 S0 also "resolves" the issue. Homing the Z using the same probe works at least.

narno2202, 2023-08-19:
@jamespearson04 . I test with current bugfix and latest `FT_MOTION` code from PR #26074, UBL works with `FT_MOTION` enabled. My printer is a standard motion XYZE printer. If you have a coreXY printer, for now this kinematic is buggy in `FT_MOTION`. In this case, your help will be appreciated if you could send feedback from this PR with the last fix proposals.

jamespearson04, 2023-08-19:
I'll test that fork out now, also just running a standard cartesian printer, so shouldn't have any issues.

thinkyhead, 2023-09-23:
We can move discussion of this issue over to #26074. This is one of the issues we should be aiming to repair in that PR. I did disable FT Motion for the duration of `G28` in that PR but perhaps not yet for the duration of `G29`. Anyway, please test with the branch associated with that PR and we'll get this fixed over there ASAP.

closed 2023-09-23 by thinkyhead