MarlinFirmware/Marlin #26108: [BUG] (Extruder won't move MARLIN 2.0/ Latest MARLIN2.1.2.1 gives an error message )

IlyaFarm, 2023-07-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

**HARDWARE: BTT Octopus V1.1; Drivers: TMC2208
FIRMWARE: Marlin 2.0.x** 
After configuring and compiling all motora are running exept of Extruder motor.
Recomendations from: #15926; #25566; #24882; did not help.
Cause of it I decided to try Marlin 2.1.2.1 latest release from official web site.
After configuring files Configuration.h; Configuration_adv.h; pins_BTT_OCTOPUS_V1_common.h
when I try to compile this version of Marlin V2.1.2.1 it gives me an error message:
**error: 'class Temperature' has no member named 'extrude_min_temp'**
Task **#25324** did not help too cause when from there I goes to 
**#25252 > ProUI general updates #25176 > https://github.com/mriscoc/Special_Configurations >** I'll get **Original Configs** and after configuring this files to my printer configuration I copy them onto me Marlin V2.1.2.1 directory and try to compile it, 
but PlatformOI ask me to update Marlin to the version specified in the files, 
which I don't know where to get.

So I don't know what to do, cause most recent version, on which I put my hopes did not work at all...

### Bug Timeline

old bug

### Expected behavior

UP

### Actual behavior

UP

### Version of Marlin Firmware

2.0.x/2.1.2.1

### Printer model

TEVO TARANTULA

### Electronics

BOARD: BTT Octopus V1.1; DRIVERS: TMC2208

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

All changes can be found by the symbols                     //!!!

ellensp, 2023-07-22:
you have disabled PREVENT_COLD_EXTRUSION so extrude_min_temp is not setup

Which is breaking 
```CPP
    #if HAS_HOTEND
      if (parser.seenval('H'))
        mat.hotend_temp = constrain(parser.value_int(), thermalManager.extrude_min_temp, thermalManager.hotend_max_target(0));
    #endif

```

IlyaFarm, 2023-07-22:
When I try to compile it as Marlin V2.1.2.1 **regardless of** on/off  Cold Extrusion, PlatformIO gives me message:
**error: 'class Temperature' has no member named 'extrude_min_temp'** 
I tried this and that....

IlyaFarm, 2023-07-22:
> you have disabled PREVENT_COLD_EXTRUSION so extrude_min_temp is not setup
....
When I try to compile it as Marlin V2.1.2.1 regardless of on/off Cold Extrusion, PlatformIO gives me message:
error: 'class Temperature' has no member named 'extrude_min_temp'
I tried this and that....

ellensp, 2023-07-22:
```DIFF
diff --git a/Marlin/src/gcode/lcd/M145.cpp b/Marlin/src/gcode/lcd/M145.cpp
index 942d20afd2..4940d3eac6 100644
--- a/Marlin/src/gcode/lcd/M145.cpp
+++ b/Marlin/src/gcode/lcd/M145.cpp
@@ -47,7 +47,7 @@ void GcodeSuite::M145() {
     preheat_t &mat = ui.material_preset[material];
     #if HAS_HOTEND
       if (parser.seenval('H'))
-        mat.hotend_temp = constrain(parser.value_int(), thermalManager.extrude_min_temp, thermalManager.hotend_max_target(0));
+        mat.hotend_temp = constrain(parser.value_int(), TERN(PREVENT_COLD_EXTRUSION, thermalManager.extrude_min_temp, EXTRUDE_MINTEMP), thermalManager.hotend_max_target(0));
     #endif
     #if HAS_HEATED_BED
       if (parser.seenval('B'))

```

ellensp, 2023-07-22:
get a fresh copy of marlin 2.1.2.1  as it only happens when PREVENT_COLD_EXTRUSION is undefined  and the above patch fixes it.

IlyaFarm, 2023-07-22:
I wrote above that I took the latest version from official site

ellensp, 2023-07-22:
so did I...   just now...

ellensp, 2023-07-22:
Change that line and compile, if it errors post full error message

ellensp, 2023-07-22:
"  #define MIN_SOFTWARE_ENDSTOP_Z false  //!!! По умолчанию стоит True поэтому Сколько бы ни ставили в Z offset, ниже нуля не двинется !!!!" is invalid.
comment it out or un comment it, not true or false

IlyaFarm, 2023-07-22:
In my case string "#define MIN_SOFTWARE_ENDSTOP_Z false", allow to axis Z move below zero, cause I ned it for my printing way.

ellensp, 2023-07-22:
no, you just comment it out  "//#define MIN_SOFTWARE_ENDSTOP_Z"

IlyaFarm, 2023-07-22:
OK, thank you, you was right about strings:
#define PREVENT_COLD_EXTRUSION
#define EXTRUDE_MINTEMP 170
After uncomment this strings, PlatformIO are compille firmware succeful.

IlyaFarm, 2023-07-22:
Just now flash firmware, still not work properly, all motors are rotate, exept extruder motor at 185C.

ellensp, 2023-07-22:
Are you sure your using correct stepper driver and plug?

IlyaFarm, 2023-07-22:
Yep I'am sure, cause if I use wrong socket for driver, Marlin wrote to me TMC driver error, and I waste two days to solve this problem throw soldering jumpers on drivers...  below is a link to a picture of my board
[image]

ellensp, 2023-07-22:
You have E0 cable plugged into z2 socket

IlyaFarm, 2023-07-22:
Why? 
they go in this order: X, Y, Z, Z2, E0, E1, E2, E3, E4
Or not?

ellensp, 2023-07-22:
Look at the image I posted
order is X, Y, Z, Z, Z2, E0, E1, E2, E3, E4

Two Z's fr running two z off 1 driver

IlyaFarm, 2023-07-22:
So what for Z2 on a board?

And yep it's working, but so loud, with a hiss.

IlyaFarm, 2023-07-22:
Thanks a lot again. You're good.

ellensp, 2023-07-22:
You need to adjust the driver currents and check correct rsense value for each axis

Eg for E0

  #define E0_CURRENT      800
  #define E0_MICROSTEPS    16
  #define E0_RSENSE         0.11
  
In Configuration_adv.h

IlyaFarm, 2023-07-22:
It's it's already done in Hardware and in Firmware.
Drivers before work fine on different board "Rumba" and was adjusted.

ellensp, 2023-07-23:
The variable resistors are ignored as soon as the UART is working. The Drivers are reconfigured via UART using the settings  in Configuration_adv.h

IlyaFarm, 2023-07-23:
Everything fine now, I uncomment string
#define STEALTHCHOP_E
and extruder works silently, as it be before.

IlyaFarm, 2023-07-23:
If you interesting, you can try use
#define DEFAULT_MINSEGMENTTIME        6592
instead of 20000, I think the best value, cause when I was use Mega2560 based boards, this value give me better print quolity in combination with
#define MIN_STEPS_PER_SEGMENT 1
#define BLOCK_BUFFER_SIZE 32
and if was work fine with Mega 2560 I think it will be not bad with 32bit ARM MCU's , but here probably other optimal value for
#define DEFAULT_MINSEGMENTTIME

closed 2023-07-24 by IlyaFarm