MarlinFirmware/Marlin #26109: [BUG] (bug summary)Some LCD pages are not in the language file

liu2-3zhi, 2023-07-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Some LCD pages are not in the language file, such as "Configuration > Debug > Endstop Test". I cannot translate these pages.
In addition, there are many language files that are incompletely translated or missing.

### Version of Marlin Firmware

02010300

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

liu2-3zhi, 2023-07-22:
There is no ZIP file for Configuration.h and Configuration_adv.h. Because it has nothing to do with Configuration.h and Configuration_adv.h.

ellensp, 2023-07-22:
Except what type of lcd ae you referring to, a Config is always needed to answer questions like this one.
 And what pages?

liu2-3zhi, 2023-07-22:
"Configuration > Debug > Endstop Test"

ellensp, 2023-07-22:
thats is one page, you said pages...

liu2-3zhi, 2023-07-22:
I only found that this one page is not in the language file. All that remains is to regenerate the language files and font data for languages other than English.

ellensp, 2023-07-22:
We rely on people who speak the language to provide correct translations 
English is the default file and should be used as a template

In Marlin/src/lcd/language/language_en.h is
`LSTR MSG_ENDSTOP_TEST                   = _UxGT("Endstop Test");`

You just add this line to your appropriate language file with the correct translation.
 
No font rebuilding or stuff like that is required.

We even provide tools to list what strings are missing from each language...
buildroot/share/scripts/findMissingTranslations.sh 
though this seems a little broken right now

from another post you made I see that your identify as speaking "Chinese"

Running findMissingTranslations.sh  zh_CN shows

249 out of 781 LCD strings need translation

<details><summary>List of missing translations</summary>
<p>

```
Missing strings for language_zh_CN.h (Simplified Chinese):
  LSTR MSG_HIGH                           = _UxGT("HIGH");
  LSTR MSG_LOW                            = _UxGT("LOW");
  LSTR MSG_ERROR                          = _UxGT("Error");
  LSTR MSG_MEDIA_INIT_FAIL                = MEDIA_TYPE_EN _UxGT(" Init Fail");
  LSTR MSG_MEDIA_SORT                     = _UxGT("Sort ") MEDIA_TYPE_EN;
  LSTR MSG_ENDSTOP_TEST                   = _UxGT("Endstop Test");
  LSTR MSG_Z_PROBE                        = _UxGT("Z Probe");
  LSTR MSG_AUTO_HOME_A                    = _UxGT("Home @");
  LSTR MSG_Z_AFTER_HOME                   = _UxGT("Z After Homing");
  LSTR MSG_FILAMENT_SET                   = _UxGT("Filament Settings");
  LSTR MSG_FILAMENT_MAN                   = _UxGT("Filament Management");
  LSTR MSG_MANUAL_LEVELING                = _UxGT("Manual Leveling");
  LSTR MSG_TRAM_FL                        = _UxGT("Front Left");
  LSTR MSG_TRAM_FR                        = _UxGT("Front Right");
  LSTR MSG_TRAM_C                         = _UxGT("Center");
  LSTR MSG_TRAM_BL                        = _UxGT("Back Left");
  LSTR MSG_TRAM_BR                        = _UxGT("Back Right");
  LSTR MSG_MANUAL_MESH                    = _UxGT("Manual Mesh");
  LSTR MSG_AUTO_MESH                      = _UxGT("Auto Build Mesh");
  LSTR MSG_ITERATION                      = _UxGT("G34 Iteration: %i");
  LSTR MSG_DECREASING_ACCURACY            = _UxGT("Accuracy Decreasing!");
  LSTR MSG_ACCURACY_ACHIEVED              = _UxGT("Accuracy Achieved");
  LSTR MSG_HOME_OFFSET_X                  = _UxGT("Home Offset X");
  LSTR MSG_HOME_OFFSET_Y                  = _UxGT("Home Offset Y");
  LSTR MSG_HOME_OFFSET_Z                  = _UxGT("Home Offset Z");
  LSTR MSG_ERR_M428_TOO_FAR               = _UxGT("Err: Too far!");
  LSTR MSG_TRAMMING_WIZARD                = _UxGT("Tramming Wizard");
  LSTR MSG_SELECT_ORIGIN                  = _UxGT("Select Origin");
  LSTR MSG_LAST_VALUE_SP                  = _UxGT("Last value ");
  LSTR MSG_PREHEAT_HOTEND                 = _UxGT("Preheat Hotend");
  LSTR MSG_LASER_TOGGLE                   = _UxGT("Toggle Laser");
  LSTR MSG_LASER_EVAC_TOGGLE              = _UxGT("Toggle Blower");
  LSTR MSG_LASER_ASSIST_TOGGLE            = _UxGT("Air Assist");
  LSTR MSG_LASER_PULSE_MS                 = _UxGT("Test Pulse ms");
  LSTR MSG_LASER_FIRE_PULSE               = _UxGT("Fire Pulse");
  LSTR MSG_FLOWMETER_FAULT                = _UxGT("Coolant Flow Fault");
  LSTR MSG_SPINDLE_TOGGLE                 = _UxGT("Toggle Spindle");
  LSTR MSG_SPINDLE_EVAC_TOGGLE            = _UxGT("Toggle Vacuum");
  LSTR MSG_SPINDLE_FORWARD                = _UxGT("Spindle Forward");
  LSTR MSG_BED_TRAMMING_MANUAL            = _UxGT("Manual Tramming");
  LSTR MSG_BED_TRAMMING_RAISE             = _UxGT("Adjust bed until the probe triggers.");
  LSTR MSG_BED_TRAMMING_IN_RANGE          = _UxGT("Corners within tolerance. Bed trammed.");
  LSTR MSG_BED_TRAMMING_GOOD_POINTS       = _UxGT("Good Points: ");
  LSTR MSG_BED_TRAMMING_LAST_Z            = _UxGT("Last Z: ");
  LSTR MSG_MESH_VIEWER                    = _UxGT("Mesh Viewer");
[... 184 lines cut ...]
  LSTR MSG_FTM_DFREQ_K_N                  = _UxGT("@ Dyn. Freq.");
  LSTR MSG_FTDI_HEATER_TIMEOUT            = _UxGT("Idle timeout, temperature decreased. Press Okay to reheat and again to resume.");
  LSTR MSG_REHEATDONE                     = _UxGT("Reheat Done");
  LSTR MSG_PROBE_WIZARD                   = _UxGT("Z Probe Wizard");
  LSTR MSG_PROBE_WIZARD_PROBING           = _UxGT("Probing Z Reference");
  LSTR MSG_PROBE_WIZARD_MOVING            = _UxGT("Moving to Probing Pos");
  LSTR MSG_XATC                           = _UxGT("X-Twist Wizard");
  LSTR MSG_XATC_DONE                      = _UxGT("X-Twist Wizard Done!");
  LSTR MSG_XATC_UPDATE_Z_OFFSET           = _UxGT("Update Probe Z-Offset to ");
  LSTR MSG_SOUND                          = _UxGT("Sound");
  LSTR MSG_TOP_LEFT                       = _UxGT("Top Left");
  LSTR MSG_BOTTOM_LEFT                    = _UxGT("Bottom Left");
  LSTR MSG_TOP_RIGHT                      = _UxGT("Top Right");
  LSTR MSG_BOTTOM_RIGHT                   = _UxGT("Bottom Right");
  LSTR MSG_CALIBRATION_COMPLETED          = _UxGT("Calibration Completed");
  LSTR MSG_CALIBRATION_FAILED             = _UxGT("Calibration Failed");
  LSTR MSG_DRIVER_BACKWARD                = _UxGT(" driver backward");
  LSTR MSG_SD_CARD                        = _UxGT("SD Card");
  LSTR MSG_USB_DISK                       = _UxGT("USB Disk");
  LSTR MSG_HOST_SHUTDOWN                  = _UxGT("Host Shutdown");
```
<p>
</details>

Please provide translations for these strings

ellensp, 2023-07-22:
Here is a CSV file with all string. Please fill out any that are Missing 
[language_zh_CN.csv](https://github.com/MarlinFirmware/Marlin/files/12136949/language_zh_CN.csv)

liu2-3zhi, 2023-07-22:
It's too late for me in China now. I'll try to finish it tomorrow. But now, I need to sleep.

liu2-3zhi, 2023-07-23:
Why are some texts I still can't find in the language files?
[image]

liu2-3zhi, 2023-07-23:
例如"open"、“TRIGGEREND‘’

EvilGremlin, 2023-07-23:
You can add missing strings if you like (entry in english required too, as it's template for everything else)
But serial output must stay all-english.

liu2-3zhi, 2023-07-23:
But they don't seem to exist in the language files, even the English language files can't find them. They seem to be embedded directly into the code.

EvilGremlin, 2023-07-23:
Yes, because no one needed to change tham to lang strings yet. You'll have to do that.

liu2-3zhi, 2023-07-23:
Error strings for language_zh_CN.h (Simplified Chinese)(Can I say there is no translation at all?):
```csharp
  LSTR MSG_ADVANCE_K                      = _UxGT("线性进给常数 K");
  LSTR MSG_ADVANCE_K_E                    = _UxGT("线性进给常数 K *");
```
Missing strings for language_zh_CN.h (Simplified Chinese)（language_zh_CN.h (Simplified Chinese) is missing strings (that I can find and understand in the code or can see in my LCD):）:
```csharp

  LSTR MSG_HIGH                           = _UxGT("高");
  LSTR MSG_LOW                            = _UxGT("低");
  LSTR MSG_ERROR                          = _UxGT("错误");
  LSTR MSG_MEDIA_INIT_FAIL                = MEDIA_TYPE_EN _UxGT("内存卡\USB设备初始化失败");
  LSTR MSG_MEDIA_SORT                     = _UxGT("种类 ") MEDIA_TYPE_EN;
  LSTR MSG_ENDSTOP_TEST                   = _UxGT("传感器测试");
  LSTR MSG_Z_PROBE                        = _UxGT("Z 探针");
  LSTR MSG_AUTO_HOME_A                    = _UxGT("回@轴原位");
  LSTR MSG_Z_AFTER_HOME                   = _UxGT("Z轴已归位");
  LSTR MSG_FILAMENT_SET                   = _UxGT("耗材设置");
  LSTR MSG_FILAMENT_MAN                   = _UxGT("耗材管理");
  LSTR MSG_MANUAL_LEVELING                = _UxGT("手动调平");
  LSTR MSG_TRAM_FL                        = _UxGT("左上角");
  LSTR MSG_TRAM_FR                        = _UxGT("右上角");
  LSTR MSG_TRAM_C                         = _UxGT("中心");
  LSTR MSG_TRAM_BL                        = _UxGT("左下角");
  LSTR MSG_TRAM_BR                        = _UxGT("右下角");
  LSTR MSG_MANUAL_MESH                    = _UxGT("手动建立调平网格");
  LSTR MSG_AUTO_MESH                      = _UxGT("自动建立调平网格");
  LSTR MSG_ITERATION                      = _UxGT("G34 重复反馈中: %i");
  LSTR MSG_DECREASING_ACCURACY            = _UxGT("精确度在降低！");
  LSTR MSG_ACCURACY_ACHIEVED              = _UxGT("达到目标精度");
  LSTR MSG_HOME_OFFSET_X                  = _UxGT("原点在X轴上的偏移");
  LSTR MSG_HOME_OFFSET_Y                  = _UxGT("原点在Y轴上的偏移");
  LSTR MSG_HOME_OFFSET_Z                  = _UxGT("原点在Z轴上的偏移");
  LSTR MSG_ERR_M428_TOO_FAR               = _UxGT("错误：偏移距离过大！");
  LSTR MSG_TRAMMING_WIZARD                = _UxGT("调平热床床角向导");
  LSTR MSG_MOVE_50MM                      = _UxGT("移动 50mm");
  LSTR MSG_FILAMENT_EN                    = _UxGT("*耗材情况");
  LSTR MSG_BABYSTEP_N                     = _UxGT("短距离移动 @");
  LSTR MSG_COLORS_RED                     = _UxGT("红色");
  LSTR MSG_COLORS_GREEN                   = _UxGT("绿色");
  LSTR MSG_COLORS_BLUE                    = _UxGT("蓝色");
  LSTR MSG_COLORS_WHITE                   = _UxGT("白色");
  LSTR MSG_UI_LANGUAGE                    = _UxGT("UI界面语言");
  LSTR MSG_SOUND_ENABLE                   = _UxGT("声音已开启");
  LSTR MSG_LOCKSCREEN                     = _UxGT("锁定屏幕");
  LSTR MSG_PROBE_WIZARD                   = _UxGT("探针Z轴偏移量自测量");
  LSTR MSG_PROBE_WIZARD_PROBING           = _UxGT("正在测量探针的Z轴偏移量");
  LSTR MSG_PROBE_WIZARD_MOVING            = _UxGT("移动到探测位置");
  LSTR MSG_XATC                           = _UxGT("X轴扭曲补偿向导");
  LSTR MSG_XATC_DONE                      = _UxGT("X轴扭曲补偿向导完成!");
  LSTR MSG_XATC_UPDATE_Z_OFFSET           = _UxGT("探针在Z轴上的偏移将被更新为");
  LSTR MSG_SOUND                          = _UxGT("声音");
  LSTR MSG_TOP_LEFT                       = _UxGT("左上角");
  LSTR MSG_BOTTOM_LEFT                    = _UxGT("左下角");
  LSTR MSG_TOP_RIGHT                      = _UxGT("右上角");
  LSTR MSG_BOTTOM_RIGHT                   = _UxGT("右下角");
  LSTR MSG_CALIBRATION_COMPLETED          = _UxGT("校准完成");
  LSTR MSG_CALIBRATION_FAILED             = _UxGT("校准失败");
  LSTR MSG_DRIVER_BACKWARD                = _UxGT("驱动器是反向的");
  LSTR MSG_SD_CARD                        = _UxGT("SD 卡);
  LSTR MSG_USB_DISK                       = _UxGT("USB 设备");
  LSTR MSG_HOST_SHUTDOWN                  = _UxGT("关闭宿主机");
```
What none of the language files have:

Wich on the "Configuration > Debug > Endstop Test" page:

Whcih In the "Marlin\Marlin\src\lcd\menu\menu_configuration.cpp" file:
```csharp
#define STOP_ITEM(A,I) _STOP_ITEM(STRINGIFY(A) STRINGIFY(I) " " TERN(A##_HOME_TO_MAX, "Max", "Min"), A##I##_ENDSTOP)
#define FIL_ITEM(N) PSTRING_ITEM_N_P(N-1, MSG_FILAMENT_EN, (READ(FIL_RUNOUT##N##_PIN) != FIL_RUNOUT##N##_STATE) ? PSTR("PRESENT") : PSTR("out"), SS_FULL);
```
Translated:
```csharp
#define STOP_ITEM(A,I) _STOP_ITEM(STRINGIFY(A) STRINGIFY(I) " " TERN(A##_HOME_TO_MAX, "最大值", "最小值"), A##I##_ENDSTOP)
#define FIL_ITEM(N) PSTRING_ITEM_N_P(N-1, MSG_FILAMENT_EN, (READ(FIL_RUNOUT##N##_PIN) != FIL_RUNOUT##N##_STATE) ? PSTR("存在") : PSTR("不存在"), SS_FULL);
```

Whcih In the "Marlin\Marlin\src\core\language.h" file:
  ```csharp
#define STR_ENDSTOP_HIT                     "TRIGGERED"
#define STR_ENDSTOP_OPEN                    "open"
```

Translated:

```csharp
#define STR_ENDSTOP_HIT                     "触发"
#define STR_ENDSTOP_OPEN                    "未触发"
```

liu2-3zhi, 2023-07-23:
The "Configuration > Debug > Endstop Test" page is sometimes useful for troubleshooting. I wish they could be added to the language files.

IlyaFarm, 2023-07-23:
There is a possibility that your display only supports Japanese and Latin characters, like mine 
REPRAP_DISCOUNT_SMART_CONTROLLER.
I would suggest checking this way: make a translation in Latin by different words not like in lang.h file and if the Latin symbols is displayed, 
then the problem is in the firmware of your display.
But I cannot guarantee it.

liu2-3zhi, 2023-07-23:
I'm very sorry, I'm not good at English.
May be my expression is wrong.
The LCD can display normally, but the display is in English. Because they are not in the lang.h file. I need to modify the code, I hope to add the corresponding text to the lang.h file

IlyaFarm, 2023-07-23:
No if your display firmware not do not support chinese language, it's too difficult to change firmware on it, and not fact that will help.
Long time ago I try by my self translate to my regional language my
REPRAP_DISCOUNT_SMART_CONTROLLER
and I couldn't did for this reason.

liu2-3zhi, 2023-07-23:
My display can display Chinese!

IlyaFarm, 2023-07-23:
Understood, so then you can translate.

liu2-3zhi, 2023-07-23:
But some LCD pages are not in the language file. Not even the English language file.

ellensp, 2023-07-26:
Updated language file with info provided

[language_zh_CN.zip](https://github.com/MarlinFirmware/Marlin/files/12171144/language_zh_CN.zip)

Not yet extended to replace all English