MarlinFirmware/Marlin #26115: [BUG] BTT Octopus Pro 1.1 (STM32H723ZE) failing to compile with `USB_FLASH_DRIVE_SUPPORT` enabled

smikutsky, 2023-07-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When compiling latest bugfix-2.1.x for BTT Octopus Pro 1.1 with STM32H723ZE, compilation fails if `USB_FLASH_DRIVE_SUPPORT` is enabled.

By disabling `USB_FLASH_DRIVE_SUPPORT`, I am able to get a successful build.

### Bug Timeline

It started yesterday when configuring for new BTT Octopus Pro 1.1

### Expected behavior

Build success:
[image]

### Actual behavior

Build failed:
[image]
This is actually all of the errors, they all fit

### Steps to Reproduce

1. Load config
2. Try to build

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

N/A

### Electronics

BTT Octopus Pro 1.1

### Add-ons

N/A

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[OctopUBS 1_1 config.zip](https://github.com/MarlinFirmware/Marlin/files/12137242/OctopUBS.1_1.config.zip)

thisiskeithb, 2023-07-22:
We're missing the `_USB` environments for this board / STM32H7.

smikutsky, 2023-07-28:
Given that I was unable to decifer the compile error in the screenshot, adding a check or warning that STM32H7 USB is not (yet) supported would be helpful to users less familiar with the platform

Hman32, 2023-08-15:
@smikutsky  there's no warning because normally you should choose the right environment to being able to compile and currently there is no "_USB" environment for the H7.

For ex. the STM32F4 for the Octopus you have 3 =/= environment for compiling with the USB.

[image]

ellensp, 2023-08-15:
"_USB environments were for flashing a board connected over USB" no not quite...  It allows sdcard to be used as a usb drive on the computer  

But there are two different _USB  the other is to enable usb host support on board that have it.

This is why https://github.com/MarlinFirmware/Marlin/pull/26131  was added and merged, to differentiate them

smikutsky, 2023-08-15:
The changes in #26131 are just for a few environments, SKR2 has USB drive support on its base environments:

[image]

While standardizing the naming is good, I don't foresee all 200+ environments coming into line quickly. A beginner-friendly error message could still help with ease of use as the USB host feature becomes (hopefully) more popular.

Steppge, 2024-03-15:
will there be support for usb flash drive at some point?

djpallax, 2025-04-27:
Any solution for this? Or there is someone working on this now?