MarlinFirmware/Marlin #26116: [BUG] Input shaping causes delay in laser actuation

Moeschus, 2023-07-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I use Marlin to control a laser engraver (using inline mode) and when I activate input shaping this causes delays in laser actuation. Therefore I cannot use input shaping, although it would be a great feature for laser engravers, because powerful laser modules are quite heavy and have to be moved around quite quickly. This generates quite a bit of vibration that can be nicely cancelled by input shaping.

The delay becomes clearly visible when I engrave text or images in bidirectional mode. This means that the laser moves across the whole text or image in lines in alternating directions (left to right, then right to left and so on). A delay causes all features to get a shifted "twin". This behavior is only present when input shaping is active (see image below). The shift between the letters in this picture is about 4 mm and the text was engraved with 200 mm/s. This means that there must be a time delay of the laser actuation of approximately 10 ms. This behavior is only present when input shaping is active.

[image]

### Bug Timeline

Present in 2.1.2.1 and bugfix-2.1.x. No other versions tested.

### Expected behavior

No laser actuation delays caused by input shaping

### Actual behavior

laser actuation delays caused by input shaping

### Steps to Reproduce

- activate input shaping
- engrave any picture or text in bidirectional fill mode

### Version of Marlin Firmware

2.1.2.1

### Printer model

custom laser engraver

### Electronics

SKR mini E3 V3.0

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/12137273/Configuration.zip)
[Test_gcode.zip](https://github.com/MarlinFirmware/Marlin/files/12137274/Test_gcode.zip)

narno2202, 2023-07-23:
@Moeschus, do you test input-shaping before 20 July Babystep commit ? If yes and works fine, try diabling Babystepping in `Configuration_ADV.h` as babystep isr has changed.

Moeschus, 2023-07-23:
Thank you for the quick answer @narno2202! I downloaded my version of bugfix-2.1.x on 19th of July. So it is before that commit. Disabling Babystepping in Configuration_ADV.h doesn't change the behavior.

narno2202, 2023-07-25:
Answering to @danbrianwhite about `FT_MOTION` and talking about buffers, i remember your issue with `INPUT_SHAPING`. Have you try to change `SHAPING_MAX_STEPRATE` ? With your config, the calculated value is 40 000 and `shaping_echoes is 836` (the buffer size), it's perhaps not the good size for X axis as Y axis is OK on your pics. Enable in `Configuration_ADV.h`  `SHAPING_MAX_STEPRATE` to a value over 40 000 and test it.

narno2202, 2023-07-25:
@Moeschus, how have you determined your shaping frequencies ? They are perhaps too low : when they increase, the buffer size diminishes as if you were decreasing SHAPING_MAX_STEPRATE (best results in your pics).
Now it's vacation time. I'll hope you find a solution.

Moeschus, 2023-07-26:
For now it was just a rough estimation based on wavelength of engraved oscillations. Frequencies on this system are really low because of the heavy weight of the toolhead.

Thank you for your help and have a nice vacation!

I won't close the issue yet, because I still think this is a bug. IS should not introduce any shift at all between motion and laser actuation imho.

JonBr306, 2024-05-04:
> IS should not introduce any shift at all between motion and laser actuation imho.

It looks like the laser feature thinks that the position includes half steps that have not yet been fully processed. So the laser is turning on slightly too soon, for example when scanning from left to right the image is shifted to the left and vice versa, the equivalent occurs with up and down.

JonBr306, 2024-05-05:
After having a chat with @tombrazier. In principle it is possible to use the `ShapingQueue` to synchronise x, y and the laser so that  even with input shaping the laser will switch on at the right intervals. This will take more RAM but it appears @Moeschus has a 32 bit main board so this probably will not be an issue. 

I think we might be able to do this and submit a PR, we would need someone with a laser cutter to help test it.

thinkyhead, 2026-10-08:
Thanks for the detailed tests. They line up with the code. Inline laser power follows the planner's unshaped timeline, while ZV input shaping replays part of every X/Y step about half a resonance period later. So the carriage lags the laser by roughly factor2 x delay (about 7–8 ms at your 25 Hz / 0.15 settings), and bidirectional raster turns that into the ghost offset. Fixing it properly means routing laser power changes through the shaping queue. That's too invasive for the 2.1.3 bugfix release, so we're tracking it for a later version. Until then, disabling shaping for raster jobs (M593 F0) or using a higher shaping frequency will reduce or remove the offset.