MarlinFirmware/Marlin #26124: [BUG] G33 does not take into account the probe offset (G30 does)

Ari-SSO, 2023-07-24:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Better explain with an example:

**1)** Dual extruder with a single block, and a touch probe attached directly to the hotend 0, so is one block with 2 nozzles, same Y and Z for both, but nozzle 0 is X=-9 and Nozzle 1 is X=+9 relative to the center of the block:
```
      #define HOTEND_OFFSET_X { 0.0, 18.00 } // (mm) relative X-offset for each nozzle
      #define HOTEND_OFFSET_Y { 0.0, 0.00 }  // (mm) relative Y-offset for each nozzle
      #define HOTEND_OFFSET_Z { 0.0, 0.00 } 
```
So, as Z-Probe is attached to nozzle 0:

```
       #define FIX_MOUNTED_PROBE
       #define NOZZLE_TO_PROBE_OFFSET { -9, 0, 0 }
```
**2)** G28 go home and after thats the block is perfectly centered to the bed X,Y plane, so nozzle 0 is X=-9 and nozzle 1 is X=+9 relative to the center of the bed.

**3)**
  **3.1)** With G30, the block of the nozzle move X+9, and after goes to the point to do the measurement. Example, G30 X11, touch the bed in its X=2   _(2=10+ (-9))_.
 **3.2)** With G33, apparently the position of the probe is not taken into account, so, for example, G33 P10 results in the circular path made by the probe touching the bed, forming a circle whose center is **X=-9, Y=0** instead of **X=0, Y=0**.

I have not tried it yet but I think that the generation of the mesh with G29 P1 does not take into account the position of the probe either.

### Bug Timeline

I don't know exactly, but with versions older than 2.1.x this happened to. I remeber figth with this issue some time ago

### Expected behavior

G33 should take into account the position of the probe and make a concentric circle to the bed

### Actual behavior

G33, apparently, daesn't take into account the porbe's offsets (X at least)

### Steps to Reproduce

1) Run G33 
2) Look where it touches the bed. This is easier to see when you get to the edge (-X) and (+X) of the bed.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

HE3D K280, DELTA, 140mm of printable radius, 600mm height

### Electronics

BTT SKR 1.3, BTT TMC2209, BTT TFT35 v3

### Add-ons

Z-Probe 

### Bed Leveling

None

### Your Slicer

None

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_Files.zip](https://github.com/MarlinFirmware/Marlin/files/12140358/Configuration_Files.zip)

Ari-SSO, 2023-07-28:
I'm not sure if this is an error or not, since perhaps to do the self-calibration it is necessary that the extruder block is as centered as possible, if that is the case then it would be good if it does not take into account the position of the extruder probe.

In that case, so the problem would be in the G29 command, which is not coherent with the G30, since the G30 takes into account the position of the probe and locates it so that it touches the bed in the designated X,Y position. While the G29 does not build the mesh taking into account the offset of the probe, so the M92 command must be executed before executing the G29, to center the probe.

Could someone please confirm what I have mentioned here?

Thanks!

Kraligor, 2023-08-25:
Don't know the exact reasoning, but the following code is in `G30.cpp`:
```
// Disable leveling so the planner won't mess with us
    TERN_(HAS_LEVELING, set_bed_leveling_enabled(false));
```

So it seems to be very much intended that way.

closed 2023-11-05 by github-actions[bot]