MarlinFirmware/Marlin #26126: [BUG] firmware ignores BLTouch signal on BTT SKR mini E3 V3.0

wb1016, 2023-07-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

no matter BLTouch Signal pin is attatched to z-stop port or attatched to dedicated sensor port, firmware ignores sensor output.
transmitting signal into BLTouch is fine. it deploys its probe when i press 'deploy probe' in the LCD menu.

### Bug Timeline

a month ago

### Expected behavior

z-homing using probe should complete when BLTouch touches the bed

### Actual behavior

z-homing does not stop even if BLTouch probe touches the bed and retracts, results slamming printhead into printbed

### Steps to Reproduce

1. download marlin 2.1.2.1 or nightly source
2. download matching config files
3. uncomment following entries
```
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define BLTOUCH
#define AUTO_BED_LEVELING_BILINEAR
#define Z_SAFE_HOMING
```
4. build and flash
5. connect BLTouch output signal pin to z-stop port or dedicated port

### Version of Marlin Firmware

2.1.2.1 and bugfix-2.1.x

### Printer model

Ender-3

### Electronics

SKR mini E3 V3.0

### Add-ons

BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/12169912/config.zip)
[a simmilar issue](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/593) report exists.

ellensp, 2023-07-26:
If you have a 3d touch (ie a clone) disable #define BLTOUCH_FORCE_SW_MODE

ellensp, 2023-07-26:
staeps to reproduce are not consistant

"#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN"

"connect BLTouch output signal pin to z-stop port or dedicated port"  NO this will only work on z-stop port, not dedicated port

ellensp, 2023-07-26:
Just tested today's bugfix on an SKR Mini E3 V3 with a genuine BLTouch connected to Z STOP (Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN) and it works fine.

closed 2023-07-26 by ellensp