MarlinFirmware/Marlin #26129: [BUG] Bugfix broken on Elegoo Neptune 2S (ZNP v1.3)

diogo580, 2023-07-27:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Marlin flashes but doesn't boot
This issue has been created to fix a M600 problem

### Expected behavior

Marlin boot screen

### Actual behavior

Complete white screen

### Version of Marlin Firmware

Bugfix

### Printer model

Elegoo Neptune 2S

### Electronics

Stock

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/12181467/Configuration.zip)

thisiskeithb, 2023-07-27:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/26090

closed 2023-07-27 by thisiskeithb

diogo580, 2023-07-27:
@thisiskeithb why did you close this? the bugfix doesn't work

thisiskeithb, 2023-07-27:
> why did you close this?

Because it’s a duplicate of https://github.com/MarlinFirmware/Marlin/issues/26090.

> the bugfix doesn't work

No one has submitted a patch, so this has not been fixed.

diogo580, 2023-07-27:
> No one has submitted a patch, so this has not been fixed.

but i literally can't use the patch? since it doesn't boot into marlin?

thisiskeithb, 2023-07-27:
> but i literally can't use the patch? since it doesn't boot into marlin?

There is no patch, so you have to wait until someone fixes the bug.

To clarify, someone opened bug report/issue (https://github.com/MarlinFirmware/Marlin/issues/26090) before you for the same reported problem(s), so your issue has been closed as a duplicate.

We don’t need multiple issues opened reporting the same thing.

Follow https://github.com/MarlinFirmware/Marlin/issues/26090 for updates. Hopefully someone will come along and submit a fix.

diogo580, 2023-07-27:
> There is no patch, so you have to wait until someone fixes the bug.

so, let me get this straight, i opened this issue because i couldn't use marlin 2.1.x, and it's a duplicate of a M600 trouble?

phobos79, 2023-07-27:
Hello, 2.1.x boot well on neptune 2, so for sure is not a bug but some error by you, M600 is one of the problems of that release, but marlin boot