MarlinFirmware/Marlin #26133: [BUG] Touch stop working after finishing a print.

gutierrezvalentino, 2023-07-29:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Hi, im trying to compile a firmware for my 3D printer (it have a MKS Robin nano 1.3 and a TFT35)
Im ussing classic marlin UI and enabled "#define MKS_ROBIN_TFT35".
When the printer power on, it start with a touch calibration screen, everything work, the touch is responsive and everything, but when a print ends, the touch stops working until the printers is turn of and on again.
ALSO: the toutch screen keeps working during the print, i can pause, resume, its still responsive, but when the print ends, it stops working until reset.

Is there a way to fix this? Or maybe i forgot to uncomment something.

### Bug Timeline

today

### Expected behavior

Expecting the touch screen to keep working after finishing a print

### Actual behavior

The touch screen stops working after finishing a print

### Steps to Reproduce

1)Start a print job from SD
2)Let it finish
3)Touch stops working

### Version of Marlin Firmware

LTS 2.1.1

### Printer model

Wanhao D12 300

### Electronics

MKS robin nano v1.3

### Add-ons

TFT35 mks

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2023-07-29:
is `TOUCH_SCREEN` enabled in configuration.h?

gutierrezvalentino, 2023-07-29:
Yes, im currently tring the last marlin bugfix version to see if the issue continues

classicrocker883, 2023-07-29:
could it be possible that the screen has a timeout feature? meaning after a certain time has passed the LCD will turn off. well if not turn off in your case, maybe there is part of the code saying "timer reached, disable connection to display". basically it's asleep and "touch" isn't being registered to wake back up.

so perhaps can you instead of making a print, like leave it on the main menu - walk away and see if the screen still works or not. like after 5 or 10 minutes or so. say it stops working after 10 minutes. that would confirm my suspicion.

closed 2023-10-08 by github-actions[bot]