MarlinFirmware/Marlin #26134: [BUG] SKR Mini E3 V3 with BTT TFT50 V3.0 "no printer attached" in Touchscreen mode and black screen for Marlin Mode

clembo1, 2023-07-29:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Board: SKR Mini E3 V3.0  (STM32G0B1RE_btt)
LCD:  BTT TFT50 V3.0 
Used Visual Studio Code with the Marlin Firmware Auto Build

Marlin firmware 2.1.2.1 and current bugfix branch when compiled for SKR Mini E3 V3.0 display a black screen on BTT TFT50 V3.0 in Marlin Mode and "No Printer Attached" on BTT Touch Screen mode. 

On Marlin firmware 2.1.2.0 (bugfix branch) both Marlin Mode and Touchscreen mode work on the LCD. To keep things simple, I will use 2.1.2.0 (Bugfix-2.1.x branch from Oct 8 ,2022) and 2.1.2.1 release or current bugfix as the firmware references.

At a minimum, a simple compile using the exact same settings for Serial and LCD on 2.1.2.1 in configuration.h should work for both modes on the LCD just like the settings on 2.1.2.0 that work. 

To ensure I didn't change some 1-off configuration, I took the stock configuration from the Creality\Ender3 for Marlin 2.1.2.0 and compiled the firmware with the configuration.h and Configuration_adv.h as-is with no further modifications, installed on the SKR Mini E3V3...both modes worked.  I repeated the exact same process with the 2.1.2.1 Release and current bugfix branch...I get the black screen or "no printer attached"

Unless there is some obscure mistake I am making...I believe something is broken in the newer Marlin firmware versions in relation to the serial communication when using a SKR Mini E3 V3.0 and a BTT TFT 50 V3.0.

I have also:

1. Checked the correct Baud Rate of 115200 on touchscreen mode on the TFT 50.
2. Confirmed the Pin files match. stm32g0 pins from 2.1.2.0 and 2.1.2.1 to see if there was any difference...files are identical. Only difference was the addition of validation.h. Additionally I checked the boards file and found that SKR Mini E3 3.0's category was changed, but that shouldn't cause the serial connections to fail.
3. checked the stm32g0.ini file on both firmware versions...seems to be the same, only thing different is on
[env:STM32G0B1RE_btt_xfer]
pre:buildroot/share/scripts/upload.py
upload_protocol = custom
4. Checked and rechecked all documentation from BTT and from internet sources to see if there were any changes to serial ports and investigated to see if anyone else is having this issue. I found various things but nothing specific. Only thing I saw that was interesting was that someone on a forum mentioned TH3D firmware as using serial port 1...but I have no way to validate, and my issue is with official Marlin build.
5. Updated the firmware for the BTT TFT50 V3.0 and re-tested.  same issue occurs. 

### Bug Timeline

Anything past 2.1.2.0  (Bugfix-2.1.x branch from Oct 8 ,2022)

### Expected behavior

Both Marlin Mode and Touchscreen mode work on current 2.1.2.x+ version of Marlin firmware

### Actual behavior

After uploading firmware 2.1.2.1 or current bugfix branch There is a black screen in Marlin Mode. In Touchscreen mode, "no printer attached" even after waiting several minutes.

### Steps to Reproduce

Download a copy of 2.1.2.1 release or current bugfix > Use File path: Marlin-2.1.2.1_Release\Configurations-release-2.1.2.1\config\examples\Creality\Ender-3\BigTreeTech SKR Mini E3 3.0...and copy files over to Marlin directory

Edit Configuration.h (note: it should already be set correctly from the example)
#define SERIAL_PORT 2
#define BAUDRATE 115200
#define SERIAL_PORT_2 -1
#define CR10_STOCKDISPLAY
--note: no other display is enabled (including the infamous reprap one.)

Compile firmware > copy firmware to microsd card > insert card into board > turn on machine

get a blank screen or "printer not found"

Turn machine off...because nothing happens. Remove microsd > reinsert microsd into pc > filename of firmware.bin is changed to firmware.cur ... so I know the firmware is loading.

At this point I revert back to the 2.1.2.0 bug fix build > everything works again. Just to make sure it wasn't some fluke, I also recompiled the 2.1.2.0 Marlin firmware using fresh Configuration.h and Configuration_adv.h and reinstalled...it works every time.

### Version of Marlin Firmware

2.1.2.1 release, current bugfix  (note: currently using 2.1.2.0 bugfix due to issue)

### Printer model

Creality Ender 3V2 with SKR Mini E3V3.0

### Electronics

Board: SKR Mini E3 V3.0  (STM32G0B1RE_btt) LCD:  BTT TFT50 V3.0 

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2023-07-29:
so one firmware version displays black, the other (older) says no printer attached? 
have you tried any version of firmware like from BTT which works so you can get a benchmark?

it could be something wrong with the board if no other firmware works. otherwise perhaps the LCD is wired differently or pinouts are different (between versions).

ellensp, 2023-07-30:
Just tested this

Using Marlin bugfix 2.1.x 

Config files from https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3/BigTreeTech%20SKR%20Mini%20E3%203.0  With no changes

On a On a TFT 3.5 V3.0  (identical screen just smaller) 

It works as expected

(bad photos, but you can see it s working)

Marlin mode (just using usb power, so getting expected tmc errors)
[image]

Touch Mode  (is connected fine)

[image]

Check your cabling

[image]
Reset cable must be on the left

[image]
Use EXP3 plug for CR10_STOCKDISPLAY

thisiskeithb, 2023-07-30:
No issues here on current `bugfix-2.1.x` with two different SKR Mini E3 V3 boards (STM32G0B1RE & STM32G0B0RE) using a BTT TFT35 E3 V3.

closed 2023-07-30 by thisiskeithb

clembo1, 2023-07-30:
Hi All,

Thank you for the input.  A couple clarifications:

1. Can you test with a BTT TFT50  V3.0?  Every reply indicates testing using the TFT35.  I understand the boards "should" be the same, but that doesn't mean everything is 100% the same.  If that's not possible, itsok; I appreciate the effort.
2. I reconfirmed cables are plugged into the correct ports
3. both modes work perfectly in 2.1.2.0 bugfix from Oct 22.  Wiring "should be" correct otherwise, I would have issues one way or another on that build.  That being said, I have two different serial cables and get the same result using either cable in Marlin mode when using 2.1.2.1 release or current bugfix.  I only have the supplied usb cable that came with the LCD, but can confirm it is plugged into the correct port on the lcd and has the correct orientation on the SKR Mini E3 V3.

classicrocker883, 2023-07-30:
sometimes all it takes is fresh files. like once for some reason one of the libraries didn't install and I assumed it had. Couldn't figure it out, deleted the folder it was supposed to be in, automatically re-downloaded and all was fine. don't know how or why it didn't before it's just one or those quirks that happen.

clembo1, 2023-08-06:
Quick update. I finally found the root cause of the issue; Visual Studio Code.

I ran yet another test with the nightly build of Marlin yesterday and had the same LCD issue show up.  I then tried an alternative path using mriscoc's pro firmware based on Marlin and switched back to stock dwin lcd...compiled...and still got the exact same issue...this time with a different LCD.  Then I tried a precompiled binary from mriscoc; stock lcd worked.  This helped me isolate the issue down MS Visual Studio Code.  I do not understand exactly how VSC could compile successfully (no errors) and still mess up serial communication.  To solve the issue, I uninstalled VSC, purged all cache/folders including the .platformio folder > reinstalled VSC Platformio and Marlin Autobuild > compiled firmware, installed on the SKR Mini E3V3.  Everything worked on the printer for each build.

classicrocker883, 2023-08-10:
yes like i said, the .platformio\packages folder had to be cleared to be redownloaded.