MarlinFirmware/Marlin #26137: [BUG] SD Card cant start g code when sellect, then SD card cant reads files

omaryouns, 2023-07-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Scenario 1:
1- Power on the machine
2- Marlin starts normally 
3- push on the sd card icon 
4- the sd carn opens with files uploaded on it 
5- selecting any g code file 
6- g code start executing normally ==> NOK

Expected:  5- g code start executing normally
Obtained:  5- after selecting the g code, the main menu appears without any response, when trying to select the sd card again it opens without any file appearing 

Scenario 2:
1- Power on the machine
2- Connect to prontoface 
3- the machine is connected normally 
4- select any g code from the sd card to spring 
5- g code start executing normally ==> NOK

Expected:  5- g code start executing normally
Obtained:  5- after selecting g code, the g code starts executing but after 20 to 30 secs machine stops suddenly  & returns SD Card read Error to prontoface 

Note: Marlin work read sd card normally with maple environment  

 

### Bug Timeline

Simillar to  #25174

### Version of Marlin Firmware

Marlin 2.1.2.1

### Printer model

Kingroon KP5L

### Electronics

Stock electronics "MKS ROBIN NANO"

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/12210046/Configuration.zip)

ellensp, 2023-07-31:
not bugfix configs...  
not even stock Marlin 2.1.2.1 kingron KP5L configs  (different motherboard and display)  and many many other changes

ellensp, 2023-07-31:
Tested on mks robin nano v1.2 with mks robin tft35 (running as a MKS_ROBIN_TFT24)

Using bugfix 2.1.x 

no issue found

I suspect your sdcard is Dying / damaged or needs a reformat

omaryouns, 2023-07-31:
> Tested on mks robin nano v1.2 with mks robin tft35 (running as a MKS_ROBIN_TFT24)
> 
> Using bugfix 2.1.x, printed a 20x20x10 test cube.
> 
> no issue found
> 
> I suspect your sdcard is Dying / damaged or needs a reformat

SD card works perfectly when I flash the same configuration but with mable option, it works perfectly also on computer, issue isn't from SD card

leftbrasil, 2023-08-24:
@omaryouns did you figure out what was going on? I have the same problem on a neptune 2 (znp robin nano 1.3) using latest bugfix

closed 2023-11-03 by github-actions[bot]