MarlinFirmware/Marlin #26138: [BUG] Issue with `hotend_max_target(0)`. should it be `hotend_max_target(1)`

classicrocker883, 2023-07-31:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

So while I was optimizing some code, I took note from [this commit here](https://github.com/MarlinFirmware/Marlin/commit/fc2272c4b473a45f2033d4dd82aef4b259b99759#diff-e275ff094f5fdecfd06e3b9cc637c13f78d98612fc3b8bbd47874feaa7c9453d)  
where many `thermalManager.*` functions were renamed.

I decided to follow suite and sort of copy that into *ProUI*.

- Here me out, even though this isn't _particular_ to the current Marlin repo and source code, it is still relevant as this may effect the same code as a whole.
<br>

The thing that didnt work out was in the file **src/lcd/e3v2/proui/dwin.cpp** and changing 
```c
#define MAX_ETEMP  (thermalManager.hotend_maxtemp[0] - (HOTEND_OVERSHOOT))
```
to -->  
```c
#define MAX_ETEMP  thermalManager.hotend_max_target(0)
```
<br>

I thought this would work out since `hotend_max_temp` is defined in **src/module/temperature.h** as
```c
static celsius_t hotend_max_target(const uint8_t e) { return hotend_maxtemp[e] - (HOTEND_OVERSHOOT); }
```
however, when manually setting the Temperature from the LCD menu limits the HotendTemp to 210.  
this is the code which gets implimented when asking to set temperature through LCD:
```c
void SetHotendTemp() { SetIntOnClick(MIN_ETEMP, MAX_ETEMP, thermalManager.degTargetHotend(0), ApplyHotendTemp); }
```

so it limits `MAX_ETEMP` not to the defined max temp of `HEATER_0_MAXTEMP` but to 210 for whatever arbitrary reason.

Then, setting it to `thermalManager.hotend_max_target(1)` solved the issue.
I mean I suppose I could revert it back to `thermalManager.hotend_maxtemp[0] - (HOTEND_OVERSHOOT))`  
but it is troubling me why `hotend_max_target(1)` works and at `(0)` it doesnt.

can someone explain if there is another type of solution, or why would the fix to `1` instead of `0` actually work? 

and could this effect the same code used in other files??

I just don't understand how `hotend_max_target(0)` passes the `0` to return essentually the same code **which** DOES work originally. but doesn't when using that same code as `returned`?

what if this was changed to

```c
static celsius_t hotend_max_target(const uint8_t e) { return thermalManager.hotend_maxtemp[e] - (HOTEND_OVERSHOOT); }
```

perhaps it is returning `hotend_maxtemp` and needs the Temperature class preceding it?

### Bug Timeline

New, but not particular to anyone else.

### Expected behavior

Should let me select hotend temperature to anything under the max limit, but this sets the max limit to 210 instead of 300 (which is what I had limited to)

### Actual behavior

Value on LCD display is limited to 210. the screen will allow me to draw a value up to 300, but selecting anything > 210 reverts it to 210. It will let me select anything =< 210, but no more.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Voxelab Aquila

### Electronics

Creality 4.2.7 board

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

possible affected files, anywhere that has `thermalManager.hotend_max_target(0)`:  
**src/gcode/lcd/M145.cpp**  
**src/lcd/tft/touch.cpp**  
**src/lcd/menu/menu_advanced.cpp**  
**src/lcd/menu/menu_tune.cpp**  
**src/lcd/menu/menu_temperature.cpp**  
**src/lcd/menu/menu_ubl.cpp**  
**src/e3v2/creality/dwin.cpp**

```c
thermalManager.hotend_max_target(0)
```

classicrocker883, 2023-07-31:
reverting back to `thermalManager.hotend_maxtemp[0] - HOTEND_OVERSHOOT` still did not work.

everything was just working normally I wonder what went wrong

thisiskeithb, 2023-07-31:
> Here me out, even though this isn't particular to the current Marlin repo and source code, it is still relevant as this may effect the same code as a whole.

This can't be a bug since the changes you're making do not exist in current code.

With current code, increasing the max temperature for both 2004 & 12864 displays works >210 and up to the config limit (minus safety).

These types of questions are best suited for our [Discord server](https://discord.gg/n5NJ59y).

closed 2023-07-31 by thisiskeithb

classicrocker883, 2023-07-31:
whatever the reason it has something to do with `AUTOTEMP_MIN`

I'm trying to back track to see what changed and for the life of me cant figure it out.

i will manually set the temperature through the LCD and it will automatically limit itself via the value of AUTOTEMP_MIN

I'll have to look to see why it is choosing this value. Even though I just noticed this, for some odd reason its not an issue with a recent build 3 days ago, but I reinstalled a June build and the issue is present.

so whatever it may be is from some Marlin Update in that timeframe.

Not sure why it lets me set the temp during a print, but will not go passed the predefined minimum autotemp. dont know what or why its being called to do that but I hope someone may help out and know.

classicrocker883, 2023-08-01:
UPDATE:
---

finally figured it out. after a dozen trial and errors I managed to narrow it down to being a custom workaround which I had been playing with the code to get right (almost 2 months ago).

So I basically incorporated the most recent code that worked and that fixed the issue. I was looking up and down for anything related to `AUTOTEMP_MIN` since it was directly related to that value. However that was not the case, for the strangest reason it came to something completely unrelated. and to make matters worse it only happened in non-Maple environment, so it was very specific as to what was wrong.

anyway just wanted to make the update for prosperity sake. file involved was **src/lcd/marlinui.cpp**

for *reasons*, many MarlinUI class function values are defined here, then also it says it is initalized in **settings.cpp**.
originally I initialized my custom workaround values here, then decided to go with the flow and have it initialized in settings.cpp.

at the time I saw no difference and left it. but here it was affecting something unrelated for no real logical reason.

is there a benefit as to why some are initialized here in **marlinui.cpp** vs **settings.cpp**?