MarlinFirmware/Marlin #26139: BUGFIX 2.1.x (latest) [Bug] Cannot compile, DISTINCT_E_FACTORS and too many initializers for 'const xyze_float_t'

LightningShark25, 2023-08-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin will not compile.  Was OK immediately prior to the following change:

From:
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80.1, 80.5, 401.6, 139 }
To:
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80.1, 80.5, 401.6, 139, 142}

Background:
Set #define DISTINCT_E_FACTORS
Number of extruders is 2 (they have been working fine for years)

### Bug Timeline

Immedietely after adding the additional E1 steps value to 'DEFAULT_AXIS_STEPS_PER_UNIT as above

### Expected behavior

Parse the additional esteps value for E1 and compile

### Actual behavior

An extract of the errors provided.. all errors are similar to this

*** [.pio\build\BIGTREE_SKR_2\src\src\feature\babystep.cpp.o] Error 1
*** [.pio\build\BIGTREE_SKR_2\src\src\MarlinCore.cpp.o] Error 1
In file included from Marlin\src\feature\../inc/MarlinConfigPre.h:39,
                 from Marlin\src\feature\pause.cpp:28:
Marlin\src\feature\../inc/../../Configuration.h:1227:68: error: too many initializers for 'const xyze_float_t' {aka 'const XYZEval<float>'}  
 1227 | #define DEFAULT_AXIS_STEPS_PER_UNIT   { 80.1, 80.5, 401.6, 139, 142}
      |                                                                    ^
c:\users\kevin\documents\3d printing\firmware\marlin\marlin\src\lcd\e3v2\common\limits.h:93:40: note: in expansion of macro 'DEFAULT_AXIS_STEPS_PER_UNIT'
   93 |                        default_steps = DEFAULT_AXIS_STEPS_PER_UNIT,
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pio\build\BIGTREE_SKR_2\src\src\feature\pause.cpp.o] Error 1
In file included from Marlin\src\feature\../inc/MarlinConfigPre.h:39,
                 from Marlin\src\feature\../inc/MarlinConfig.h:28,
                 from Marlin\src\feature\tmc_util.cpp:23:
Marlin\src\feature\../inc/../../Configuration.h:1227:68: error: too many initializers for 'const xyze_float_t' {aka 'const XYZEval<float>'}  
 1227 | #define DEFAULT_AXIS_STEPS_PER_UNIT   { 80.1, 80.5, 401.6, 139, 142}
      |                                                                    ^
Marlin\src\feature\../lcd/e3v2/creality/../common/limits.h:93:40: note: in expansion of macro 'DEFAULT_AXIS_STEPS_PER_UNIT'
   93 |                        default_steps = DEFAULT_AXIS_STEPS_PER_UNIT,
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\gcode\bedlevel\abl\../../../inc/MarlinConfigPre.h:39,
                 from Marlin\src\gcode\bedlevel\abl\../../../inc/MarlinConfig.h:28,
                 from Marlin\src\gcode\bedlevel\abl\G29.cpp:27:
Marlin\src\gcode\bedlevel\abl\../../../inc/../../Configuration.h:1227:68: error: too many initializers for 'const xyze_float_t' {aka 'const XYZEval<float>'}

### Steps to Reproduce

Change value

From:
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80.1, 80.5, 401.6, 139 }
To:
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80.1, 80.5, 401.6, 139, 142}

Background:
Set #define DISTINCT_E_FACTORS
Number of extruders is 2 (they have been working fine for years)

### Version of Marlin Firmware

BugFix 2.1.x

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

LightningShark25, 2023-08-01:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/12223135/Configuration.zip)

ellensp, 2023-08-01:
Issue is with Ender-3 v2 OEM display, most of the UI's have an min_steps_edit_values
In Marlin/src/lcd/e3v2/common/limits.h
This is hardcoded to expect only 4 values....

LightningShark25, 2023-08-01:
Thanks @ellensp 
I have commented out the display and can confirm it compiles
I only tinker around the edge of code.. is this an 'easily' fixable issue or a showstopper?

Thanks

classicrocker883, 2023-08-03:
that definitely is the case, the LCD must be disabled for it to compile successful. however, I'm sure youd be able to edit the LCD DWIN display code to incorporate the extra variable. 

OTHERWISE, I believe you can manually set this with a gcode command. M92 is it?

ellensp, 2023-08-03:
I suspect reverting commit https://github.com/MarlinFirmware/Marlin/commit/5626638739698ba07f0e8720836d16d6c9bb07c5 
would fix it. 

As the original poster used constexpr float function[] which scales with the number of axis..