MarlinFirmware/Marlin #26148: [BUG] Character LCD Menus Misaligned

SparkyDan555, 2023-08-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Some LCD submenus are displayed with misalignment. See photos below.

### Bug Timeline

Started sometime between eca80db and 87231b6

### Expected behavior

Menus displayed correctly.

### Actual behavior

Alignment of menu items is wrong and some menus just look generally messed up. 

Examples:
[image]
[image]
[image]

### Steps to Reproduce

1. Upload latest bugfix
2. Navigate to submenus on LCD
3. Observe incorrect display

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Geeetech A10M

### Electronics

Stock board, stock LCD

### Add-ons

None

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/12263785/Archive.zip)

thisiskeithb, 2023-08-04:
This doesn't show up on all menus, but I can confirm the bug here on a `REPRAP_DISCOUNT_SMART_CONTROLLER`.

thisiskeithb, 2023-08-05:
This was broken in #26073

```
d036097bca52b128b421342afcdb908cb1d8d5ab is the first bad commit
commit d036097bca52b128b421342afcdb908cb1d8d5ab
Author: lukasradek <lukasradek@users.noreply.github.com>
Date:   Thu Jul 27 23:08:21 2023 +0200

    🚸 Fix HD44780 EDIT_ITEM position (#26073)

 Marlin/src/lcd/HD44780/lcdprint_hd44780.cpp | 2 +-
 Marlin/src/lcd/HD44780/marlinui_HD44780.cpp | 3 ++-
 Marlin/src/lcd/TFTGLCD/lcdprint_TFTGLCD.cpp | 2 +-
 Marlin/src/lcd/dogm/lcdprint_u8g.cpp        | 2 +-
 Marlin/src/lcd/lcdprint.cpp                 | 2 ++
 5 files changed, 7 insertions(+), 4 deletions(-)
```

lukasradek, 2023-08-05:
There is a misunderstanding between callers and callees here 

https://github.com/MarlinFirmware/Marlin/blob/f66323ecfbe067667234df286e14d43d409bfaea/Marlin/src/lcd/lcdprint.cpp#L47C9-L47C9

The underlying `lcd_put_u8str_P` declares `Return the given maxlen minus the number of characters emitted, i.e., the number of unused columns` while all of its callers just return that value as output width.

SparkyDan555, 2023-08-20:
I have tested #26154 and it fixes this issue. I will close this issue as others confirmed the fix and it looks likely to be merged.

closed 2023-08-20 by SparkyDan555