MarlinFirmware/Marlin #26155: [FR] Add an extra argument to M600, to show to which filament the user has to change.

plampix, 2023-08-05:
### Is your feature request related to a problem? Please describe.

When doing rather simple multi-colour prints with just one nozzle, no mmu, and you use more than 2 colours, it can be hard to tell to which filament you need to change. Just adding a number to the message on screen could help a lot.

### Describe the feature you want

An extra argument to M600 that contains text that can be displayed in the 'Insert filament and press button' message, for example replace it with 'Insert filament 3 and press button'.

DerAndere1, 2023-08-06:
This would mean that we introduce a new standard that duplicates the purpose of tool change (command T). With your proposal, slicer, firmware and host software have to track both, tool number and filament number. I think it is more consistent to just use the established tool change (T command) and implement a manual toolchange procedure that in your case only includes a filament change (see pull request https://github.com/MarlinFirmware/Marlin/pull/23575). Then, each tool number represents a different filament.

plampix, 2023-08-06:
Yeah, that should certainly work, and be less redundant. I'll keep watching that PR, and close this one.

closed 2023-08-06 by plampix

DerAndere1, 2023-08-07:
for your specific usecase, after merging PR 23575, you will have to add this  line of code: 
```M600();```
somewhere here https://github.com/MarlinFirmware/Marlin/pull/23575/files#diff-9876938bb66214705f89adf5ab50fd49d5504d318662b5c996ac0eba2c434587L646

plampix, 2023-08-07:
Either that, or add M702/M701. I'll figure it out when MANUAL_SWITCHING_TOOLHEAD becomes available.

DerAndere1, 2023-08-07:
Probably the merge of that PR into official bugfix-2.1.x won't happen anytime soon. You can checkout the pull request yourself: 

lets say you have forked Marlin and https://github.com/MarlinFirmware/Marlin is the `upstream` remote repository. 

Then you can fetch an upstream PR like this: 

git fetch upstream pull/$ID/head:$BRANCHNAME

where $ID is the pull request id and $BRANCHNAME is the name of the new branch that you want to create (replace both placeholders!). Once you have created the branch, then simply checkout your newly created branch:

git checkout $BRANCHNAME

(modified from https://stackoverflow.com/questions/27567846/how-can-i-check-out-a-github-pull-request-with-git)

EvilGremlin, 2023-08-07:
On-screen status message like "Insert green PETG" will be very helpful, but it can be done in two ways.
First is to include status message on filchange screen, then use M117 in color/tool change gcode.
Second is to add separate string to M600 and filchange screens.
In any case, slicers must support inserting _next filament name_ as variable, which i don't think any do.

plampix, 2023-08-08:
PrusaSlicer can do this, and I already use that with M117 and  M118, but the message disappears when the display shows the change filament status screen.

But there's no rush, I can always check my Octoprint to check which filament I need. Would be nice to also show it on the printer though, for people that print from SD.

plampix, 2023-08-08:
Hmm, I could just show it directly after changing filament, instead of just before changing. Then I just prepare for the next change, and do whatever was prepared when M600 runs. I'm stupid :)