MarlinFirmware/Marlin #26156: [BUG] SKR Mini E3 V3 stop mid print - with touch BTT GD TFT 3.5 V3

jacksonport, 2023-08-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

On Ender-5 Plus, when printing with BTT SKR Mini E3 V3.0 with BTT GD TFT35 V3 attached in touch screen mode, printer reboots at some random moment and then powers on again. Print ruined as printing not continued after it and don't see unfinished job.
Happens inconsistently. Last time print lasted about 24 hours before reboot.

### Bug Timeline

Started when trying new SKR Mini E3 V3 board

### Expected behavior

Expected print to complete.

### Actual behavior

Printer reboots at some random moment with heaters off.

### Steps to Reproduce

Start a print from TFT Touchscreen.

### Version of Marlin Firmware

2.1.2.1

### Printer model

Creality Ender-5 Plus

### Electronics

BIGTREETECH SKR MINI E3 V3.0

### Add-ons

CRTouch connected via Z-Stop

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Screen firmware version: BIGTREE_GD_TFT35_V3.0.27.x

Prusa Slicer v2.6.0 
G-code flavour: Marlin 2

My compiled firmware attached in archive. Can be installed only with microSD cards of size 4GB or less.

thisiskeithb, 2023-08-06:
> Screen firmware version: BIGTREE_GD_TFT35_V3.0.27.x Feb 15 2023

I'd recommend updating the TFT firmware since it was last updated on June 4:
[image]

> Version of Marlin Firmware
> 2.1.2.1

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

jacksonport, 2023-08-06:
> I'd recommend updating the TFT firmware since it was last updated on June 4: [image]
>
>
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test [...]

Thanks, I'll try a new TFT firmware. 
Why it's still called "[BIGTREE_GD_TFT35_V3.0.27.x.bin] (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIGTREE_GD_TFT35_V3.0.27.x.bin)", while it's 4 months older than the previous one that has the same name?

jacksonport, 2023-08-06:
> I'd recommend updating the TFT firmware since it was last updated on June 4: [image]
>
>
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test [...]

Just updated TFT with bin from June 4 archive and it still shows "BIGTREE_GD_TFT35_V3.0.27.x Feb 15 2023".

classicrocker883, 2023-08-08:
rename your firmware file then reflash

jacksonport, 2023-08-08:
> rename your firmware file then reflash
[image]

Here is an excerpt from BTT documentation above and that is pretty much what I've got. Everything on a microSD card became .CUR after update. Menu icons theme also updated.

classicrocker883, 2023-08-09:
> Here is an excerpt from BTT documentation above and that is pretty much what I've got. Everything on a microSD card became .CUR after update. Menu icons theme also updated.

have you reformatted the SD card? and sometimes using 8gb or smaller works. there are issues formatting using a MAC as well.

jacksonport, 2023-08-09:
> > Here is an excerpt from BTT documentation above and that is pretty much what I've got. Everything on a microSD card became .CUR after update. Menu icons theme also updated.
> 
> have you reformatted the SD card? and sometimes using 8gb or smaller works. there are issues formatting using a MAC as well.

I used 4Gb microSD. All update process went well - no errors.

radek8, 2023-08-10:
Test FW from Kisslorand
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2761

jacksonport, 2023-08-10:
> Test FW from Kisslorand [bigtreetech/BIGTREETECH-TouchScreenFirmware#2761](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2761)

Thank you. I'll have a look.

husakki, 2023-08-11:
Ive been having similar issues buth with the BTT TFT35 E3 V3.

I used Kisslorand FW for the TFT and Marlin newest bugfix branch. Both did not really do much in my case **BUT** could be different in yours.

_I saw that you have input shaping enabled._
Currently I'm printing with a marlin version that has no input shaping in it, because I kinda suspected that it would be because of that and my "test print" (the one where i've been encountering issues with) has not failed yet. So maybe even disable that although it gave me quite some quality prints.

jacksonport, 2023-08-12:
> Ive been having similar issues buth with the BTT TFT35 E3 V3.
>
> I used Kisslorand FW for the TFT and Marlin newest bugfix branch. Both did not really do much in my case **BUT** could be different in [...]

I have a sneaky suspicion that this problem is caused by the content of the microSD card. Once I've inserted some card with many folders in it, connected USB to laptop, started Octoprint and pressed "Connect". Right after connection it attempted to get list of files on microSD in printer and that caused a reset. Reconnect causes reset all the time. I've changed a microSD on another one 4GB with only several files in the root and all worked fine.

I've also observed that when printing in touchscreen mode it receives some distorted messages at random moment, like "B0sy: Processing". Once I've seen "SD card OK" during printing. This message shown normally when I insert microSD card into slot onboard.

classicrocker883, 2023-08-12:
> messages at random moment, like "B0sy: Processing".

I don't think it's supposed to say "B0sy", it should say "Busy"

husakki, 2023-08-12:
I dont keep folders on my "USB Stick" which I plug into the TFT. I testest different kinda of USB's and also tested the micoSD onboard. Altough it came into my mind to also test everything out in marlin mode, I mean if i just have to switch to marlin mode for printing a part that would be fine for me, if it means I can put input shaping back on.

#26055 Here Nerdcaught switch to Octoprint and didn't entcounter any weird issues anymore (can be seen on the video)

jacksonport, 2023-08-13:
> > messages at random moment, like "B0sy: Processing".
> 
> I don't think it's supposed to say "B0sy", it should say "Busy"

That's right. Often in these messages letters are substituted with numbers.

closed 2023-10-24 by github-actions[bot]