MarlinFirmware/Marlin #26157: [BUG] (Crealty Ender S1 Plus) does not print objects with large area

mkemlogic, 2023-08-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Tested with a629471
Have tried to print object with area of 195x237cm or more on Ender S1 Plus. The printer does not react when the file is selected for printing.  No movement and no heating.
 Smaller objects print fine. Don't know where the limit goes. Same objects print fine with Crealty FW Ender-3S1Plus_HWv24S1_301_SWV2.0.8.26F4H_F401_FDM_LASER.bin.

### Bug Timeline

Just started using Marlin FW so don't know if it is old or new

### Expected behavior

I expected the extruder to move into start position and turn on heat.

### Actual behavior

No movement and no heating when a file with a large object is selected for printing

### Steps to Reproduce

1. Create an object with area 195x237cm
2. Try to print on Ender S1 Plus

### Version of Marlin Firmware

commit a629471

### Printer model

Ender S1 Plus

### Electronics

Extruder for Ender S1 Pro, but config is for Ender S1 Plus extruder

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/12268909/Configuration.zip)

classicrocker883, 2023-08-08:
i want to say this could have to do with the gcode file / slicer.

what if you took the top portion - start code - of the gcode file that works, and replace it with the gcode file of the one that causes the issue.

mkemlogic, 2023-08-08:
> 
I can retry everything I did in case I missed something. If I did everything as I remember it, I used the same gcode file with both Marlin FW and Crealty FW and it only worked with Crealty.
I will come back with new results.

mkemlogic, 2023-08-08:
I am now able to print full size objects with this FW. I have no idea why it did not work in first place.

thisiskeithb, 2023-08-08:
Glad it works!

closed 2023-08-08 by thisiskeithb