MarlinFirmware/Marlin #26161: flash marlin2.1.x to esp32s3

yangyuchang, 2023-08-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Marlin2.1. x can run normally on esp32s when I flash Marlin2.1. x to esp32s3. It will make this error
[image]

### Version of Marlin Firmware

marlin2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin-2.1.2.1.zip](https://github.com/MarlinFirmware/Marlin/files/12271640/Marlin-2.1.2.1.zip)

ellensp, 2023-08-07:
Clearly not a marlin issue, but a build tools issue.

ellensp, 2023-08-07:
Attached config files are not a zip file, It is a rar files

ellensp, 2023-08-07:
Provided Configs are for a BOARD_GEN7_CUSTOM which is a  ATMega644P, not any sort of ESP32

ellensp, 2023-08-07:
Marlin uses espressif32@2.1.0 released Dec 3, 2020, ie this is rather old

yangyuchang, 2023-08-09:
@ellensp Thanks for your answer, I will try

closed 2023-08-11 by yangyuchang