MarlinFirmware/Marlin #26162: [BUG] Tronxy STMF446ZET6 board - Error with TFT

Pelu88, 2023-08-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I configured all the Configuration.h, Configuration_adv.h and platformio.ini for a Tronxy's board with STM32F446ZET chip. I selected the TFT_TRONXY_X5SA LCD as my screen. When I compile I get this error:

Tool Manager: framework-arduinoststm32@4.10900.200819 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
In file included from c:\users\manonif\documents\marlin-bugfix-2.1.x-pelu\marlin\src\pins\pins.h:800,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:36,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
c:\users\manonif\documents\marlin-bugfix-2.1.x-pelu\marlin\src\pins\stm32f4/pins_TRONXY_V10.h:190:4: error: #error "TFT_TRONXY_X5SA is not yet supported."
  190 |   #error "TFT_TRONXY_X5SA is not yet supported."
      |    ^~~~~
Error: Failed to parse Marlin features. See previous error messages.

I'd like to know which TFT I have to define.

### Version of Marlin Firmware

Marlin-bugfix-2.1.x

### Printer model

Tronxy XY3 Pro-V2

### Electronics

Stock electronics (STM32F446ZET6)

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/12284720/Configuration.files.zip)

Pelu88, 2023-08-07:
> It's not a bug, it's feature request. Somweone with skills and hardware have to do this.

Sorry I didn't know where to post this, I'm new in this

jmz52, 2023-08-16:
@Pelu88, F446 code available at https://github.com/jmz52/Marlin/tree/stm32f4-tft-spi-fix
Please note that I had to disable `CONTROLLER_FAN_MIN_SOC_TEMP` and `LIN_ADVANCE` features in your configuration files to avoid compilation errors.

Pelu88, 2023-08-16:
> @Pelu88, F446 code available at https://github.com/jmz52/Marlin/tree/stm32f4-tft-spi-fix Please note that I had to disable `CONTROLLER_FAN_MIN_SOC_TEMP` and `LIN_ADVANCE` features in your configuration files to avoid compilation errors.

@jmz52 Alex, I've just downloaded it. I'm seeing it and I have a few questions. Do I have to set the env in platformio don't I? and also I have to configure all the configuration.h and configuration_adv.h? As soon as I can compile and test it I would make you know if it's ok.
Thanks a lot for your time

jmz52, 2023-08-17:
Yes, the board-specific changes to platformio.ini and configuration files are not included.
I was using `[env:STM32F446_tronxy]` build environment and configuration files you've provided.

Pelu88, 2023-08-19:
@jmz52 Alex, I've setted up the configuration.h and configuration_adv.h disabling the CONTROLLER_FAN_MIN_SOC_TEMP and LIN_ADVANCE features as you recommended, but I still having compilation errors.
Here it is the terminal debug:

~~~~
 *  Ejecutando tarea: C:\Users\franc\.platformio\penv\Scripts\platformio.exe run 

Processing STM32F446_tronxy (board: marlin_STM32F446ZET_tronxy; platform: ststm32@~12.1; framework: arduino)
-----------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
In file included from d:\compis marlin\marlin-2.1.2.1-gavrilenko\marlin\src\pins\pins.h:800,
                 from d:\compis marlin\marlin-2.1.2.1-gavrilenko\marlin\src\inc\marlinconfig.h:36,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
d:\compis marlin\marlin-2.1.2.1-gavrilenko\marlin\src\pins\stm32f4/pins_TRONXY_V10.h:190:4: error: #error "TFT_TRONXY_X5SA is not yet supported."
  190 |   #error "TFT_TRONXY_X5SA is not yet supported."
      |    ^~~~~
Error: Failed to parse Marlin features. See previous error messages.

~~~~

Then I've tried defining with generic TFT and I get this errors:

~~~~
In file included from Marlin\src\HAL\STM32\tft\xpt2046.cpp:31:
Marlin\src\HAL\STM32\tft\xpt2046.h:39:4: error: #error "TOUCH_MISO_PIN is not defined."
39 | #error "TOUCH_MISO_PIN is not defined."
| ^~~~~
In file included from C:\Users\franc.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:4,
from C:\Users\franc.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/pinmap.h:22,
from C:\Users\franc.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PeripheralPins.h:34,
from C:\Users\franc.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/analog.h:44,
from C:\Users\franc.platformio\packages\framework-arduinoststm32\cores\arduino/board.h:8,
from C:\Users\franc.platformio\packages\framework-arduinoststm32\cores\arduino/wiring.h:41,
from C:\Users\franc.platformio\packages\framework-arduinoststm32\cores\arduino/Arduino.h:36,
from d:\compis marlin\marlin-2.1.2.1-gavrilenko\marlin\src\hal\shared\marduino.h:36,
from Marlin\src\HAL\STM32\tft../../../inc/../HAL/../HAL/STM32/HAL.h:28,
from Marlin\src\HAL\STM32\tft../../../inc/../HAL/HAL.h:30,
from Marlin\src\HAL\STM32\tft../../../inc/MarlinConfig.h:33,
from Marlin\src\HAL\STM32\tft\xpt2046.cpp:27:
Marlin\src\HAL\STM32\tft\xpt2046.h: In static member function 'static void XPT2046::dataTransferBegin()':
Marlin\src\HAL\STM32\tft\xpt2046.h:76:87: error: 'TOUCH_CS_PIN' was not declared in this scope; did you mean 'TOUCH_INT_PIN'?
76 | static void dataTransferBegin() { if (SPIx.Instance) { HAL_SPI_Init(&SPIx); } WRITE(TOUCH_CS_PIN, LOW); };
| ^~~~~~~~~~~~
C:\Users\franc.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNamesTypes.h:129:34: note: in definition of macro 'STM_PORT'
129 | #define STM_PORT(X) (((uint32_t)(X) >> 4) & 0xF)
| ^
Marlin\src\HAL\STM32\tft../../../inc/../HAL/../HAL/STM32/fastio.h:59:49: note: in expansion of macro 'digitalPinToPinName'
59 | #define _WRITE(IO, V) (FastIOPortMap[STM_PORT(digitalPinToPinName(IO))]->BSRR = _BV32(STM_PIN(digitalPinToPinName(IO)) + ((V) ? 0 : 16)))
| ^~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32\tft../../../inc/../HAL/../HAL/STM32/fastio.h:70:33: note: in expansion of macro '_WRITE'
70 | #define WRITE(IO,V) _WRITE(IO,V)
| ^~~~~~
Marlin\src\HAL\STM32\tft\xpt2046.h:76:81: note: in expansion of macro 'WRITE'
76 | static void dataTransferBegin() { if (SPIx.Instance) { HAL_SPI_Init(&SPIx); } WRITE(TOUCH_CS_PIN, LOW); };
| ^~~~~
Marlin\src\HAL\STM32\tft\xpt2046.h: In static member function 'static void XPT2046::dataTransferEnd()':
Marlin\src\HAL\STM32\tft\xpt2046.h:77:41: error: 'TOUCH_CS_PIN' was not declared in this scope; did you mean 'TOUCH_INT_PIN'?
77 | static void dataTransferEnd() { WRITE(TOUCH_CS_PIN, HIGH); };
| ^~~~~~~~~~~~
C:\Users\franc.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNamesTypes.h:129:34: note: in definition of macro 'STM_PORT'
129 | #define STM_PORT(X) (((uint32_t)(X) >> 4) & 0xF)
| ^
Marlin\src\HAL\STM32\tft../../../inc/../HAL/../HAL/STM32/fastio.h:59:49: note: in expansion of macro 'digitalPinToPinName'
59 | #define _WRITE(IO, V) (FastIOPortMap[STM_PORT(digitalPinToPinName(IO))]->BSRR = _BV32(STM_PIN(digitalPinToPinName(IO)) + ((V) ? 0 : 16)))
| ^~~~~~~~~~~~~~~~~~~
Marlin\src\HAL\STM32\tft../../../inc/../HAL/../HAL/STM32/fastio.h:70:33: note: in expansion of macro '_WRITE'
70 | #define WRITE(IO,V) _WRITE(IO,V)
| ^~~~~~
Marlin\src\HAL\STM32\tft\xpt2046.h:77:35: note: in expansion of macro 'WRITE'
77 | static void dataTransferEnd() { WRITE(TOUCH_CS_PIN, HIGH); };
| ^~~~~
[... 73 lines cut ...]
Marlin\src\HAL\STM32\tft\tft_fsmc.h:97:69: error: 'GPIO_AF12_FSMC' was not declared in this scope; did you mean 'GPIO_AF12_FMC'?
97 | #define FSMC_PIN_DATA STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF12_FSMC)
| ^~~~~~~~~~~~~~
C:\Users\franc.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNamesTypes.h:93:29: note: in definition of macro 'STM_PIN_DEFINE'
93 | ((AFNUM & STM_PIN_AFNUM_MASK) << STM_PIN_AFNUM_SHIFT))
| ^~~~~
Marlin\src\HAL\STM32\tft\tft_fsmc.h:97:27: note: in expansion of macro 'STM_PIN_DATA'
97 | #define FSMC_PIN_DATA STM_PIN_DATA(STM_MODE_AF_PP, GPIO_NOPULL, GPIO_AF12_FSMC)
| ^~~~~~~~~~~~
[... cut ...]
*** [.pio\build\STM32F446_tronxy\src\src\HAL\STM32\tft\tft_fsmc.cpp.o] Error 1
~~~~

I don't really know what's happening since I don't know about programming. I'm only trying to set up a Firmware for my machine as the official from Tronxy has an issue with the filament change or `ADVANCED_PAUSE_FEATURE` which causes the printer to not do the routine as it should.
I re-attached the files that threw these build errors.
[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/12386319/Configuration.files.zip)

thisiskeithb, 2023-08-19:
@jmz52: Did you mean to link to https://github.com/jmz52/Marlin/tree/tft-fmc-stm32f446 instead?

jmz52, 2023-08-19:
Indeed.
@Pelu88, link I've provided was for another issue.
Please use branch https://github.com/jmz52/Marlin/tree/tft-fmc-stm32f446

Pelu88, 2023-08-19:
> Indeed. @Pelu88, link I've provided was for another issue. Please use branch https://github.com/jmz52/Marlin/tree/tft-fmc-stm32f446

@jmz52 Downloaded this branch and I could compile it. I flashed and it worked. But, there are some problems with the screen.
[image]
As you can see, the middle of the screen is smudged. Not in all menus, but it's random that it appears. If I compile de "Marlin" version provided by Tronxy, I don't have this issue. I don't know if anyone can compare the files, maybe that way can find out where's the problem

TheThomasD, 2023-12-14:
Just an idea: Tronxy offers a (hopefully) compilable branch of Marlin here: https://github.com/tronxy3d/F4xx-SIM480x320

There is a lot of stuff in there, including pin mappings and variant configurations for the chipset. Shouldn't it "just" be an application of the things they did in their repo to the standard Marlin version?

TheThomasD, 2023-12-14:
Oh, now I remember :-D I think I did a similar thing back then for my Tronxy Gemini S. I guess it has the same display and the config I created below works for the Gemini. Most likely I just have to apply the changes that I did there as well...

https://github.com/TheThomasD/Marlin/compare/2.1.x...TheThomasD:Marlin:2.1.x-tronxy-gemini-s

TheThomasD, 2023-12-15:
This works for my Tronxy v10 board including the TFT. My board doesn't even have the STM32 chip but the GD32 clone, but the config still seems to work. I basically applied the changes I did back then to the Gemini S and imported the pin config from the tronxy3d-repo mentioned above. That did the trick...

https://github.com/TheThomasD/Marlin/tree/2.1.x-tronxy-xy-2-pro-2e

SyncroSeb, 2024-01-11:
@TheThomasD Grüß dich, danke für die Mühe!
I used your fork to compile for my X5SA Pro with a 446 V6 Board successfully, but the firmware freezes every 30-60 seconds and restart, until after the 3rd or 4th time it stalls with a grey screen, so the screen works fine, the rest not so good.
The screen calibration error i avoid by using a pointy tool like a fine allen wrench, then it works just fine.

The question now is, have you ever had it running? My Board uses a STMF446 CPU not an GP32, I ried to change it like it was done by tronxy in the platformio.ini ("DMCU_TYPE=0", change it to "DMCU_TYPE=4") but I cant find this option...

Any Idea what to try next? Maybe I should try your branch unmodified first, even its a different printer, it should at least show if its stable on th board or not...

I´m sorry, but I don´t know anything about coding, teched myself a bit in modifying firmwares, and thats it. Tell me when I get annoying.

SyncroSeb, 2024-01-11:
Ok tried it with the unmodified version, its unstable as well. Every second command it reinitialises.
Maybe I have to find out what needs to be changed to fit to the STM32F446 CPU not the GD32.

My fork https://github.com/SyncroSeb/ThomasD-Marlin

TheThomasD, 2024-01-11:
Hi @SyncroSeb,

as for the firmware, it works flawlessly on my Tronxy XY-2 PRO 2E and I was able to print multi-color things without an issue (even longer ones).

I also own an X5SA-400, but for that I use a different branch: https://github.com/TheThomasD/Marlin/tree/2.1.x-tronxy-x5sa-400
I fear that my X5SA is running another chip (BOARD_CHITU3D_V6 // Chitu3D TronXY X5SA V6 Board (STM32F103ZE)), so this might not work for you.

My Tronxy Gemini S, however, should have a similar chip (see repo https://github.com/TheThomasD/Marlin/tree/2.1.x-tronxy-gemini-s). Maybe you (or someone else) can try starting the journey from there?!

Unfortunately, I don't have an X5SA with another chip, so I cannot provide a solution here. If you don't have any experience in programming (and potentially in electronics), it might be hard for you to solve this on your own.

SyncroSeb, 2024-01-12:
@TheThomasD looks a lot like I´m stuck here with my limited knowledge.
I´m pretty much limited to comparing the configration.h and configration_adv.h of the different versions (Yours, Tronxys, Main branch etc) and trying to reverse engeneer the changes you made for the use with dual extruders, Tronxy made to make our lives hard etc. 
I also tried to mod the actual 2.1.x branch to use with the 446 motherboard without touchscreen, but there are so many f*cked up dependencies I cant even manage to do this, each time i try to compile theres another error asking for files and lines releated to the TFT. I just cannot find where to disable the screen completely. I have a thinclient ready to use with repetierserver and a 10" Tablet as  Touchscreen. But without a modern firmware there no use.

Its not only about me, I´d really like to find a solution to this challenge, because I know there are a lot of poeple who do not want or cant change to klipper, change the whole board etc. 
I´m installing the old firmware (Marlin 2.0, no Input shaping etc) again and wait some months for someone who´s a bit better in coding than a Schornsteinfeger ;-)

TheThomasD, 2024-01-12:
@SyncroSeb if you have telegram then we can maybe chat or talk through that. You can reach me here https://telegram.me/thethomasd

SyncroSeb, 2024-01-20:
Thanx to and with a lot of  TheThomasD I was able to configure a repo based on his gemini S firmware for my X5SA Pro 446 V6.

Use to your liking. Its roughly configured like the original Marlin example config for the X5SA F103.

https://github.com/SyncroSeb/Marlin-for-Tronxy-X5SA-Pro-446

thisiskeithb, 2024-06-02:
Added in https://github.com/MarlinFirmware/Marlin/pull/26747

closed 2024-06-02 by thisiskeithb