MarlinFirmware/Marlin #26166: [FR] Add calibration patterns for Input Shaping to M593

tombrazier, 2023-08-08:
### Describe the feature you want

When I get time this is what I want to add to `M593` but if someone else gets there before me, that would be fantastic...

The Input Shaping calibration patter we ~~stole~~ borrowed from Klipper does not do a good job. It uses a lot of filament and takes a long time to print. But, worst, longitudinal oscillations are often a lot more visible on the pattern than transverse oscillations meaning the X and Y resonant frequencies are overlaid on top of each other the value measured for X is often actually the value for Y and vice versa.

A much better alternative is to draw a zig-zag line at increasing speed and, thereby, performs a frequency scan on just one axis. I have written a [Python script](https://github.com/tombrazier/Marlin/blob/speedboatrace/ringing_pattern.py) to generate gcode for this. This gives much more accurate results and also makes zeta calibration a lot easier.

Sample generated gcode files are:

[Frequency scans in Y and X](https://github.com/tombrazier/Marlin/blob/speedboatrace/freq.gcode).
[Zeta calibration pattern for X](https://github.com/tombrazier/Marlin/blob/speedboatrace/zeta_x.gcode).
[Zeta calibration pattern for Y](https://github.com/tombrazier/Marlin/blob/speedboatrace/zeta_y.gcode).

It would be fairly trivial to incorporate these calibration patterns into `M593` but I do not have the capacity for that at present. In the meantime, this FR records what needs to be done.

tombrazier, 2023-08-08:
For anyone who does not want to delve into the Python script to find out how to use these calibration patterns, here is the blurb in the documentation inside the script:
```
This script generates gcode for one of three patterns to tune input shaping. Each pattern
performs a number of frequency scans, starting from 0Hz and going up to some maximum (default
60Hz). A frequency scan is a zig-zag line which is drawn at steadily increasing speed.

Horizontal lines are oscillating in the Y direction. e.g.

  ^^^^^^^^^^^^^^^^^^^^

And vertical lines are oscillating in the X direction. e.g.

  >
  >
  >
  >
  >
  >
  >
  >

Don't get them confused! You read X frequencies off the vertical lines and Y frequencies off
the horizontal line.

At the resonant frequency, the oscillations become noticeably larger and there may even
be layer shifts. If this is a problem, reduce the value of the amplitude variable.

The "freq" pattern draws one frequency scan for Y and then one for X. Measure from the start of
the line to where the oscillations are worst. Divide distance in mm by wavelength (default 2)
to get the frequency.

The "zeta_x" and "zeta_y" patterns draw a series of frequency scans all in one direction. The
zeta value increases by 0.05 for every frequency scan drawn, starting with a value of 0.05.
Find the line which has the most uniform amplitude of oscillation across the whole line. This
gives you the X or Y zeta value.
```

revetuzo, 2023-09-14:
I have tested this on an Ender 3 V2 with some mods (dual z, direct drive). It works very well and allows to find the resonance frequencies and the damping values relatively easily.

I love the idea to take the measurement from the time domain (ringing tower) to the frequency domain (single layer) without using an accelerometer. 

It should be mentioned in the docs that the zeta_x and zeta_y patterns have to be printed after setting the resonance frequencies determined with the freq pattern. Only then you see the transition from an overcompensation at the resonant frequency with too small zeta (too small amplitude) to an undercompensation with too large zeta (too large amplitude).

KimmoHop, 2023-11-05:
I tried this yesterday.

Frequency calibration was sort of simple, after re-reading explanation few times to find out what to look for in the pattern. I got Y starting to go wild at  ~64mm -> 32Hz, while X didn't break -> 60Hz. Maybe I should run the script to generate higher speed - or loosen X belt a bit to bring resonant frequency down?

But that damping... I think that reference images from different printers and explanation what to look for (on M593 page?) would help a lot. My guesses were 0.35 for X and 0.5 for Y. Ender-3-style bedslinger with light direct drive extruder and PEI magnetic steel plate.

If this would become included in firmware, G29 could be optional (removed it after 1st try :D ) + retract & raise Z at the end?
Maybe the code could be used as Cura postprocessing pluging, or web generator (like linear advance calibration pattern generator)?

Other benefits of this test are speed and the sound it makes :)

I'm not sure if python script could be turned into Cura postprocessing plugin?

tombrazier, 2023-11-05:
> X didn't break -> 60Hz. Maybe I should run the script to generate higher speed - or loosen X belt a bit to bring resonant frequency down?

If the X resonant frequency really is higher than 60Hz, that's great. The higher the frequency the lower the amplitude. You won't necessarily see the pattern "go wild" at the resonant frequency but you should see a higher amplitude, i.e. the peaks should be higher. For an Ender-3 style printer I would have thought the resonant frequency would be below 60Hz, based on the X axis design being similar to my own which resonates at about 30Hz.

> But that damping... I think that reference images from different printers and explanation what to look for

Good idea. I am still kind of hoping someone else will pick all this up and implement it.

> Other benefits of this test are speed and the sound it makes :)

:)

KimmoHop, 2023-11-05:
> > But that damping... I think that reference images from different printers and explanation what to look for
> 
> Good idea. I am still kind of hoping someone else will pick all this up and implement it.

I can at least try to create calibration pattern page to docs (based on linear advance calibration page). I guess current M593 page would be the place for pattern images and how to interpret them - where I will need some help to stay in style.

@tombrazier & @revetuzo  - if you have taken images from your calibration patterns, that would help! I will also re-test calibration and take images or scan the bed plate.

KimmoHop, 2023-11-11:
I'd like comments from @tombrazier what I should be looking at :)

In this picture (sweep to 50 Hz for Y), at 1 there starts to be a little shifting, at 2 it's getting bad and at 3 it's totally bonkers. Where is resonant frequency?
[image]

In this picture (sweep to 70 Hz for X), at 4 there starts to be a little shifting (maybe a bit earlier?), at 5 it's getting bad and at 6 control is lost. ?
[image]

Here is partial shot of page. I'll create draft PR to MarlinDucumentation and we can add images later :)
[image]

tombrazier, 2023-11-11:
> Edit: hurdle # 1 passed: Jekyll running under WSL :)

I recognise that feeling of victory from my own work with Marlin docs!

> @tombrazier & @revetuzo - if you have taken images from your calibration patterns, that would help

Here's a picture of my X and Y axes being tested. You can clearly see where the amplitude is higher:

[image]

Here is my X axis going through a damping factor scan after setting the X frequency:

[image]

In your images it isn't completely clear to me where the resonant frequencies are. I think towards the right hand side of the Y axis scan, beyond 3, there is an inertia effect that is causing the print head to get closer and further from the bed. It does also look like the amplitude gets greater here but you can't see it clearly because of the inertia effect. I have outlined what looks like the envelope of the oscillations to make it clearer.

[image]

What kind of printer do you have? i.e. bedslinger, delta, core, etc?

The page that generates the pattern looks great. As to where to put it, I think an Input Shaping specific page under Help -> Features makes sense but I also vaguely remember @thinkyhead saying was something like he wanted to move away from those kinds of pages or refactor them or something.

tombrazier, 2023-11-11:
Just checking, @KimmoHop, was IS disabled when you ran your scans? It should have been.

tombrazier, 2023-11-12:
Thinking about it, I wonder if it is possible to improve the printing past point 3 on the Y pattern. If you try a higher layer height then the inertial effect may be less prominent and a lower amplitude will reduce the strength of the inertial effects.

KimmoHop, 2023-11-12:
@tombrazier how do you want to be attributed?

Yes, IS is disabled with M593 F0 for resonant frequency sweep.

My printer is Voxelab Aquila (Ender 3 v2 clone) with Orbiter 2 direct extruder and magnetic PEI sheet instead of glass.
Some of the problems could be
- Octoprint @ 115kbps, I have to try printing from SD
- Creality heat sink may have hard time keeping volcano heating block in control. I have ordered V6 heating block and nozzles (shorter and lighter) + maybe there could be support for the heatsink?
- Filament spool at top, but klipper bedslingers are the same
- Print speed, though Volcano should handle it, might benefit from increasing temperature

tombrazier, 2023-11-13:
> @tombrazier how do you want to be attributed?

Do you mean in the web tool you're building? Just @tombrazier will do, maybe with a link to my github profile.

I think your printer is accelerating correctly, which is why the printing at the end of the line goes wild. You can see the same effect in my calibration patterns, the line is a lot messier at higher frequencies. So for the Y calibration line, for example, the whole bed is probably tilting a little bit under the high acceleration and so the surface is getting closer to / further from the nozzle. This is why I suggest reducing the amplitude of the oscillations and increasing the distance to the bed.

KimmoHop, 2023-11-22:
Well... I changed hotend (probably doesn't matter much but V6 heatsink is much easier to cool than Creality non-Neo) and tightened POM wheels to "internet standard" (wheels are not easy to rotate against each other). The bed was actually pretty loose, though moved very smoothly. Now the jumping in Y pattern is pretty much gone :)

I also increased layer height to 0.30. This is Y frequency pattern, which shows how amplitude increases. I settled for 34Hz on Y. Should it actually be higher?
[image]

Damping patterns are harder. This is for Y. Right top has increased amplitude. Green marks what looks like lower amplitude? I took 5th line (0.2) loosely based on that. I'm not really sure what to look for or if frequency was right or if printer is resonating too much at multiple frequencies :/
[image]

Anyway I think I got improvement in print speed _and_ quality ;)
On the left is new calibration cube, printed with IS, 2.5k acceleration and 60 mm/s for outer walls + solids / 120 mm/s for inner walls & infill. LA was a bit too high. IMO it came out quite nice :)
On the right is some older cube, it may have guessed IS frequencies and default damping, no more than 2k acceleration and most likely 45/80 mm/s print speed.
New one has less ringing and at lower frequency. So my IS settings probably are still a bit off, but better.
[image]
[image]

KimmoHop, 2023-11-23:
After thinking this a bit, and asking from ChatGPT (I don't know control theory), I think both my interpretations and values are wrong. Correct me, but am I not looking for
- resonant frequency where amplitude has reached it's peak value?
    - so for my printer Y axis it should be ~41Hz (81mm)
- damping factor is best, where amplitude from 0 to resonant frequency is fairly constant, and attenuates above that?
    - my Y freqency is off, so there is no clear indication

After next run, I'll try to scan the sheet, or use proper camera :)

tombrazier, 2023-11-24:
Yes, you are looking for the maximum amplitude. For Y that is about 80mm. Damping is best when the entire line has fairly constant amplitude. So, yes, if resonant frequency is not set correctly the damping pattern won't tell you much.

Your photo quality is great as it is.

tombrazier, 2023-12-11:
Many thanks to @KimmoHop for implementing a pattern generator in the Marlin docs. Closing this FR.

KimmoHop, 2023-12-11:
> Many thanks to @KimmoHop for implementing a pattern generator in the Marlin docs. Closing this FR.

It was quite fun, actually, only drawback being my stubborn printer not shakin' in the best way :D