MarlinFirmware/Marlin #26168: [BUG] MKS Monster V2 and Mini_12864 v3 does not work without onboard SD in

jesterhead82, 2023-08-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Im using todays Bugfix 2.1.x

MKS Monster V2
MKS Mini12864 V3

default_envs = mks_monster8
#define SDCARD_CONNECTION LCD
//#define USB_FLASH_DRIVE_SUPPORT

The Display does not come up if there is no SDcard in the onboard slot, independent of the LCD card slot.
If I put SD card onboard and restart, display comes up and the displays SD slot work, too.

### Expected behavior

Display comes up without onboard SD in.

### Actual behavior

Display does not come up with onboard SD in.

### Version of Marlin Firmware

https://github.com/MarlinFirmware/Marlin/commit/2bac7835e801c413c3d9e75a1dbe3e94dc689f3d

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/12303450/Configuration_adv.zip)

rondlh, 2023-08-09:
I'm not sure what's happening there...
From the pinout both onboard and EXT1/2 ports are completely independent. 
The Mini12864  always has power right (check the backlight)?

You can try this:
default_envs = mks_monster8_usb_flash_drive_msc
Then the onboard micro SD card can be mounted via a USB C cable, and you have serial access too. (Note, you might need to unplug the USB cable a few times).
or
;default_envs = mks_monster8_usb_flash_drive
This one can mount the micro SD card, but no serial port access.

jesterhead82, 2023-08-09:
I tried already with mks_monster8_usb_flash_drive, same behavior.

The backlight stays off if I do a power cycle.

If I start with onboard SD, unplug SD onboard and reboot using reset switch, backlight stays in the last state (color).

closed 2023-10-19 by github-actions[bot]

jesterhead82, 2023-11-28:
was trying todays Bugfix, and somehow the issue is gone...

jesterhead82, 2023-12-10:
Na, its not gone ;) I had a SD in the Displays slot...