MarlinFirmware/Marlin #26169: [BUG] `HAS_GCODE_PREVIEW` thumbnail preview not working

classicrocker883, 2023-08-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

with `HAS_GCODE_PREVIEW` it doesn't popup to show the thumbnail of the model preview.

also, this is the 2nd part to the issue, when it was working, it only allowed selecting 1 or 2 previews at a time before freezing/rebooting.

### Expected behavior

suppose to show a popup with a 3D rendering image of the gcode image.  
asking to confirm to print or cancel.

### Actual behavior

just goes straight to print, no popup to preview and ask to confirm or cancel

### Steps to Reproduce

1. Enable `DWIN_LCD_PROUI`
2. make sure `HAS_GCODE_PREVIEW` is enabled
3. flash onto board
4. go to Print menu
5. select file with known preview image

### Version of Marlin Firmware

current bugfix

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin-Configs.zip](https://github.com/MarlinFirmware/Marlin/files/12304354/Marlin-Configs.zip)

when I was able to have this working, there was also an issue when selecting the thumbnail preview file again.

basically anytime you go to select another or the same gcode file with preview, it would freeze/crash then reboot automatically.

a way around this was to after seeing the thumbnail - go to another file that did not have one or simply go back to another folder directory. once there it basically refreshes itself and you can select another file to preview.

I noticed that files which had larger thumbnail "cache" or whatever you want to call it - whatever was being written to the FLASH memory - or wherever its store (SRAM?) - you could only select these files once or twice in a row before it freezes/reboots.

files with smaller image sizes (which also load the preview faster) can be loaded several times 3 or 4 in a row before the issue happens.

if there is a way to clear the cache or memory after you "Cancel"/ not confirm the file to print, inorder to view another file, that would fix that issue.

i've tried a few things like
```y
   { fileprop.thumbsize = 0; Preview_Invalidate(); delete[] fileprop.thumbdata; fileprop.clear(); card.closefile(); return }
```
ect... dont know where to go from there.

mriscoc, 2023-08-10:
```cpp
delete[] fileprop.thumbdata;
```
Should clear all thumbnail data. Be sure that your thumbnail does not exceed 10kB with the current implementation.

classicrocker883, 2023-08-10:
> ```c++
> delete[] fileprop.thumbdata;
> ```
> 
> Should clear all thumbnail data. Be sure that your thumbnail does not exceed 10kB with the current implementation.

sorry,  
but where can this be used?
```y
  void onClickConfirmToPrint() {
    dwinResetStatusLine();
    if (hmiFlag.select_flag) {     // Confirm
      gotoMainMenu();
      return card.openAndPrintFile(card.filename);
    }
    else {
      delete[] fileprop.thumbdata;  // added here
      hmiReturnScreen();
    }
  }
```
I tried here, yet doesn't help. i may have mentioned a quick temporary fix is to select another file without a thumbnail so it says  `Thumbnail not found` or to another folder directory and it resets or clears the thumbnail stored memory automatically.

another note to make is should `gotoMainMenu();` be part of the **Confirm** button? because in my testing, I omitted that and there was no difference.

classicrocker883, 2023-08-10:
unrelated:  
I hope you see the other comment and understand I wasnt trying to copy something that was yours, I respect you and any one else to not do that. so sorry if it came off that way.

classicrocker883, 2023-08-11:
I also tried
```y
return delete[] fileprop.thumbdata, 
      hmiReturnScreen(); 
```
maybe I have it in the wrong spot?

I would assume this is correct, it is where it selects to print or to cancel on the display.

I will say removing `gotoMainMenu();` did not make a difference. I am just wondering why is this function here when return is to go ahead and print file.  
maybe it was supposed to be something like redrawDash?

HAS_GCODE_PREVIEW enabled still doesn't work within this Marlin fork. I have to test using the Ender3V2S1 fork. the files and code is the same mostly except for PROUI_EX.

closed 2023-10-22 by github-actions[bot]