MarlinFirmware/Marlin #26171: [BUG] SERIAL_FLOAT_PRECISION no longer works as intened

ellensp, 2023-08-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

From https://github.com/MarlinFirmware/Marlin/pull/18367
### Description
Introduces DEFAULT_PRINT_FLOAT_PRECISION

### Benefits
Allows increasing the number of fractional digits when printing floats and doubles to serial port.
Effectively increasing the reported precision of probing operations(G30, G38, M48) and configs (M503).

This no longer works.

### Bug Timeline

Unknown

### Expected behavior

increasing DEFAULT_PRINT_FLOAT_PRECISION > 2 should increase the precision of G30, G38 and M48 and M503

### Actual behavior

It does not.

### Steps to Reproduce

1 enable a probe (I use FIX_MOUNTED_PROBE as it work in the sim)
2 enable Z_SAFE_HOMING
3 define DEFAULT_PRINT_FLOAT_PRECISION 4
4 build (I build for BOARD_SIMULATED)
5 Do  G28
6 Do a G30

these do no show increased precision 

### Version of Marlin Firmware

Bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

GMagician, 2023-08-12:
All these function use fixed precision and G30/M48 (I just checked them) use different precision depending on value shown.
May be a solution (extrapolated from M48) to do something like:
`p_float_t(min, _MAX(3, SERIAL_FLOAT_PRECISION))`?

thephantom1492, 2023-08-12:
Using a nightly snapshot from 2023-05-14, SERIAL_FLOAT_PRECISION is working.
The precision of 2 break the corner leveling of BTT TFT screens (atleast TFT70, >1 year and yesterday firmware) as it require 4 digits.
At 2 digits the TFT70 just display 0.0000 when you try to level the corners.

#define SERIAL_FLOAT_PRECISION 4
Recv: X:43.0000 Y:43.0000 Z:5.0000 E:0.0000 Count X:13760 Y:13760 Z:32000

#define SERIAL_FLOAT_PRECISION 6
Recv: X:360.000000 Y:360.000000 Z:0.000000 E:0.000000 Count X:115200 Y:115200 Z:0

ellensp, 2023-08-13:
was broken in this commit 2ef71c6ebaa1f1b496eb369879fe22c69cf9adb4  on Jun 13

thinkyhead, 2023-08-19:
`SERIAL_FLOAT_PRECISION` applies in some places but may not apply everywhere. Please submit PRs to patch the serial reports where it is currently supposed to apply but doesn't.

The issue with some BTT serial displays requiring exactly 4 digits is a BTT display bug. They will need to fix the firmware for these displays to handle any number of digits. Meanwhile, we can add a hack as a custom flag to ensure 4 digits is sent in all reports where this display requires 4 digits, or to assert that `SERIAL_FLOAT_PRECISION` must be set to 4 whenever one of these displays is enabled in the configuration.

thephantom1492, 2023-08-19:
While I do agree that the BTT screen does have a bug, I find it more than ordinary that the SERIAL_FLOAT_PRECISION got removed and fixed to 2 positions. The returned position precision should have been configurable IMO as it was before.

thisiskeithb, 2023-08-19:
> The issue with some BTT serial displays requiring exactly 4 digits is a BTT display bug. They will need to fix the firmware for these displays to handle any number of digits. Meanwhile, we can add a [...]

There is no display option to enable when connecting these serial displays...via serial. BTT has instructions in their TFT firmware readme to set `SERIAL_FLOAT_PRECISION` in Marlin to `4`, which used to work fine until setting `SERIAL_FLOAT_PRECISION` no longer worked as intended since it now sets it to 2 or less no matter what you have set in your config thanks to:

https://github.com/MarlinFirmware/Marlin/blob/f7d5188b2f0a59b752fbabf6b96ff1ed4aed8112/Marlin/src/module/temperature.cpp#L4223

And of course https://github.com/MarlinFirmware/Marlin/commit/2ef71c6ebaa1f1b496eb369879fe22c69cf9adb4 broke things further.

thinkyhead, 2023-09-09:
> There is no display option to enable when connecting these serial displays

The code in `print_heater_state` has been emitting no more than 2 digits for temperature precision since #20602 published in Jan 2021 (so if they were aware or the issue, BTT has had since Dec 2020 to patch their TFT firmware). In the meantime we can provide an obscure option to override the maximum use of 2 digits in `print_heater_state`, which will mostly only apply as a workaround for BTT TFT. #26253

ellensp, 2023-09-10:
The feature needs to come back. We should not expect everyone else to update their code because it was accidentally omitted/broken. I am presuming it was accidentally removed as the configuration option remains.

closed 2023-10-03 by thinkyhead