MarlinFirmware/Marlin #26179: [FR] Generate update.cbd and firware.bin files for Tronxy X5SA CXY-V6-191017 Chitu_V6

Plum6767, 2023-08-15:
### Is your feature request related to a problem? Please describe.

First time trying to compile firmware so may be doing this incorrectly - Sorry in advance

When compiling Marlin 2.1.2.1 for my Tronxy X5SA pro with CXY-V6-191017 motherboard and using the Chitu-V6 board within Marlin, it all compiles successfully, but it generates an update.cbd file at the end.

I have only had the machine about a month and got Tronxy to send me the latest version of their version of Marlin (which is way out of date), and that update successfully but was a fmw_x5sa_pro.bin, file.

It looks like later versions of this printer don't like *.cbd files but do like *.bin files.

When I compiled the firmware I see from the text generated that a firmware.bin file is created, then encrypted to an update.cbd file. then the firmware.bin file seems to get deleted as I can't find it anywhere.

Why can't it not generate both firmware.bin and update.cbd, then whichever is appropriate can be used, or if not can you document how to stop the compilation at the firmware.bin stage

### Are you looking for hardware support?

No

### Describe the feature you want

Can firmware.bin and update.cbd files be generated so appropriate one can be used.

Or if not document how firmware.bin can be generated instead of update.cbd

ellensp, 2023-08-16:
please attach fmw_x5sa_pro.bin, but I suspect it will be encrypted

Plum6767, 2023-08-16:
[Tronxy.zip](https://github.com/MarlinFirmware/Marlin/files/12356316/Tronxy.zip)
.zip file contains everything in the folder that Tronxy sent to update the firmware on my machine, including fmw_x5sa_pro.bin.

Thanks for the assistance.

suportaries, 2023-12-06:
hola tengo el mismo problema como lo has solucionado?

SlackHacky, 2024-01-14:
all about bootloader version - early bootloaders  use a single update.cdb file, later use a folder on sdcard with a .bin in it not sure when it changed (around v1.56?) but i flashed a v1.63 bootloader to my 191017 board and it updates via folder and not .cdb file now - so the later bootloader (v1.63) works on both f103 and f446 mcu boards

inspectionsbybob, 2024-07-06:
I have tried putting the Update.cbd in the root, and in a folder called "Update" and nothing worked.

thinkyhead, 2024-08-14:
@SlackHacky — Does it work to just change `MOTHERBOARD` to `BOARD_TRONXY_CXY_446_V10` and then flash using the `fmw_tronxy.bin` file, or is there more involved?

inspectionsbybob, 2024-08-14:
I didn't see that motherboard in my boards.h file, I will re-download and
try again.

-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

closed 2024-09-07 by thinkyhead