MarlinFirmware/Marlin #26180: [BUG] If you launch a bed tramming just after the z-probe wizard, the bltouch doesn't retract

SGr33n, 2023-08-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi!
Marlin 2.1.2.1, on an Ender 3 / SKR E3 Mini v2 / BL Touch v2.
If you launch a bed tramming just after the z-probe wizard, the bl touch doesn't retract, so when you press on "Next" the bl touch collides with the bed. Fortunately the tip of the bl touch is blunt so it doesn't break.

Thanks!

### Steps to Reproduce

1. Do a Z-Probe Wizard
2. Launch a bed tramming
3. Click on "Next": Warning, the bl touch will collide with the bed.

### Version of Marlin Firmware

2.1.2.1

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/12363940/Configuration.zip)

classicrocker883, 2023-08-19:
you manually deployed the probe. either "Stow" the probe after, or do not hit "Deploy".

otherwise maybe someone could make a failsafe inside the code, but its not a concern.

SGr33n, 2023-08-19:
Thanks! So my fault, I misunderstood the deploy command.

closed 2023-08-19 by SGr33n