MarlinFirmware/Marlin #26182: [BUG] Error compiling with SDSUPPORT enabled (STM32F103RC)

raillpl, 2023-08-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Firmware failed to compile with SDSUPPORT turned on, otherwise compile successfully.
Using lattest Vcode and PIO.

### Version of Marlin Firmware

Latest NIGHTLY

### Printer model

Own build

### Electronics

BTT SKM Mini E3 V2

### Add-ons

MKS Mini 12864 v3

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Error log:
 *  Executing task: C:\Users\d9\.platformio\penv\Scripts\platformio.exe run --environment STM32F103RC_btt_USB 

Processing STM32F103RC_btt_USB (board: genericSTM32F103RC; platform: ststm32@~12.1; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
Copying variant MARLIN_F103Rx to framework directory...
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 (12.1.1) > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 @ 0.0.0
 - framework-cmsis @ 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 15 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SPI @ 1.0
|-- SoftwareSerial @ 1.0
|-- Wire @ 1.0
|-- TMCStepper @ 0.7.3
|-- Adafruit NeoPixel @ 1.8.7
|-- U8glib-HAL @ 0.5.2
|-- SlowSoftI2CMaster @ 1.0.0
|-- SlowSoftWire @ 0.0.0+20230817020129
|-- IWatchdog @ 1.0.0
|-- EEPROM @ 2.0.1
|-- Servo @ 1.1.2
Building in release mode
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\HAL.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\HAL_SPI.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\MarlinSPI.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\MarlinSerial.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\MinSerial.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\Servo.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\eeprom_bl24cxx.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\eeprom_flash.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\eeprom_if_iic.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\eeprom_sdcard.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\eeprom_sram.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\eeprom_wired.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\fast_pwm.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\fastio.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\msc_sd.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\sdio.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\timers.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\usb_host.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\usb_serial.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\shared\Delay.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\shared\HAL.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\shared\MinSerial.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\shared\backtrace\backtrace.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\shared\eeprom_api.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\shared\eeprom_if_i2c.cpp.o
Compiling .pio\build\STM32F103RC_btt_USB\src\src\HAL\shared\esp_wifi.cpp.o
Marlin\src\HAL\STM32\msc_sd.cpp: In member function 'virtual bool Sd2CardUSBMscHandler::Write(uint8_t*, uint32_t, uint16_t)':
Marlin\src\HAL\STM32\msc_sd.cpp:74:34: error: no matching function for call to 'DiskIODriver::writeStart(uint32_t&)'
   74 |       sd2card->writeStart(blkAddr);
      |                                  ^
In file included from Marlin\src\HAL\STM32\../../sd/usb_flashdrive/Sd2Card_FlashDrive.h:30,
                 from Marlin\src\HAL\STM32\../../sd/SdVolume.h:38,
                 from Marlin\src\HAL\STM32\../../sd/SdBaseFile.h:34,
                 from Marlin\src\HAL\STM32\../../sd/SdFile.h:33,
                 from Marlin\src\HAL\STM32\../../sd/cardreader.h:43,
                 from Marlin\src\HAL\STM32\msc_sd.cpp:28:
Marlin\src\HAL\STM32\../../sd/usb_flashdrive/../disk_io_driver.h:56:16: note: candidate: 'virtual bool DiskIODriver::writeStart(uint32_t, uint32_t)'
   56 |   virtual bool writeStart(const uint32_t block, const uint32_t) = 0;
      |                ^~~~~~~~~~
Marlin\src\HAL\STM32\../../sd/usb_flashdrive/../disk_io_driver.h:56:16: note:   candidate expects 2 arguments, 1 provided
Marlin\src\HAL\STM32\msc_sd.cpp:76:12: error: 'uint32' was not declared in this scope; did you mean 'uint'?
   76 |       for (uint32 i = blkLen; i--;) {
      |            ^~~~~~
      |            uint
Marlin\src\HAL\STM32\msc_sd.cpp:76:31: error: 'i' was not declared in this scope
   76 |       for (uint32 i = blkLen; i--;) {
      |                               ^
Marlin\src\HAL\STM32\msc_sd.cpp: In member function 'virtual bool Sd2CardUSBMscHandler::Read(uint8_t*, uint32_t, uint16_t)':
Marlin\src\HAL\STM32\msc_sd.cpp:106:12: error: 'uint32' was not declared in this scope; did you mean 'uint'?
  106 |       for (uint32 i = blkLen; i--;) {
      |            ^~~~~~
      |            uint
Marlin\src\HAL\STM32\msc_sd.cpp:106:31: error: 'i' was not declared in this scope
  106 |       for (uint32 i = blkLen; i--;) {
      |                               ^
*** [.pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\msc_sd.cpp.o] Error 1
======================================================= [FAILED] Took 51.60 seconds =======================================================

Environment          Status    Duration
-------------------  --------  ------------
STM32F103RC_btt_USB  FAILED    00:00:51.602
================================================== 1 failed, 0 succeeded in 00:00:51.602 ================================================== 

 *  The terminal process "C:\Users\d9\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'STM32F103RC_btt_USB'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

jmz52, 2023-08-18:
It was PR https://github.com/MarlinFirmware/Marlin/pull/25340 broke the code, but your `Configuration_adv.h` file seems outdated and incorrectly configured.

Outdated part is
```
    #define SDSORT_LIMIT       256     // Maximum number of sorted items (10-256). Costs 27 bytes each.
    #define FOLDER_SORTING     -1     // -1=above  0=none  1=below
    #define SDSORT_GCODE       false  // Allow turning sorting on/off with LCD and M34 G-code.
```
when it should be 
```
    #define SDSORT_REVERSE     false  // Default to sorting file names in reverse order.
    #define SDSORT_LIMIT       40     // Maximum number of sorted items (10-256). Costs 27 bytes each.
    #define SDSORT_FOLDERS     -1     // -1=above  0=none  1=below
    #define SDSORT_GCODE       false  // Enable G-code M34 to set sorting behaviors: M34 S<-1|0|1> F<-1|0|1>
```
`FOLDER_SORTING` option was renamed and now causes compilation error.

Incorrectly configured part is
```
  #define USB_FLASH_DRIVE_SUPPORT
```
This option should not be used unless you are using external MAX3421E board.
And if you do use this external board, then you need to configure pins this board is connected to.

Once these two configuration issues are addressed you need to revert commit https://github.com/MarlinFirmware/Marlin/commit/6af6060aa0240ccc03020edba661fe42125d3713 to be able to compile code once again.

ellensp, 2023-08-18:
User misunderstanding of USB_FLASH_DRIVE_SUPPORT.

jmz52, 2023-08-18:
@ellensp I was able to reproduce reported problem without `USB_FLASH_DRIVE_SUPPORT` option and pinpoint it to exact commit that breaks the code in Marlin\src\HAL\STM32\msc_sd.cpp - https://github.com/MarlinFirmware/Marlin/commit/6af6060aa0240ccc03020edba661fe42125d3713

raillpl, 2023-08-18:
Thanx guys for effort, with USB_FLASH_DRIVE_SUPPORT and SDCARD_SORT_ALPHA both disabled, there is another error at same poin:
`marlin\src\HAL\STM32\msc_sd.cpp: In member function 'virtual bool Sd2CardUSBMscHandler::Write(uint8_t*, uint32_t, uint16_t)':
marlin\src\HAL\STM32\msc_sd.cpp:74:34: error: no matching function for call to 'DiskIODriver::writeStart(uint32_t&)'
   74 |       sd2card->writeStart(blkAddr);
      |                                  ^
In file included from marlin\src\HAL\STM32\../../sd/Sd2Card.h:38,
                 from marlin\src\HAL\STM32\../../sd/SdVolume.h:44,
                 from marlin\src\HAL\STM32\../../sd/SdBaseFile.h:34,
                 from marlin\src\HAL\STM32\../../sd/SdFile.h:33,
                 from marlin\src\HAL\STM32\../../sd/cardreader.h:43,
                 from marlin\src\HAL\STM32\msc_sd.cpp:28:
marlin\src\HAL\STM32\../../sd/disk_io_driver.h:56:16: note: candidate: 'virtual bool DiskIODriver::writeStart(uint32_t, uint32_t)'
   56 |   virtual bool writeStart(const uint32_t block, const uint32_t) = 0;
      |                ^~~~~~~~~~
marlin\src\HAL\STM32\../../sd/disk_io_driver.h:56:16: note:   candidate expects 2 arguments, 1 provided
marlin\src\HAL\STM32\msc_sd.cpp:76:12: error: 'uint32' was not declared in this scope; did you mean 'uint'?
   76 |       for (uint32 i = blkLen; i--;) {
      |            ^~~~~~
      |            uint
marlin\src\HAL\STM32\msc_sd.cpp:76:31: error: 'i' was not declared in this scope
   76 |       for (uint32 i = blkLen; i--;) {
      |                               ^
marlin\src\HAL\STM32\msc_sd.cpp: In member function 'virtual bool Sd2CardUSBMscHandler::Read(uint8_t*, uint32_t, uint16_t)':
marlin\src\HAL\STM32\msc_sd.cpp:106:12: error: 'uint32' was not declared in this scope; did you mean 'uint'?
  106 |       for (uint32 i = blkLen; i--;) {
      |            ^~~~~~
      |            uint
marlin\src\HAL\STM32\msc_sd.cpp:106:31: error: 'i' was not declared in this scope
  106 |       for (uint32 i = blkLen; i--;) {
      |                               ^
*** [.pio\build\STM32F103RC_btt_USB\src\src\HAL\STM32\msc_sd.cpp.o] Error 1`

raillpl, 2023-08-18:
I am not C programmer, and maybe it is not correct, but this made building process to pass ok for me:

marlin\src\HAL\STM32\msc_sd.cpp:

line 74:       sd2card->writeStart(blkAddr, blkLen);
line 76:       for (uint i = blkLen; i--;) {
line 106:     for (uint i = blkLen; i--;) {

Basically added second argument to writeStart() function and changed "i" variable delaration from uint32 to uint following log errors.

classicrocker883, 2023-08-19:
I was able to compile successfully 
>without `USB_FLASH_DRIVE_SUPPORT` option 
 using the provided Configuration files, all i did was disable USB_FLASH_DRIVE_SUPPORT
 
 I dont see how anyone was able to get the above errors or why **msc_sd.cpp** being changed worked any different.

did i do something wrong here?

narno2202, 2023-08-23:
Same problem with MKS Monster 8 and trying to compile mks_monster8_usb_flash_drive_msc. I also changed the faulty `sd2card->writeStart(blkAddr)` and compilation went fine.

github4657, 2023-09-16:
I found this same error after loading bugfix on sept 14 2023
Using board BOARD_BTT_SKR_V2_0_REV_B
#define USB_FLASH_DRIVE_SUPPORT was commented out.

Reported error at line 76 "for (uint32 i = blkLen; i--;) {"

Line 62 had "bool Write(uint8_t *pBuf, uint16_t blkAddr, uint16_t blkLen) {"

Looking at code blkLen was sometimes 16 bit other times it was 32 bits

Changing line 76 to  "for (uint16 i = blkLen; i--;) {" got rid of the error

gogeekness, 2023-11-10:
Hello,  I am getting a similar compile error when I try to build Marlin.
I am running a Big Tree SKR Mini E3 v3.0 with a  Big Tree TFT35-E3 V3.01 touchscreen.

I have upgraded my Tronxy 5ea Pro with this board and it has been a hard process.
Recently, I made some basic prints and compiled working firmware.

To track all of the changes to the config files I moved them to GitHub and cloned Marlin to rebuild the firmware.
I cloned to the latest [bugfix 2.1.2.1

```
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:51,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:27:
Marlin\src\HAL\STM32\../../inc/Changes.h:665:4: error: #error "FOLDER_SORTING is now SDSORT_FOLDERS."
  665 |   #error "FOLDER_SORTING is now SDSORT_FOLDERS."
```

```
====================================================== 1 failed, 0 succeeded in 00:00:13.774 =========================================================
 *  The terminal process "C:\Users\rdese\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'STM32G0B1RE_btt'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
 *  Executing task in folder Marlin: C:\Users\rdese\.platformio\penv\Scripts\platformio.exe run --environment STM32G0B1RE_btt 
```

I see the previous fixes in the files, but I don't know what is triggering this error.  

The error (for me) starts here: Commit: b5a08c251853e0b7122bc33d1c6fcd3de967e478    (Aug 22, 2012)
Before this commit, I can compile to completion without any errors.

I will build on the next earliest revision.

ellensp, 2023-11-10:
@gogeekness your issue is completely unrelated to this issue

The error is clear 

#error "FOLDER_SORTING is now SDSORT_FOLDERS."

You have an out dated Configuration_adv.h file that contains     #define FOLDER_SORTING vs the needed SDSORT_FOLDERS

See Example
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/Configuration_adv.h#L1753