MarlinFirmware/Marlin #26183: [BUG]  Compiles fine without Input shaping enables with only 12.2% or RAM used, but fails due to lack of memory when Input Shaping is enabled.

andy-spoo, 2023-08-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have a working Marlin 2.1.2.1 which compiles just fine which has input shaping disabled. 
Compiled using Visual Studio, Platformio and Auto Build Marlin.
Board is a Bigtreetech Mini E3 V3.0
This good build reports = 

RAM:   [=         ]  12.2% (used 18036 bytes from 147456 bytes)
Flash: [=====     ]  50.0% (used 262084 bytes from 524288 bytes)
Building .pio/build/STM32G0B1RE_btt/firmware.bin
============================================================================================== [SUCCESS] Took 111.74 seconds ==============================================================================================

Environment      Status    Duration
---------------  --------  ------------
STM32G0B1RE_btt  SUCCESS   00:01:51.736

When I enable Input Shaping =

#define INPUT_SHAPING_X
#define INPUT_SHAPING_Y

and compile, it get this error =

Linking .pio/build/STM32G0B1RE_btt/firmware.elf
/home/andy/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/STM32G0B1RE_btt/firmware.elf section `.bss' will not fit in region `RAM'
/home/andy/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: region `RAM' overflowed by 192248 bytes
collect2: error: ld returned 1 exit status
*** [.pio/build/STM32G0B1RE_btt/firmware.elf] Error 1
=============================================================================================== [FAILED] Took 14.21 seconds ===============================================================================================

Environment      Status    Duration
---------------  --------  ------------
STM32G0B1RE_btt  FAILED    00:00:14.205

Surely there must be way more RAM available than required?? Going from 12.2% to fail for lack of memory seems excessive.
I've ran 'CLEAN' and 'FULL CLEAN' a few times and restarted Visual Studio code / Platformio a few times and still get this error.

Any ideas?

### Bug Timeline

New. Started on enabling input shaper

### Expected behavior

To compile correctly as it should have bags of memory available.

### Actual behavior

Fails with    =  firmware.elf section `.bss' will not fit in region `RAM'

### Steps to Reproduce

1. Uncomment #define INPUT_SHAPING_X and #define INPUT_SHAPING_Y
2. Compile.

### Version of Marlin Firmware

2.1.2.1

### Printer model

Creality ender 5 Plus

### Electronics

Bigtreetech Mini E3 V3.0

### Add-ons

TFT35 V3

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Ender 5 Plus.zip](https://github.com/MarlinFirmware/Marlin/files/12372795/Ender.5.Plus.zip)

ellensp, 2023-08-17:
The line #define DEFAULT_MAX_FEEDRATE          { 8000, 8000, 15, 15000 }
also sets the size of the memory buffers used. lower these vales to fit in the hardware limitations of your controller

closed 2023-08-17 by ellensp

andy-spoo, 2023-08-17:
OK. Thanks for your help. Greatly appreciated.