MarlinFirmware/Marlin #26184: [BUG] Cannot use BED_TRAMMING_USE_PROBE and NOZZLE_TO_PROBE_OFFSET at the same time

ADeliciousMan, 2023-08-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Unable to use BED_TRAMMING_USE_PROBE and NOZZLE_TO_PROBE_OFFSET at the same time, not sure if this is intended behavior. 

### Bug Timeline

Just noticed it, but no previous experience with Marlin either

### Expected behavior

I accepted to be able to have Tramming use the bed, while also having the nozzle offset after a homing.

### Steps to Reproduce

Turn on NOZZLE_TO_PROBE_OFFSET and BED_TRAMMING_USE_PROBE

### Version of Marlin Firmware

2.1.2.1

### Printer model

Ender 3 4.3.7 Board 

### Electronics

4.3.7 Board 

### Add-ons

CR Touch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/12375709/configs.zip)

ellensp, 2023-08-19:
The error is 

`Marlin/src/lcd/menu/menu_bed_corners.cpp:157:79: error: static assertion failed: BED_TRAMMING_INSET_LFRB right inset is not reachable with the default NOZZLE_TO_PROBE offset and PROBING_MARGIN.
`
The error is a deliberate error to stop you configuring  your machine with impossible settings

In Your Configuration files you have set 
```
#define NOZZLE_TO_PROBE_OFFSET { -44, -7, 0 }
#define BED_TRAMMING_INSET_LFRB { 30, 30, 30, 30 } // (mm) Left, Front, Right, Back insets
```

Your configuration has told marlin that the furthermost right position it can probe is 44 mm from the right edge, 33 is outside of that area.

You need  #define BED_TRAMMING_INSET_LFRB { 30, 30, 44, 30 }

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2023-08-19 by ellensp

ADeliciousMan, 2023-08-19:
Ah thank you, I'm not familiar enough for Marlin yet to know the difference between it being a bug and it stopping me from making a mistake.
Appreciate the insight.