MarlinFirmware/Marlin #26187: [BUG]

IlyaFarm, 2023-08-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

At the end of printing from SD card Printer DO things which are not in G-Code.
for example it can extrude plastic, make retraction/deretraction make moves like it make print.

### Bug Timeline

May be new, but I don't thik so.

### Version of Marlin Firmware

Marlin 2.1.2.1

### Printer model

TEVO TARANTULA

### Electronics

BOARD: BTT Octopus V1.1; DRIVERS: TMC2208

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I've seen this 3 times so far.
erlier I was think it's problem with my SD Card, or errors which occure's wher printer read G-Code from SD, But now I think it's problem with G-Code parser in Marlin.

ellensp, 2023-08-19:
Please provide:
The configuration files
The example gcode used 
video of what it is doing incorrectly with that gcode

IlyaFarm, 2023-08-19:
Done

IlyaFarm, 2023-09-10:
I think it's after all my cheap SD card, cause if I format it before write new files, everything OK. If not, somthing can go wrong.

thisiskeithb, 2023-09-10:
> I think it's after all my cheap SD card, cause if I format it before write new files, everything OK.  

Quality SD cards are an important part of the process 👍

closed 2023-09-10 by thisiskeithb

IlyaFarm, 2023-09-17:
Nope problem was not in SD card quality, cause I change SD card source from board to LSD where I use my older good SD card and today it happened again. The printer parked and turned off the print head, before finishing printing.