MarlinFirmware/Marlin #26188: [BUG] Compilation error with DWIN_CREALITY_LCD_JYERSUI (error: 'MSG_MAIN_MENU' is not a member of 'Language_en'; )

stoubian, 2023-08-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

With DWIN_CREALITY_LCD_JYERSUI defined,
When compiling in VSCode with PIO,
The compiler throws the error: 'MSG_MAIN_MENU' is not a member of 'Language_en'; did you mean 'MSG_MMU2_MENU'?

The fix proposed on reddit by nanmetal worked fine after test.
[link here](https://www.reddit.com/r/MarlinFirmware/comments/14zk3uz/trying_to_compile_marline_2121_on_mac_fatals_need/?rdt=62548)

I make a PR asap

### Bug Timeline

new

### Expected behavior

I expect the code compiles without error.

### Actual behavior

The compiler throws the error: 'MSG_MAIN_MENU' is not a member of 'Language_en'; did you mean 'MSG_MMU2_MENU'?

### Steps to Reproduce

1. take the example default configuration.h file or any configuration.h file
2. uncomment DWIN_CREALITY_LCD_JYERSUI 
3. compile
4. error

### Version of Marlin Firmware

2.1.2.1

### Printer model

Creality Ender-3 V2

### Electronics

Stock electronics

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2023-08-19:
Are you using a fork? `MSG_MAIN_MENU` is in the language file(s):

https://github.com/MarlinFirmware/Marlin/blob/1d17c34e123f74d820f42fcd179104c884719007/Marlin/src/lcd/language/language_en.h#L76

Using the [stock Ender-3 V2 Creality UI config](https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x/config/examples/Creality/Ender-3%20V2/CrealityV422/CrealityUI) and enabling `DWIN_CREALITY_LCD_JYERSUI` instead of `DWIN_CREALITY_LCD` works fine.

closed 2023-08-19 by thisiskeithb

stoubian, 2023-08-19:
I am using the code revision at tag 2.1.2.1 and the error is in that version but I see it has been fixed by this commit on the branch bugfix-2.1.x : [link to to the commit](https://github.com/MarlinFirmware/Marlin/commit/92721d1bf7d3516ffa591bf7fa9b0b03099f5156)

thisiskeithb, 2023-08-19:
> I am using the code revision at tag 2.1.2.1 and the error is in that version but I see it has been fixed by this commit on the branch bugfix-2.1.x : [link to the commit](https://github.com/MarlinFirmware/Marlin/commit/92721d1bf7d3516ffa591bf7fa9b0b03099f5156)

That's why we ask if you've tested the latest `bugfix-2.1.x` before opening up new issues:

> Did you test the latest `bugfix-2.1.x` code?
> Yes, and the problem still exists.

stoubian, 2023-08-19:
My bad, I don't know why I was thinking that tag 2.1.2.1 was in advance of branch bugfix-2.1.x but this is clearly not the case 😅 
Thanks for your fast answer (and for maintianing the project)!