MarlinFirmware/Marlin #26190: [BUG] SanityCheck reports error when GRID_MAX_POINTS_X is set to 2

JudithRichardson, 2023-08-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am using a Lulzot Taz6. I am trying to update to the latest Marlin version (bugfix-2.1.x) from an older version (2.1.1). The Taz uses the nozzle as the probe, and needs to do bed levelling by probing on 4 washers in the corners of the bed. I do not have a BLTouch on this machine.
I need to use AUTO_BED_LEVELING_LINEAR to probe at these four specific positions. I have set the PROBING_MARGIN_* variables to specify these positions relative to the X/Y_MIN/MAX_POS and X/Y_BED_SIZE values.
In the older version setting #define GRID_MAX_POINTS_X 2 compiles without complaint, and the code runs fine on the Taz, probing at the intended 4 points.
In the bugfix-2.1.x SanityCheck.h has 
#error "GRID_MAX_POINTS_[XY] must be between 3 and 255."
which kills the build

### Bug Timeline

This seems to have changed some time between 2.1.1 and 2.1.2

### Expected behavior

I expected it to continue to compile with GRID_MAX_POINTS_X 2

### Actual behavior

Fails to compile

### Steps to Reproduce

#define AUTO_BED_LEVELING_LINEAR 
#define GRID_MAX_POINTS_X 2
try to compile in VSCode

### Version of Marlin Firmware

bugfix_2.1.x

### Printer model

Lulzbot Taz 6

### Electronics

stock electronics

### Bed Leveling

ABL Linear grid

### Your Slicer

Prusa Slicer

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I haven't included the full configuration files since you can reproduce this by just changing 2 lines in the distribution files

ellensp, 2023-08-20:
was added in https://github.com/MarlinFirmware/Marlin/pull/25831

thisiskeithb, 2023-08-22:
I updated the sanity check in https://github.com/MarlinFirmware/Marlin/pull/26199, but it is untested. Can you give this a test and let us know if that works for you?

thisiskeithb, 2023-08-22:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/26199

closed 2023-08-22 by thisiskeithb