MarlinFirmware/Marlin #26196: [BUG] Flow rate sometimes is 101%?

classicrocker883, 2023-08-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Sometimes I will start the printer and it will say 101% for flow. It should be defined as 100 on startup.

Even changing the value in **planner.cpp** - - `flow_percentage` = 150 (or any other value), it would turn on showing 150%. but at normal values 9 times out of 10 it will be 101%

I seem to have corrected the issue by changing the typedef `int16_t` to `uint16_t`.

Im just wondering if anyone else has experienced this or if *this* fix is the right fix.

having it unsigned vs signed just makes sense to me since Flow cannot be a negative number. unless it needs to be otherwise.

just looking for any ideas. thanks. 

### Bug Timeline

New, probably has been happening but only noticed recently 

### Expected behavior

should be 100% at every startup

### Actual behavior

shows 101% for flow

### Version of Marlin Firmware

current bugfix

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2023-08-21:
Please attach current `bugfix-2.1.x` configs that exhibit this issue.

Sonicboom247, 2023-08-21:
ok, I thought I was going crazy and hopefully this does apply, but I have seen where if I start with flow rate at 96% in my slicer, then I bump it one percent to 97% during the print, it now reads flow rate 100%. In the past it did read 97% or whatever percent I bumped it.  I have only had to tweak once or twice so never really paid it much attention.

mriscoc, 2023-08-23:
I saw that before, inclusive some variables with weird values only at power on and not on reboot. That can be an indicative of memory corruption. Sometimes it is only software related, such as a bad array/string/record length.

classicrocker883, 2023-08-25:
[Marlin_Configs.zip](https://github.com/MarlinFirmware/Marlin/files/12435347/Marlin_Configs.zip)

I have noticed that changing just one thing, such as from signed to unsigned `int` will cause things to not work right. or even having something initialized to not initialized, basically just a simple small change that you would think not matter be like the butterfly effect.

closed 2023-11-05 by github-actions[bot]