MarlinFirmware/Marlin #26203: [BUG] wifi_upload blocks temperature processing, so hotend and bed can overheat

vlsi, 2023-08-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Uploading a big file via wifi can cause overheat of the hotbed and hotend

### Expected behavior

wifi upload should not stop processing critical tasks like temperature monitoring

### Actual behavior

wifi_upload seem to stop all the other processes

Can it be caused by a spinloop in https://github.com/MarlinFirmware/Marlin/blob/f6de698fc8bf284c76a2ae09b1c7b5041c74d495/Marlin/src/lcd/extui/mks_ui/wifi_upload.cpp#L678-L681 ?

It looks like:
1. The file takes a lot to transfer
2. The transfer performs multiple packets, so it iterates many times in the loop
3. Every time it calls `watchdog_refresh`, so I guess it resets the watchdog timer, so watchdog never fires

I guess the fix would be moving `hal.watchdog_refresh();` outside of the loop.

### Steps to Reproduce

1. Start preheating
2. Start uploading a big file (e.g. 500M)

### Version of Marlin Firmware

2.1.1

### Printer model

FlyingBear Reborn 2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[R2S_2.1.1_GD32F407.zip](https://github.com/MarlinFirmware/Marlin/files/12421622/R2S_2.1.1_GD32F407.zip)

vlsi, 2023-11-03:
It is still relevant

ellensp, 2023-11-03:
Temperature control is on a timer interrupt, ie they keep happening...  

Did you actually verify this, or it it unfounded speculation?

vlsi, 2023-11-03:
Could you please hint where timer interrupt-based temp control is activated?

The "overheat during wifi transfer" issue is very known in Reborn 2 community.

Here's how it looks like: https://youtu.be/7MxLkfBDbUM (not mine)

I have pesronally run into the issue as well

vlsi, 2023-11-03:
Here's a typical precaution: https://t.me/reborn2support/42670

ellensp, 2023-11-03:
Video contains no useful information. Only more speculation.

`I haven't checked the firmware source code from their website yet, but I strongly suspect that Flying Bear disabled the thermal runaway protection for the units coming out of the factory.  This is not likely to be one of those rare cases where a mosfet fails "on" as the hot end stayed cold after powering on.
`

Motherboard listed In Config BOARD_MKS_NANO6 Is not even in stock Marlin

EvilGremlin, 2023-11-03:
BOARD_MKS_NANO6 do not exist in vanilla marlin, so your issue is kinda invalid in the first place. But MKS_UI is buggy orphaned code in general, so yeah, no wonder. Basically, don't use MKS_UI if you aren't willing to fix it yourself.

ellensp, 2023-11-03:
reborn 2 firmware https://drive.google.com/drive/folders/15Dw9_xIY9br-wts5MLN6GH2n5i6SZ7ik

ellensp, 2023-11-03:
also from the comments on that video "Had another thermal runaway event while preheating.  This time not using the wifi."

vlsi, 2023-11-03:
> Video contains no useful information.

Which information could help to diagnose and analyze the issue?

> Basically, don't use MKS_UI if you aren't willing to fix it yourself

Many users might be hit by the bug, so I wonder if something can be fixed/improved in Marlin itself so the upcoming mks-based builds still work.
For instance, there might be a trivial "disable heater" at the beginning of `wifi_upload` and "re-enable it after the upload is done.

> reborn 2 firmware

For reference, here's an exact firmware https://github.com/vlsi/reborn2-marlin/tree/2.1.1-rb2_autolevel (see commit https://github.com/vlsi/reborn2-marlin/commit/527c4aa9a8b28f3440138f3f2de7a8f897271f6c)

Unfortunately, the official firmware sources contain many irrelevant changes (like random whitespace change, a random addition of `/* C-declarations for C++ */` comments), so I've made a small cleanup in https://github.com/vlsi/reborn2-marlin/commit/6952091136c4d1bdd5a01292130e6322b5484122 so the diff is easier to analyze.

EvilGremlin, 2023-11-03:
Yep, but no one willing to fix many failings on MKS_UI yet, so general advice - just don't use it until it's fixed.

vlsi, 2023-11-03:
@EvilGremlin , I am willing to add "disable and re-enable heater to wifi_upload.cpp". Will that be accepted?

thisiskeithb, 2023-11-03:
> @EvilGremlin , I am willing to add "disable and re-enable heater to wifi_upload.cpp". Will that be accepted?

Since thermal runaways were reported with Wi-Fi disabled, that’s likely not going to fix the issue. Something else is causing these thermal runaway events.

vlsi, 2023-11-03:
>Something else is causing these thermal runaway events.

wifi transfer makes it 100% reproducible

thinkyhead, 2023-11-07:
One thing you can do to look for problems is to enable `1PINS_DEBUGGING` and then send `M43` to the machine to get a pins report. It's a little trickier to check on timer conflicts, but that is also possible. If neither of those is the case, then the next thing to look for is weird differences between a firmware that works, and some nearby version of Marlin that doesn't work.

closed 2024-09-23 by github-actions[bot]