MarlinFirmware/Marlin #26205: [BUG] Ender 3 V2 Dacai Display Rendering Bugginess

Soldr, 2023-08-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have tested Marlin latest release and bugfix-2.1.x.

The UI partially displays but is bugged out, however I can navigate the marlin menu.

I am using Marlin UI on an Ender 3 V2 with a V4.2.2. board and STMF103RE processor.

Creality UI is even worse as it has no text at all, but I do not have pictures of it.

### Bug Timeline

Today, when I installed Marlin2.1.x

### Expected behavior

The display works properly, and I see Marlin UI

### Actual behavior

I see Marlin UI but it is fragmented and displayed horribly

### Steps to Reproduce

1. Build Marlin (bugfix-2.1.x)
2. Flash the LCD (dacai) by the private folder in LCD_FILES
3. Flash the printer with the built marlin binary
4. Look at the screen

### Version of Marlin Firmware

Marlin bugfix-2.1.x (Aug 26 2023 00:26:44) 

### Printer model

Creality Ender 3 V2

### Electronics

Stock Ender 3 V2

### Add-ons

CR Touch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/12444594/Marlin.zip)

https://github.com/MarlinFirmware/Marlin/assets/66232265/f1d0ebd8-3909-4674-a300-4588e7054d31

[image]

[image]

ellensp, 2023-08-26:
See https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/Creality/Ender-3%20V2

DACAI
✅ Creality UI
✅ MRiscoC Pro UI
✅ JyersUI
✅ MarlinUI Portrait
❌ MarlinUI Landscape
Lack of support for display rotation

You CANNOT use DWIN_MARLINUI_LANDSCAPE on this display!
It is a limitation of the firmware in the display.

You also holding it in Portrait mode so should be using DWIN_MARLINUI_PORTRAIT

ellensp, 2023-08-26:
[image]

closed 2023-08-26 by ellensp