MarlinFirmware/Marlin #26206: [BUG] Z_MAX endstop is not operational in MKS_ROBIN_NANO_v1v2

bkiziuk, 2023-08-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I moved from some older bugfix-2.0.x to the current bugfix-2.1.x branch.
I found out that the printer does not respect Z_MAX endstop which was used during Z_HOMING process to ensure that (because of a need to provide clearance for Z_PROBE) the bed will not be lowered beyond the physical machine limits.

### Bug Timeline

No information since I've stopped to use newer versions of bugfix-2.0.x and jumped in to the recent bugfix-2.1.x

### Expected behavior

When hitting Z_MAX endstop during homing, the bed should stop moving down, but the process of homing should not be interrupted.

### Actual behavior

The Z_MAX endstop is not triggered.
The M119 command does not list Z_MAX endstop at all.

### Steps to Reproduce

1. Initiate Z_HOMING
2. press the Z_MAX endstop while the bed moves down

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

TwoTrees Sapphire Pro

### Electronics

MKS_ROBIN_NANO_1.2

### Add-ons

BL_TOUCH, Z_MAX endstop

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/12445837/config.zip)

M119 output:
17:29:47.540 : Reporting endstop status
17:29:47.542 : x_min: open
17:29:47.542 : y_max: open
17:29:47.545 : z_min: TRIGGERED
17:29:47.545 : filament: TRIGGERED

ellensp, 2023-10-26:
Is meant to be addressed here https://github.com/MarlinFirmware/Marlin/pull/25758 with Z_SAFETY_STOP

bkiziuk, 2023-10-27:
This works as expected. Thank you.

closed 2023-10-27 by bkiziuk