MarlinFirmware/Marlin #26207: Marlin 2.1.2.1 Laser control menu

bobgreen64, 2023-08-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have K40 laser with Arduino Mega + Ramp 1.4 board.  Everything works fine with CONFIGURATION_H_VERSION 02000902

Rebuild my code with same options on Marlin 2.1.2.1 #define CONFIGURATION_H_VERSION 02010201
found the PWM and laser menu controls are not working the same.  I manually edited the new config and advance config sheet to match my previously working config files.

### Expected behavior

Test pulse should give short laser burst
Toggle should toggle on and off the laser

### Actual behavior

Toggle remain indicated as OFF
Test Pulse trigger one pulse on the enable but the PWM remain on the other pin
Using the Toggle discontinue the PWM

### Steps to Reproduce

Laser Control Menu Action	Pin 4	Pin 6	Desc
1. Power on, pin 4 =0v, pin 6 =0v
2. press Laser Toggle, pin 4=0	pin 6=0, indication remain "OFF"
3. press Laser Toggle, pin4=0, pin 6=0, no change to indication 
4. press Test pulse, pin4=short_Pulse (010), pin6=PWM, indication shows Laser PWR =51 and PWM remain on pin 6
5. press Laser Toggle, pin4=0, pin6=0, indication no change but the PWM on pin 6 is discontinued.

### Version of Marlin Firmware

CONFIGURATION_H_VERSION 02010201

### Printer model

K40 laser

### Electronics

Arduino Mega 

### Add-ons

RAMP 1.4

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

bobgreen64, 2023-08-26:
[marlin2.1.2.1_ConfigFiles.zip](https://github.com/MarlinFirmware/Marlin/files/12446616/marlin2.1.2.1_ConfigFiles.zip)

Notrica-ir, 2023-08-29:
I have a problem like this 
Toggle remain indicated as OFF
Test Pulse trigger one pulse on the PWM pin and ends

I'm using SKR1.4 and Marlin 2.1.2.1

ellensp, 2023-08-30:
"press Laser Toggle, pin 4=0 pin 6=0, indication remain "OFF""

this is happening because you have FLOWMETER_SAFETY and it is not detecting coolant flow, so it will not allow you to turn on the laser.

Temporarily disable #define FLOWMETER_SAFETY       
And I bet  It will turn on...  

So you need to be looking into your LASER_COOLANT_FLOW_METER settings

bobgreen64, 2023-08-30:
I have the minimum flow set to 1.5 meter and read 2.2 L/min.  I will verify but I am pretty sure I did turn it off during my troubleshooting.  Also the flowmeter is setup the same with the previous version and it worked okay with the older code

bobgreen64, 2023-09-02:
> 
There is definitely a bug in the behavior.  I selected out the Flowmeter and now the toggle indicate ON and OFF but the laser remain active and will not shut off until I press the "Fire Pules" a few times.  This is in my original bug description.

  I KDiff the pins_RAMPS.h between the working version and the latest to confirm "M3/M4/M5 - Spindle/Laser Control" were on the same exact pins.  The only thing different is an extra lines around the PWM pin.  I also commented out 
          #ifndef + #endif line but that did not make any differences.
                #ifndef SPINDLE_LASER_PWM_PIN
                #define SPINDLE_LASER_PWM_PIN            6  // Hardware PWM
           #endif

I selected out the Flowmeter completely and now the toggle will change between ON and OFF but the laser remain firing.  Not even the test pulse have any effects.

I selected out the coolant temp with no luck.

closed 2023-11-16 by github-actions[bot]