MarlinFirmware/Marlin #26211: Ender-3 V2/S1 Professional Firmware 20230805 (LCD Dimming Option)

Ducktour, 2023-08-29:
### Describe the feature you want

The LCD dimming was introduced in Ender-3 V2/S1 Professional Firmware 20230805.

Can this be an optional toggle for those who prefer LCD fully visible at all times?

Ducktour, 2023-09-13:
Workaround; add M255 S0 in Start G-code

closed 2023-09-13 by Ducktour

thisiskeithb, 2023-09-13:
> The LCD dimming was introduced in Ender-3 V2/S1 Professional Firmware 20230805.

Feature requests for @mriscoc's firmware should go to his fork: https://github.com/mriscoc/Ender3V2S1

mriscoc, 2023-09-14:
Write a configuration [g-code file](https://github.com/mriscoc/Ender3V2S1/wiki/Configuration-files) with this command:
```gcode
M255 S0
```