MarlinFirmware/Marlin #26220: [BUG] 8 Axis on Octopus, no longer compiles

ellensp, 2023-09-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

8 axis no longer works

### Bug Timeline

Unknown

### Expected behavior

It should build

### Actual behavior

it errors with:
Marlin/src/HAL/STM32/../../inc/../core/types.h:982:68: error: expected unqualified-id before numeric constant
  982 |       bool LOGICAL_AXIS_LIST(E:1, A:1, B:1, C:1, _I:1, _J:1, _K:1, _U:1, _V:1, _W:1);
      |                                                                    ^~

### Steps to Reproduce

1. use Configs https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/linear_axes/Octopus%208%20LINEAR_AXES
2. build
3. see the error

### Version of Marlin Firmware

Bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2023-09-01:
Issue is in Marlin/src/core/types.h

```
    struct {
      bool LOGICAL_AXIS_LIST(E:1, A:1, B:1, C:1, _I:1, _J:1, _K:1, _U:1, _V:1, _W:1);

```

but A is defined as      #define A(CODE) " " CODE "\n\t"  in Marlin/src/core/macros.h
and _U is defined as  #define	_U	01   in  ,.platformio/packages/toolchain-gccarmnoneeabi@1.90201.191206/arm-none-eabi/includ/ctype.h

ellensp, 2023-09-01:
Simply adding  #undef _U before the strurt  will allow it to build.  But what does this break?

andrea199, 2023-09-01:
Yes exactly is strange that all the other letters do not have this error.
I tried with 6 axes and it compiles

[image]

sjasonsmith, 2023-11-19:
Symbols beginning with an underscore followed by a capital letter are reserved for use by the C++ implementation. Marlin should not be using any of these defined on this line.

batista352, 2023-12-19:
estou com o mesmo problema não consigo fazer funcionar mesmo com #undef _U #undef _V   fica assim 

*** [.pio/build/STM32H723ZE_btt/src/src/MarlinCore.cpp.o] Error 1
Compiling .pio/build/STM32H723ZE_btt/src/src/gcode/feature/trinamic/M906.cpp.o
In file included from Marlin/src/feature/../inc/MarlinConfigPre.h:37,
                 from Marlin/src/feature/../inc/MarlinConfig.h:28,
                 from Marlin/src/feature/babystep.cpp:23:
Marlin/src/feature/../module/stepper.h:92:60: error: 'I_ENABLE_PIN' was not declared in this scope; did you mean 'Z_ENABLE_PIN'?
   92 |   NUM_AXIS_LIST_(X_ENABLE_PIN, Y_ENABLE_PIN, Z_ENABLE_PIN, I_ENABLE_PIN, J_ENABLE_PIN, K_ENABLE_PIN, U_ENABLE_PIN, V_ENABLE_PIN, W_ENABLE_PIN)
      |                                                            ^~~~~~~~~~~~
Marlin/src/feature/../inc/../core/macros.h:322:44: note: in definition of macro 'LIST_8'
  322 | #define LIST_8( A,B,C,D,E,F,G,H,...) A,B,C,D,E,F,G,H
      |                                            ^
Marlin/src/feature/../inc/../core/macros.h:333:24: note: in expansion of macro '_LIST_N'
  333 | #define LIST_N(N,V...) _LIST_N(N,V)
      |                        ^~~~~~~
Marlin/src/feature/../inc/../core/types.h:41:31: note: in expansion of macro 'LIST_N'
   41 | #define NUM_AXIS_LIST(V...)   LIST_N(NUM_AXES, V)
      |                               ^~~~~~
Marlin/src/feature/../inc/../core/types.h:77:33: note: in expansion of macro 'NUM_AXIS_LIST'
   77 | #define NUM_AXIS_LIST_(V...)    NUM_AXIS_LIST(V) NUM_AXES_SEP
      |                                 ^~~~~~~~~~~~~
Marlin/src/feature/../module/stepper.h:92:3: note: in expansion of macro 'NUM_AXIS_LIST_'
   92 |   NUM_AXIS_LIST_(X_ENABLE_PIN, Y_ENABLE_PIN, Z_ENABLE_PIN, I_ENABLE_PIN, J_ENABLE_PIN, K_ENABLE_PIN, U_ENABLE_PIN, V_ENABLE_PIN, W_ENABLE_PIN)
      |   ^~~~~~~~~~~~~~

batista352, 2023-12-19:
quando não uso #undef _U  fica assim 

realloc(): invalid pointer
arm-none-eabi-gcc: internal compiler error: Aborted signal terminated program as
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
*** [.pio/build/STM32H723ZE_btt/SrcWrapper/src/stm32/clock.c.o] Error 4
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfigPre.h:37,
                 from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:28,
                 from Marlin/src/HAL/STM32/HAL.cpp:26:
Marlin/src/HAL/STM32/../../inc/../core/types.h:983:68: error: expected unqualified-id before numeric constant
  983 |       bool LOGICAL_AXIS_LIST(E:1, A:1, B:1, C:1, _I:1, _J:1, _K:1, _U:1, _V:1, _W:1);
      |                                                                    ^~
Marlin/src/HAL/STM32/../../inc/../core/macros.h:322:50: note: in definition of macro 'LIST_8'
  322 | #define LIST_8( A,B,C,D,E,F,G,H,...) A,B,C,D,E,F,G,H

ellensp, 2023-12-19:
This is issue is addressed in https://github.com/MarlinFirmware/Marlin/pull/26174, so closing

closed 2023-12-19 by ellensp

thisiskeithb, 2024-01-24:
FYI: The fix was broken out & comitted directly in https://github.com/MarlinFirmware/Marlin/commit/18e65f5eb4a1dc4b08d4bece67f2c182e8359ff6.

commit 18e65f5e (2024-01-23): 🩹 Fix _U and other conflicts

Fix #26220