MarlinFirmware/Marlin #26221: [DIRECT_STEP] DIRECT_STEPPING DO NOT SUPPORT MORE THEN 3 AXIS

andrea199, 2023-09-01:
### Is your feature request related to a problem? Please describe.

I'm trying to use the direct stepping mode for more than 3 axis but I get this error 

`In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:52,
                 from Marlin\src\HAL\STM32\MarlinSerial.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1073:4: error: #error "DIRECT_STEPPING does not currently support more than 3 axes (i.e., XYZ)."`

How can this problem be solved?

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/12500851/Configuration.zip)

### Describe the feature you want

It would be amazing to have the direct stepping mode for more then 3 axis

ellensp, 2023-09-02:
@colinrgodsey Would you be interested in extending your Direct Stepping feature for more axis?

andrea199, 2023-09-04:
I can try to do this and improve it but I don't know where to start, if someone could give me some advice it could be helpful.

colinrgodsey, 2023-09-05:
hmmm what's the extra axis for- is it an additional z-axis?

thisiskeithb, 2023-09-05:
> hmmm what's the extra axis for- is it an additional z-axis?

Marlin supports a bunch of new types:

https://github.com/MarlinFirmware/Marlin/blob/11f98adcce51f06a85ef0dd64e6e6b16139a34bd/Marlin/Configuration.h#L165-L181

On example of how it can be used: https://www.youtube.com/watch?v=Ej6fWSUrHws

It can be used for foam cutters, CNC, etc.

ellensp, 2023-09-05:
or robotics arns with N joints.

colinrgodsey, 2023-09-05:
ah yes, just curious if there was a specific use-case in mind that needed to leverage direct stepping. so there's nothing that really prevents us from expanding the protocol. the big issue just comes down to testing. the current support we have in place was developed against a sort of "reference" implementation  (https://github.com/colinrgodsey/step-daemon), which is an external planner/stepper for 3 axis cartesian. 

so it is possible to produce more page/chunk formats that can refer to more axes, but I worry about the best way to go about testing.

colinrgodsey, 2023-09-05:
so the TL;DR- nothing is really preventing this support, but it would be ideal to have a physical test device and the software to drive the protocol. so if there is a good use case that somebody has (and corresponding device), it'll make this a bit safer

PendulumPit, 2023-09-05:
Would this also extend the capabilities of Input Shaping?  I did point this stepdaemon to Tom but he said the development of it was old.

colinrgodsey, 2023-09-07:
input shaping is possible in step daemon, or whatever external planner is being used with direct stepping. there's also a beta version of linear advance, but one important feature missing from step daemon is look-ahead capable of slowing moves enough to cope with either those features. the raw math of either of those can produce really fast deltas that can exceed the possible step rate when operating at fast speeds. 

that and bezier curves i think are the biggest missing features of step daemon right now (outside of multi-axis support, delta, etc). its still pretty capable for normal cartesian printers, but yeah its been a while since I've worked on it. The main issues preventing development are just: I have no office space for my printer right now (new house, still renovating, office coming soon), and I got caught up trying to add support for 32-bit boards in marlin, which has proven to be a little odd. But i do plan to find some way to pick up development again this winter.