MarlinFirmware/Marlin #26222: [BUG] G61 does not return to saved position when Axis specified without an offset

karcaw, 2023-09-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When you use the G61 Return to Saved Position feature, and specify any axis to restore, and do not put a offset value, it does not return to the saved location, it does not change that axis at all.  It stems from Line 89:

https://github.com/MarlinFirmware/Marlin/blob/11f98adcce51f06a85ef0dd64e6e6b16139a34bd/Marlin/src/gcode/feature/pause/G61.cpp#L87-L89

Which defaults to the current position, and not to `stored_position[slot][i]` as it should.

The online line documentation here: https://marlinfw.org/docs/gcode/G061.html
does not even mention that offsets are an option, so anyone using this command with axis restores is bound to fail.(As I did)

### Bug Timeline

In the original code of G61 (#16557)

### Expected behavior

I expect it to move back to the proper position.

### Actual behavior

It does not move

### Steps to Reproduce

for Example if you run:
```
G28
G0 X10 Y10 Z10
G60
G0 X100 Y100 Z20
G61 X Y  
```
No movement is seen on the G61.  if instead you change that to `G61 X0 Y0` it will properly move back to the `X10 Y10` position.
a bare `G61` does work properly.

### Version of Marlin Firmware

Marlin 2.1.2

### Printer model

Creality Ender 3 Pro

### Electronics

Stock 4.2.2 Board

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[EF8_G16Debug.zip](https://github.com/MarlinFirmware/Marlin/files/12504987/EF8_G16Debug.zip)

karcaw, 2023-11-02:
yup, still valid

karcaw, 2024-05-31:
still not fixed

karcaw, 2024-07-16:
Any chance we can get this merged?

thisiskeithb, 2024-07-16:
> Any chance we can get this merged?

Since you've provided a fix, please open a proper [Pull Request](https://github.com/MarlinFirmware/Marlin?tab=readme-ov-file#contributing-patches) for review.

See [Contributing Code with Pull Requests](https://marlinfw.org/docs/development/getting_started_pull_requests.html) for more details.

karcaw, 2024-07-16:
I made one,...  Seems I pointed it at the wrong master: https://github.com/karcaw/Marlin/pull/1

karcaw, 2024-07-16:
Added proper PR

closed 2024-07-26 by thinkyhead