MarlinFirmware/Marlin #26224: [BUG] BTT SKR Mini E3 V3.0 and FAST_PWM_FAN_FREQUENCY

unitrix0, 2023-09-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I wanted to reduced the noise from my part cooling fans when not running on 100%. So I enabled `FAST_PWM_FAN`, but to what ever value I set `FAST_PWM_FAN_FREQUENCY`, the fans wont even turn on anymore. I've tried `125490`, which I think would be calculated value for my board and `25000`. And since I run 12V fans I set `FAN_MAX_PWM` to 110.

I have also found this older [bug report ](https://github.com/MarlinFirmware/Marlin/issues/22375) but I'm not sure what the conclusion is. Is it not possible to do this with this board?

### Expected behavior

Setting the PWM frequency by enabling `FAST_PWM_FAN` and setting `FAST_PWM_FAN_FREQUENCY`

### Actual behavior

The fans don't even turn on with 100% fan speed.

### Version of Marlin Firmware

Marlin bugfix-2.1.x (Sep  3 2023 12:41:13)

### Electronics

BTT SKR Mini E3 V3.0

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/12506355/Configurations.zip)

ellensp, 2023-09-03:
running 12v fan on 24v will not work long term

You still sending pulses of 24v to a 12v device  (its not just a restive load)

Also setting FAN_MAX_PWM to 110 is also wrong... if you trying to limit the power (if it was just a restive load) 

to get equivalent to power   you need 1/4 ie FAN_MAX_PWM 64

power = voltage * current

so for a 12ohm fan   (just for simplicity of the example)

@12v it will draw 1 amp.  so that is 12watts
@24 it will draw 2 amps, so that is 48watts 

ie 4 time the wattage, so to keep the same wattage you need 1/4 duty cycle , not approximately half

unitrix0, 2023-09-04:
Strange, for a moment I felt like I should never ever touch anything in my Marlin configuration again...

Anyways, I had the old 24V fans still laying around. So I commented out `FAN_MAX_PWM` and tried it again with `FAST_PWM_FAN_FREQUENCY = 25000`. Now the fans work when speed set to 100% but anything lower they stand still.

ellensp, 2023-09-04:
why did you ignore the warning?

```
Marlin/src/module/temperature.cpp: In member function 'void Temperature::init()':
Marlin/src/module/../inc/../../Configuration_adv.h:646:34: warning: unsigned conversion from 'int' to 'uint16_t' {aka 'short unsigned int'} changes value from '125490' to '59954' [-Woverflow]
  646 |   #define FAST_PWM_FAN_FREQUENCY 125490  // Define here to override the defaults below
      |                                  ^~~~~~
Marlin/src/module/temperature.cpp:376:64: note: in expansion of macro 'FAST_PWM_FAN_FREQUENCY'
  376 |   #define SET_FAST_PWM_FREQ(P) hal.set_pwm_frequency(pin_t(P), FAST_PWM_FAN_FREQUENCY)
      |                                                                ^~~~~~~~~~~~~~~~~~~~~~
Marlin/src/module/temperature.cpp:380:47: note: in expansion of macro 'SET_FAST_PWM_FREQ'
  380 | #define INIT_FAN_PIN(P) do{ _INIT_FAN_PIN(P); SET_FAST_PWM_FREQ(P); }while(0)
      |                                               ^~~~~~~~~~~~~~~~~
Marlin/src/module/temperature.cpp:2858:5: note: in expansion of macro 'INIT_FAN_PIN'
 2858 |     INIT_FAN_PIN(FAN0_PIN);
 
```

First thing is our tying to set to high a fan frequency  125490, but it is being truncated down to 59954
this has a maximum value of 65535

reduce this value to acceptable levels.

unitrix0, 2023-09-04:
why do you not read my comments completely?

ellensp, 2023-09-04:
also did you see "Setting very high frequencies can damage your hardware." ?

did you look at the specification of the mosfet to check it can actually do this high speed without burning?

Started with provided config files...  you got this HUGE warning and you ddnt even mention it...

unitrix0, 2023-09-04:
> ...tried it again with `FAST_PWM_FAN_FREQUENCY = 25000`.

closed 2023-09-04 by unitrix0