MarlinFirmware/Marlin #26225: [BUG] EEPROM and card support not workng

IoakeimSogiakas, 2023-09-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Weird issues when flashing the alfawise U20 printer. I discovered that not only does it not detect an sd card, it can't save to eeprom, but both are definitely defined. Also, I found out that flashing the non-maple environment sets the machine into a boot loop. Do you have any clue what I can do? This has happened on every version of Marlin that I have tried. Also although clearly enabled touch screen calibration auto-save is not working

### Bug Timeline

Old i presume

### Expected behavior

I am expecting the printer to be able to save to eeprom and read from it and also detected the presence of an sd carddetect

### Actual behavior

Although compiling with a breeze the behavior is as stated above

### Steps to Reproduce

Although compiling with a breeze the behavior is as stated above
[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/12506424/Configurations.zip)

### Version of Marlin Firmware

02010300

### Printer model

Alfawise U20

### Electronics

Sotck

### Add-ons

Able to print clay but compiling stock at the moment

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[image]
[image]

classicrocker883, 2023-09-25:
your additional file uploads aren't working.

what kind of bed leveling version are you using? Bilinear, Mesh, UBL?

IoakeimSogiakas, 2023-09-25:
I have by know dropped the project and forgot to close the issue sorry. If I remember correctly I tried both bilinear and ubl but they gave me the same thing

closed 2023-09-25 by IoakeimSogiakas