MarlinFirmware/Marlin #26227: [BUG] MKS_MONSTER_V2 NEOPIXEL_PIN redefined error

AssortedMapTacks, 2023-09-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When using a MKS Monster8 V2 board with the MKS MINI 12864 V3 (FYSETC_MINI_12864_2_1), the variable "NEOPIXEL_PIN" is defined twice. This is because the Monster8 V2 board includes a new neopixel pin on the board and was given a name that conflicts with the name of the neopixel pin used for the display. This throws an error when trying to build.

Marlin\src\pins\stm32f4\pins_MKS_MONSTER8_common.h
338: #define NEOPIXEL_PIN             EXP1_06_PIN

Marlin\src\pins\stm32f4\pins_MKS_MONSTER8_V2.h
55: #define NEOPIXEL_PIN                         PC5

Commenting out line 55 in V2.h allows successful build and use of neopixel on the display.

### Bug Timeline

unknown

### Version of Marlin Firmware

2.1.1.1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2023-09-04:
This is an ongoing issue for boards & LCDs with NeoPixels:
- https://github.com/MarlinFirmware/Marlin/issues/25236

thisiskeithb, 2023-12-03:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/26492